Koenkk/zigbee2mqtt #15711: Zigbee2MQTT stopped working after onth

different-thinking, 2022-12-23:
### What happened?

Good morning,

two days ago Zigbee2Mqtt just stopped working. I always get the following message:

`info 2022-12-23 07:06:39: Starting Zigbee2MQTT version 1.28.0 (commit #03ba647)`
`info 2022-12-23 07:06:39: Starting zigbee-herdsman (0.14.62)`
`error 2022-12-23 07:07:00: Error while starting zigbee-herdsman`
`error 2022-12-23 07:07:00: Failed to start zigbee`
`error 2022-12-23 07:07:00: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions`
`error 2022-12-23 07:07:00: Exiting...`
`error 2022-12-23 07:07:00: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)​``

Before that, it worked exactly in this configuration for several months (also with previous version).

I have tried the following:

-     worked through https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html - no solution
-     tried two different sticks
-         SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P (new)
-         Texas Instruments CC2531 (it worked with it for a long time)

Everything did not help.

How can I proceed now? 

### What did you expect to happen?

Plugin ist startung

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

UI not starting

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P & Texas Instruments CC2531

lux73, 2022-12-23:
https://www.zigbee2mqtt.io/guide/faq/#zigbee2mqtt-crashes-after-some-time

try to disable USB Autosuspend

different-thinking, 2022-12-23:
Thanks @lux73 . It's running on Raspi so I think there is no grub, isn't it.

lux73, 2022-12-23:
for sure there is grub in RasPi...  even RasPi is running a normal Linux Distri

different-thinking, 2022-12-23:
Unfortunately that doesn't solved the issue. @lux73

lux73, 2022-12-23:
what shows if you type:

`lsusb`

in ssh commandline?

different-thinking, 2022-12-23:
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 028: ID 10c4:ea60 Cygnal Integrated Products, Inc. CP2102/CP2109 UART Bridge Controller [CP210x family]
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

different-thinking, 2022-12-23:
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

lux73, 2022-12-23:
pkay your Stick is named

`Bus 001 Device 028: ID 10c4:ea60 Cygnal Integrated Products, Inc. CP2102/CP2109 UART Bridge Controller [CP210x family]`

connect the Stick and post the Output from:

`ls /dev/tty*`

different-thinking, 2022-12-23:
dmesg is showing this after inserting the stick:

 87.126072] usb 1-1.2: new full-speed USB device number 4 using xhci_hcd
[   87.271549] usb 1-1.2: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[   87.271559] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   87.271566] usb 1-1.2: Product: Sonoff Zigbee 3.0 USB Dongle Plus
[   87.271573] usb 1-1.2: Manufacturer: Silicon Labs
[   87.271581] usb 1-1.2: SerialNumber: 0001
[   87.275138] cp210x 1-1.2:1.0: cp210x converter detected
[   87.281413] usb 1-1.2: cp210x converter now attached to ttyUSB0

lux73, 2022-12-23:
seems good - the Device is connected to `ttyUSB0`

please show your z2m Configuration

different-thinking, 2022-12-23:
root@loxberry:/opt/loxberry# ls /dev/tty*
/dev/tty    /dev/tty11  /dev/tty15  /dev/tty19  /dev/tty22  /dev/tty26  /dev/tty3   /dev/tty33  /dev/tty37  /dev/tty40  /dev/tty44  /dev/tty48  /dev/tty51  /dev/tty55  /dev/tty59  /dev/tty62  /dev/tty9
/dev/tty0   /dev/tty12  /dev/tty16  /dev/tty2   /dev/tty23  /dev/tty27  /dev/tty30  /dev/tty34  /dev/tty38  /dev/tty41  /dev/tty45  /dev/tty49  /dev/tty52  /dev/tty56  /dev/tty6   /dev/tty63  /dev/ttyAMA0
/dev/tty1   /dev/tty13  /dev/tty17  /dev/tty20  /dev/tty24  /dev/tty28  /dev/tty31  /dev/tty35  /dev/tty39  /dev/tty42  /dev/tty46  /dev/tty5   /dev/tty53  /dev/tty57  /dev/tty60  /dev/tty7   /dev/ttyprintk
/dev/tty10  /dev/tty14  /dev/tty18  /dev/tty21  /dev/tty25  /dev/tty29  /dev/tty32  /dev/tty36  /dev/tty4   /dev/tty43  /dev/tty47  /dev/tty50  /dev/tty54  /dev/tty58  /dev/tty61  /dev/tty8   /dev/ttyUSB0

Stick should be connected to ttyUSB0
Serial by Id is: /dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0

different-thinking, 2022-12-23:
This is what I can configure at the loxberry interface:

[image]

lux73, 2022-12-23:
? you're not running zigbee2mqtt? i've never seen such an Interface

i don't know about loxberry - you should browse their Forums and ask there for Help - sorry

different-thinking, 2022-12-23:
Well, it is a zigbee2mqtt. It's just something like a installation an configuaration wrapper tp make it more easy to install.

lux73, 2022-12-23:
sure but we don't know what there has changed "under the hood"

so many Topics with z2m Problems related to third Party z2m Plugins... i'm not able to help, sorry

different-thinking, 2022-12-24:
I do understand your point. Thank you for your support.

closed 2022-12-24 by different-thinking