Koenkk/zigbee2mqtt #15716: [OTA] lumi.airrtc.agl001: Update of '0x54ef4410006751c4' failed (Failed to find endpoint which support OTA cluster)

lboue, 2022-12-23:
### What happened?

I am trying to update my lumi.airrtc.agl001 TRV and I am getting this error:
**Update of '0x54ef4410006751c4' failed (Failed to find endpoint which support OTA cluster)**

SRTS-A01 - Xiaomi Aqara Smart Radiator Thermostat E1
- modelId: "lumi.airrtc.agl001",
- installed_version":"0.0.0_0025"

```
Zigbee2MQTT:info  2022-12-23 14:37:22: Updating '0x54ef4410006751c4' to latest firmware
Zigbee2MQTT:info  2022-12-23 14:37:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Updating '0x54ef4410006751c4' to latest firmware","meta":{"device":"0x54ef4410006751c4","status":"update_in_progress"},"type":"ota_update"}'
Zigbee2MQTT:info  2022-12-23 14:38:22: MQTT publish: topic 'zigbee2mqtt/0x54ef4410006751c4', payload '{"away_preset_temperature":"5.0","battery":100,"calibrated":true,"child_lock":"LOCK","linkquality":255,"local_temperature":23.7,"occupied_heating_setpoint":22.5,"preset":"manual","sensor":"internal","system_mode":"heat","update":{"installed_version":"0.0.0_0025","state":"available"},"update_available":true,"valve_alarm":true,"valve_detection":"ON","voltage":null,"window_detection":"ON","window_open":false}'
Zigbee2MQTT:info  2022-12-23 14:38:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of '0x54ef4410006751c4' failed (Failed to find endpoint which support OTA cluster)","meta":{"device":"0x54ef4410006751c4","status":"update_failed"},"type":"ota_update"}'
Zigbee2MQTT:info  2022-12-23 14:38:22: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"0x54ef4410006751c4"},"error":"Update of '0x54ef4410006751c4' failed (Failed to find endpoint which support OTA cluster)","status":"error"}'
Zigbee2MQTT:error 2022-12-23 14:38:22: Update of '0x54ef4410006751c4' failed (Failed to find endpoint which support OTA cluster)
```

Is there a way to unlock this update?

Regards

### What did you expect to happen?

Upgrade to last firmware version: **0.0.0_0925_20221122**

```
    {
        "fileVersion": 2329,
        "fileSize": 261468,
        "manufacturerCode": 4447,
        "imageType": 5017,
        "modelId": "lumi.airrtc.agl001",
        "sha512": "40e4980fd6518dc32b19f6afd0aa3e49dd38ffa0aee1f2f7b05709cdf90273fb836e35468b6362e9756c63f19293907f50d381c379d5b9921074394e0cef4d62",
        "url": "https://github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/20221122180726_OTA_lumi.airrtc.agl001_0.0.0_0925_20221122_A6C469.ota",
        "path": "images/Xiaomi/20221122180726_OTA_lumi.airrtc.agl001_0.0.0_0925_20221122_A6C469.ota"
    }
```

### How to reproduce it (minimal and precise)

Send message to topic **zigbee2mqtt/bridge/request/device/ota_update/update**

### Zigbee2MQTT version

1.28.4 (commit #52e545f)

### Adapter firmware version

x

### Adapter

x

thunderj27, 2023-01-20:
Same story on my side.

paulbrause, 2023-01-21:
Same problem here

Koenkk, 2023-01-22:
This is fixed by https://github.com/Koenkk/zigbee-herdsman-converters/pull/5316 which is already in the dev branch (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) and will be included in the 1 February release.

closed 2023-01-22 by Koenkk