Koenkk/zigbee2mqtt #15718: [New device support]:  NEO Siren _TZE204_t1blo2bj

matteodallefeste, 2022-12-23:
### Link

https://it.aliexpress.com/item/4000813903996.html

### Database entry

{"id":5,"type":"Router","ieeeAddr":"0xa4c1380c0f78a599","nwkAddr":52004,"manufId":4417,"manufName":"_TZE204_t1blo2bj","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"��8+\u0013��8+e��8+\u0012","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":74}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1671817671957,"defaultSendRequestWhen":"immediate"}

### Comments

I'm trying to pair the siren, i got this error.

Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE204_t1blo2bj'

i have the Zigbee2MQTT Edge version 1.28.4-dev (commit #2550ca5)
and zigbee-herdsman (0.14.83)

i don't find any discussion about _TZE204_t1blo2bj devices.
I'm triing to follow this possible solution: https://giters.com/Koenkk/zigbee2mqtt/issues/10348
used for _TZE200_t1blo2bj version

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE204_t1blo2bj'}],
    model: 'NAS-AB02B2',
    vendor: 'Neo',
    description: 'Siren CUSTOM',
    fromZigbee: [fz.temperature, fz.humidity],
    toZigbee: [],
    exposes: [e.temperature(), e.humidity()],
    onEvent: tuya.onEventSetLocalTime,
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        const binds = ['msTemperatureMeasurement', 'msRelativeHumidity'];
        await reporting.bind(endpoint, coordinatorEndpoint, binds);
        await reporting.temperature(endpoint);
        await reporting.humidity(endpoint);
    },
};

module.exports = definition;
```

jerrm, 2022-12-24:
I think you tried a version of the converter that never worked.

There are two prior "\_TZE200_" versions of the siren.  

The one generally sold "with batteries" has the temp/humidity sensors, does not operate as a router, and will live a long life on batteries alone. My version of the "with" unit reports as TS0601/_TZE200_d0yu2xgi.

The one sold "without batteries" does not have the temp/humidity sensors.  It acts as a router, and while it does support installing batteries, they are only good for short term backup. The batteries will drain in a day or two if the device is not connected to mains.  My version of the "without" unit reports as TS0601/_TZE200_t1blo2bj.

Try the below converter, no promises. It's just taking the existing "without batteries" TZE200_t1blo2bj converter from the source tree and adds your "\_TZE20***4***_" fingerprint.

```
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_t1blo2bj'}, {modelID: 'TS0601', manufacturerName: '_TZE204_t1blo2bj'}],
        zigbeeModel: ['1blo2bj'],
        model: 'NAS-AB02B2 External Test',
        vendor: 'Neo External Test',
        description: 'Alarm',
        fromZigbee: [fz.neo_alarm, fz.ignore_basic_report],
        toZigbee: [tz.neo_alarm],
        exposes: [
            e.battery_low(),
            exposes.binary('alarm', ea.STATE_SET, true, false),
            exposes.enum('melody', ea.STATE_SET, Array.from(Array(18).keys()).map((x)=>(x+1).toString())),
            exposes.numeric('duration', ea.STATE_SET).withUnit('second').withValueMin(0).withValueMax(1800),
            exposes.enum('volume', ea.STATE_SET, ['low', 'medium', 'high']),
            exposes.numeric('battpercentage', ea.STATE).withUnit('%'),
        ],
        onEvent: tuya.onEventSetLocalTime,
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await endpoint.command('manuSpecificTuya', 'dataQuery', {});
            await endpoint.command('manuSpecificTuya', 'mcuVersionRequest', {'seq': 0x0002});
        },
    };

module.exports = definition;
```

matteodallefeste, 2022-12-26:
Hi @jerrm, thankyou for your fast reply.
I ordered the "with battery" version but i received a t1blo2bj model, so without humidity and temp sensor.
At first I didn't really understand how converters work. 
Your example works perfectly, thank you! 
I had found the same solution in the meantime and proposed a Merge Request where I added the fingerprint for the template '_TZE204_t1blo2bj'

matteodallefeste, 2023-01-26:
Resolved adding support for this device

closed 2023-01-26 by matteodallefeste

pabsi, 2023-12-11:
Sorry to sort of revive this by adding a comment in a closed PR, but before I open a new issue, I wanna make sure:
is it meant to not be reporting the battery level? I only have the `siren` entity and also only after I changed the device type of the switch...it'd be super convenient to have the battery level being reported (I already run out of the batteries shipped with the alarm and found out by surprise)..
thanks