Koenkk/zigbee2mqtt #15720: [New device support]: ZM-TB242A

rodrigogbs, 2022-12-23:
### Link

https://pt.aliexpress.com/item/1005003822349807.html?spm=a2g0o.order_list.order_list_main.4.21efcaa4XGDVOs&gatewayAdapt=glo2bra

### Database entry

{"id":28,"type":"Router","ieeeAddr":"0xa4c138118e92fcd9","nwkAddr":61844,"manufId":4417,"manufName":"_TZ3000_6s5dc9lx","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"o�6+\u0012","65506":54,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_6s5dc9lx","powerSource":1,"zclVersion":3,"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"powerOnBehavior":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024cae281","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":266,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":203301649},"lastSeen":1671822859671,"defaultSendRequestWhen":"immediate"}

### Comments

New device, variation of ZM-TB24, but with 1 20A socket.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
fingerprint: [
{
modelID: 'TS011F',
manufacturerName: '_TZ3000_6s5dc9lx'
}
],
model: 'TS011F_2_gang',
vendor: 'TuYa',
description: '2 gang wall outlet 20A',
extend: tuya.extend.switch({backlightMode: true, endpoints: ['l1', 'l2']}),
endpoint: () => {
return {'l1': 1, 'l2': 2};
},
meta: {multiEndpoint: true},
configure: async (device, coordinatorEndpoint) => {
const firstEndpoint = device.getEndpoint(1);
await firstEndpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
await reporting.bind(firstEndpoint, coordinatorEndpoint, ['genOnOff']);

const secondEndpoint = device.getEndpoint(1);
await secondEndpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
await reporting.bind(secondEndpoint, coordinatorEndpoint, ['genOnOff']);
}
};

module.exports = definition;
```

Koenkk, 2022-12-23:
Does it also work with this external converter?

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_mvn6jl7x'},
        {modelID: 'TS011F', manufacturerName: '_TZ3000_raviyuvk'}, {modelID: 'TS011F', manufacturerName: '_TYZB01_hlla45kx'},
        {modelID: 'TS011F', manufacturerName: '_TZ3000_92qd4sqa'}, {modelID: 'TS011F', manufacturerName: '_TZ3000_zwaadvus'},
        {modelID: 'TS011F', manufacturerName: '_TZ3000_k6fvknrr'}, {modelID: 'TS011F', manufacturerName: '_TZ3000_6s5dc9lx'}],
    model: 'TS011F_2_gang_wall',
    vendor: 'TuYa',
    description: '2 gang wall outlet',
    extend: tuya.extend.switch({backlightModeLowMediumHigh: true, childLock: true, endpoints: ['l1', 'l2']}),
    whiteLabel: [{vendor: 'ClickSmart+', model: 'CMA30036'}],
    endpoint: (device) => {
        return {'l1': 1, 'l2': 2};
    },
    meta: {multiEndpoint: true},
    configure: tuya.configureMagicPacket,
};

module.exports = definition;
```

rodrigogbs, 2022-12-26:
Yes it works, but withou the backlight configuration:

```
original version:
![Captura de Tela 2022-12-26 às 18 45 36](https://user-images.githubusercontent.com/71797145/209585056-7aa0945d-f2f9-4518-b94e-ad025ef2d11b.png)
![Captura de Tela 2022-12-26 às 18 45 30](https://user-images.githubusercontent.com/71797145/209585060-470ca7e9-2a6f-4d2d-b392-af244510a386.png)
```

```
new version:
![Captura de Tela 2022-12-26 às 18 31 00](https://user-images.githubusercontent.com/71797145/209585073-7599fc70-ed49-4d02-9619-2c9f1f55582c.png)
![Captura de Tela 2022-12-26 às 18 30 44](https://user-images.githubusercontent.com/71797145/209585075-71251229-6cb4-4082-a3f6-f4c7800f298a.png)
```

rodrigogbs, 2022-12-26:
With this little adjust it works pretty well with all functions:

`backlightMode: true`

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_mvn6jl7x'},
        {modelID: 'TS011F', manufacturerName: '_TZ3000_raviyuvk'}, {modelID: 'TS011F', manufacturerName: '_TYZB01_hlla45kx'},
        {modelID: 'TS011F', manufacturerName: '_TZ3000_92qd4sqa'}, {modelID: 'TS011F', manufacturerName: '_TZ3000_zwaadvus'},
        {modelID: 'TS011F', manufacturerName: '_TZ3000_k6fvknrr'}, {modelID: 'TS011F', manufacturerName: '_TZ3000_6s5dc9lx'}],
    model: 'TS011F_2_gang_wall',
    vendor: 'TuYa',
    description: '2 gang wall outlet',
    extend: tuya.extend.switch({backlightMode: true, childLock: true, endpoints: ['l1', 'l2']}),
    whiteLabel: [{vendor: 'ClickSmart+', model: 'CMA30036'}],
    endpoint: (device) => {
        return {'l1': 1, 'l2': 2};
    },
    meta: {multiEndpoint: true},
    configure: tuya.configureMagicPacket,
};

module.exports = definition;
```

[image]

Koenkk, 2022-12-27:
Is the backlight mode correct; should it be low, medium, high OR off, normal, inverted?

rodrigogbs, 2022-12-27:
My mistake, you're right, the behavior is OFF / NORMAL / INVERTED.

Koenkk, 2022-12-28:
Great, added the device.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-12-28 by Koenkk