Koenkk/zigbee2mqtt #15725: Ikea Styrbar remote not working anymore after upgrade to newest 2.4.5

rdu70, 2022-12-23:
### What happened?

After upgrading firmware version from 1.0.24 (20200708) stock firmware at device delivery to newest firmware 2.5.4 (20221028), Styrbar remote does not work anymore.  Pairing is working but after that no message are received and power status is reported as 255 instead of 0-100.
Update done using IKEA hub (zigbee2mqtt did not want to update saying 1.0.24 is newer that online version).  Remote is working fine when paired with IKEA hub.  Version 1.0.24 was working correctly throught zigbee2mqtt (and other ones not yet upgraded still works).
Log below show pairing and configuration, then a few OTA request from the device then nothing else happens even when pressing buttons.

### What did you expect to happen?

Remote with latest firmware working

### How to reproduce it (minimal and precise)

upgrade a remote to latest firmware from IKEA (2.5.4)

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20210708

### Adapter

Sonoff Zigbee 3 USB

### Debug log

debug 2022-12-23 23:04:29: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"ercpe-2","value":true}'
info  2022-12-23 23:04:29: Zigbee: allowing new devices to join.
info  2022-12-23 23:04:29: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"ercpe-2"}'
info  2022-12-23 23:04:33: Device '0x385b44fffe38ba50' joined
info  2022-12-23 23:04:33: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x385b44fffe38ba50","ieee_address":"0x385b44fffe38ba50"},"type":"device_joined"}'
info  2022-12-23 23:04:33: MQTT publish: topic 'homeassistant/sensor/0x385b44fffe38ba50/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x385b44fffe38ba50"],"manufacturer":"IKEA","model":"STYRBAR remote control (E2001/E2002)","name":"0x385b44fffe38ba50","sw_version":"2.4.5"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","name":"0x385b44fffe38ba50 battery","state_class":"measurement","state_topic":"zigbee2mqtt/0x385b44fffe38ba50","unique_id":"0x385b44fffe38ba50_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
info  2022-12-23 23:04:33: MQTT publish: topic 'homeassistant/sensor/0x385b44fffe38ba50/action/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x385b44fffe38ba50"],"manufacturer":"IKEA","model":"STYRBAR remote control (E2001/E2002)","name":"0x385b44fffe38ba50","sw_version":"2.4.5"},"enabled_by_default":true,"icon":"mdi:gesture-double-tap","name":"0x385b44fffe38ba50 action","state_topic":"zigbee2mqtt/0x385b44fffe38ba50","unique_id":"0x385b44fffe38ba50_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
info  2022-12-23 23:04:33: MQTT publish: topic 'homeassistant/sensor/0x385b44fffe38ba50/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x385b44fffe38ba50"],"manufacturer":"IKEA","model":"STYRBAR remote control (E2001/E2002)","name":"0x385b44fffe38ba50","sw_version":"2.4.5"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"0x385b44fffe38ba50 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x385b44fffe38ba50","unique_id":"0x385b44fffe38ba50_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
info  2022-12-23 23:04:33: MQTT publish: topic 'homeassistant/sensor/0x385b44fffe38ba50/update_state/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x385b44fffe38ba50"],"manufacturer":"IKEA","model":"STYRBAR remote control (E2001/E2002)","name":"0x385b44fffe38ba50","sw_version":"2.4.5"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:update","name":"0x385b44fffe38ba50 update state","state_topic":"zigbee2mqtt/0x385b44fffe38ba50","unique_id":"0x385b44fffe38ba50_update_state_zigbee2mqtt","value_template":"{{ value_json['update']['state'] }}"}'
info  2022-12-23 23:04:33: MQTT publish: topic 'homeassistant/binary_sensor/0x385b44fffe38ba50/update_available/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x385b44fffe38ba50"],"manufacturer":"IKEA","model":"STYRBAR remote control (E2001/E2002)","name":"0x385b44fffe38ba50","sw_version":"2.4.5"},"device_class":"update","enabled_by_default":false,"entity_category":"diagnostic","name":"0x385b44fffe38ba50 update available","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x385b44fffe38ba50","unique_id":"0x385b44fffe38ba50_update_available_zigbee2mqtt","value_template":"{{ value_json['update']['state'] == \"available\" }}"}'
info  2022-12-23 23:04:33: MQTT publish: topic 'homeassistant/update/0x385b44fffe38ba50/update/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/bridge/request/device/ota_update/update","device":{"identifiers":["zigbee2mqtt_0x385b44fffe38ba50"],"manufacturer":"IKEA","model":"STYRBAR remote control (E2001/E2002)","name":"0x385b44fffe38ba50","sw_version":"2.4.5"},"device_class":"firmware","entity_picture":"https://github.com/Koenkk/zigbee2mqtt/raw/master/images/logo.png","latest_version_template":"{% if value_json['update']['state'] == \"available\" %}{{ 'newer' }}{% else %}{{ value_json['update']['installed_version'] }}{% endif %}","latest_version_topic":"zigbee2mqtt/0x385b44fffe38ba50","name":"0x385b44fffe38ba50","payload_install":"{\"id\": \"0x385b44fffe38ba50\"}","state_topic":"zigbee2mqtt/0x385b44fffe38ba50","unique_id":"0x385b44fffe38ba50_update_zigbee2mqtt","value_template":"{{ value_json['update']['installed_version'] }}"}'
info  2022-12-23 23:04:33: Starting interview of '0x385b44fffe38ba50'
info  2022-12-23 23:04:33: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x385b44fffe38ba50","ieee_address":"0x385b44fffe38ba50","status":"started"},"type":"device_interview"}'
debug 2022-12-23 23:04:34: Device '0x385b44fffe38ba50' announced itself
info  2022-12-23 23:04:34: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x385b44fffe38ba50","ieee_address":"0x385b44fffe38ba50"},"type":"device_announce"}'
debug 2022-12-23 23:04:34: Received MQTT message on 'homeassistant/sensor/0x385b44fffe38ba50/battery/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x385b44fffe38ba50"],"manufacturer":"IKEA","model":"STYRBAR remote control (E2001/E2002)","name":"0x385b44fffe38ba50","sw_version":"2.4.5"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","name":"0x385b44fffe38ba50 battery","state_class":"measurement","state_topic":"zigbee2mqtt/0x385b44fffe38ba50","unique_id":"0x385b44fffe38ba50_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
debug 2022-12-23 23:04:34: Received MQTT message on 'homeassistant/sensor/0x385b44fffe38ba50/action/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x385b44fffe38ba50"],"manufacturer":"IKEA","model":"STYRBAR remote control (E2001/E2002)","name":"0x385b44fffe38ba50","sw_version":"2.4.5"},"enabled_by_default":true,"icon":"mdi:gesture-double-tap","name":"0x385b44fffe38ba50 action","state_topic":"zigbee2mqtt/0x385b44fffe38ba50","unique_id":"0x385b44fffe38ba50_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
debug 2022-12-23 23:04:34: Received MQTT message on 'homeassistant/sensor/0x385b44fffe38ba50/linkquality/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x385b44fffe38ba50"],"manufacturer":"IKEA","model":"STYRBAR remote control (E2001/E2002)","name":"0x385b44fffe38ba50","sw_version":"2.4.5"},"enabled_by_default":false,"ent [... cut ...]
debug 2022-12-23 23:07:21: Received Zigbee message from '0x385b44fffe38ba50', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":33816581,"imageType":4555,"manufacturerCode":4476}' from endpoint 1 with groupID 0
debug 2022-12-23 23:07:21: Device '0x385b44fffe38ba50' requested OTA
debug 2022-12-23 23:07:24: Received Zigbee message from '0x385b44fffe38ba50', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":33816581,"imageType":4555,"manufacturerCode":4476}' from endpoint 1 with groupID 0
debug 2022-12-23 23:07:24: Device '0x385b44fffe38ba50' requested OTA

MattWestb, 2022-12-24:
The batters _255_ is one undocumented future (= Bug) in the new firmware.
Also the firmware is having the right version in the GBL file so its working OK then getting it flashed on the device but the OTA meta data is wrong do ZHA. Z2M and Dirigera is not updating it then the version is lower then the device one. The old GW is only using the information from the update index file and is not looking at the meta data so it updating the device.

rdu70, 2022-12-26:
Thanks for that information.
Is it possible to downgrade the firmware as the old hub has automaticaly upgrade some of my remote wich are now useless in Zigbee2mqtt ?
Or will be some update to handle the remote with the new firmware ?

MattWestb, 2022-12-26:
I have not testing downgrading Syrbar but all first and second gen controller is blocked in the firmware for doing it and the original firmware was having the same settings if reading the certificate papers (the new is not posted as of today).
You can try but i think its not worth it better waiting for some one adding support for it in Z2M.
I have making one PR for fixing it in ZHA and have getting all function working OK but sadly it have loosing the group binding that was working in the original firmware.

namadori, 2022-12-30:
I have the styrbar remote on FW 1.0.024 (20200708) and I found out that some commands are not recognized anymore. Some are reporting the error **"Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'meta')}"**.

Commands working: Off, Brightness up, Brightness down, Brightness stop, Arrow right click, Arrow left click.
Commands not working: On, Arrow right hold, Arrow left hold, Arrow release

Are this somehow related?

sstrutyn, 2023-01-30:
I confirm that most recent (2.4.5) firmware completely broke support for Styrbar remotes. 
@rdu70 have you ended up with downgrading your remote, or perhaps there is a better temporary solution?

rdu70, 2023-02-03:
@sstrutyn Non I don't know how to proceed to downgrade.  If someone can provide an how-to, I'll be glad to test it.

darnox, 2023-02-11:
For me, manually pairing the remote with any bulb (using the button on the back) and then adding it to Z2MQTT got the remote on firmware v2.4.5 working again.

Freeswiss, 2023-02-11:
> For me, manually pairing the remote with any bulb (using the button on the back) and then adding it to Z2MQTT got the remote on firmware v2.4.5 working again.

So what you did was:
1. Manually pair the Remote with any Ikea Bulb.
2. Then pair the Remote with Z2MQTT.

Can you then unpair the Bulb again and the remote still works? Can the remote control other lights through Z2MQTT?

I'd like to test this.

darnox, 2023-02-11:
That is exactly what I did @Freeswiss. After this procedure the remote should start working as usual, so you are able to unbind it from the bulb and configure everything through Z2MQTT.

mdartu, 2023-02-12:
I am experiencing the same problem with a remote that has been upgraded to firmware 2.4.5 by the tradfri hub. It seems the `genOnOff` cluster has to be bound explicitly now in order for the remote to generate events (and the same may hold for the `genLevelCtrl` and `genScenes` clusters). Applying the following patch to `zigbee-herdsman-converters` works for me. However, I am not sure if this will break remotes with the older firmware, so this may need to be conditional on firmware version. 

```diff
diff --git a/devices/ikea.js b/devices/ikea.js
index 7a4f3cdd..e2a2a9ca 100644
--- a/devices/ikea.js
+++ b/devices/ikea.js
@@ -650,7 +650,7 @@ module.exports = [
         configure: async (device, coordinatorEndpoint, logger) => {
             // Binding genOnOff is not required to make device send events.
             const endpoint = device.getEndpoint(1);
-            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
+            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'genOnOff']);
             await reporting.batteryPercentageRemaining(endpoint);
         },
     },
```

Koenkk, 2023-02-13:
Could you check if the issue is fixed with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['Remote Control N2'],
    model: 'E2001/E2002',
    vendor: 'IKEA',
    description: 'STYRBAR remote control',
    fromZigbee: [ikea.fz.remote_battery, ikea.fz.styrbar_on, fz.command_off, fz.command_move, fz.command_stop, fz.ikea_arrow_click,
        fz.ikea_arrow_hold, ikea.fz.styrbar_arrow_release],
    exposes: [e.battery().withAccess(ea.STATE_GET), e.action(['on', 'off', 'brightness_move_up', 'brightness_move_down',
        'brightness_stop', 'arrow_left_click', 'arrow_right_click', 'arrow_left_hold',
        'arrow_right_hold', 'arrow_left_release', 'arrow_right_release'])],
    toZigbee: [tz.battery_percentage_remaining],
    ota: ota.tradfri,
    meta: {battery: {dontDividePercentage: true}},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        const version = device.softwareBuildID.split('.').map((n) => Number(n));
        // https://github.com/Koenkk/zigbee2mqtt/issues/15725
        const v245OrLater = version[0] > 2 || (version[0] == 2 && version[1] >= 4);
        const binds = v245OrLater ? ['genPowerCfg', 'genOnOff'] : ['genPowerCfg'];
        await reporting.bind(endpoint, coordinatorEndpoint, binds);
        await reporting.batteryPercentageRemaining(endpoint);
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

Koenkk, 2023-02-13:
Updated https://github.com/Koenkk/zigbee2mqtt/issues/15725#issuecomment-1428389352, after this only on/off actions will works (but this is just to see if it works at all)

rdu70, 2023-02-14:
> Updated #15725 (comment)

Hi,

Thanks.  It start to works :-)

With this external converter :

Firmware v2.4.5 -> "Off" command working after pairing. "ON" do NOT work.
Default pairing show those link : PowerCfg, GenPollCtrl AND OnOff.  After
adding "LevelCtrl" link to coordinator, "brightness_move_up",
"brightness_move_down" and "brithness_stop" are working.  Battery is
showing 255%.

Firmware v1.0.24 -> "Off", "Brightness_move_up", "Brightness_move_down" and
"brigthness_stop" works directly after pairing. "ON" do NOT work.  Default
pairing show only PowerCfg and GenPollCtrl.  No OnOff and No LevelCtrl by
default.  Battery is showing correctly value less than 100%

Hope this helps.

rdu70, 2023-02-14:
Hi,
Made some more test after disabling the external converter to restore my other working devices.
Now, with no external converter it's working properly, including "on".
This seem to be due to the added binding of "OnOff" and "BrightnessCtrl".
Only the left and right arrow where not working, so I added "Scene" to the binding as seen in the following screenshot and everything work correctly.  Those binding where not needed in previous firmware and are not activated by default.
The only remaining problem : the battery is still reporting 255%
[image]

Koenkk, 2023-02-15:
@rdu70 does the `v1.0.24` work without an external converter after re-pairing (factory reset?)

rdu70, 2023-02-15:
@Koenkk Yes just tried now :

v1.0.24 work correctly without anything after pairing (I'm running zigbee2mqtt v1.30.1). In the bind section, only genPollCtrl and PwrCfg are selected (it's the default) and all functions works. Note that the "scenes" option does not appear in the binding list.

v2.5.4 -> if pairing without external converter, it pair but nothing works.  Binding only show the same 2 selected : genPollCtrl and PwrCfg and it's NOT possible to add other binding.

v2.5.4 -> if pairing WITH external converter, it pair AND 'off' works.  Binding show 3 binds by default (genPollCtrl, PwrCfg AND OnOff).  I'm able to add 2 other binding : LevelCtrl and Scene.  After that, all functions works EXCEPT "ON".  Then, I REMOVE the external converter, and everything works fine.

It seems that the v2.5.4 version needs the 3 more bindings (OnOff, LevelCtrl and Scenes) but they are not selected by default and without external converter it's not possible to do the binding (as this need to have a button pushed just after requesing the bind to wake up the remote).

In any cases, v2.5.4 does not repport the battery level correctly (show 255%)

Hope that helps

Koenkk, 2023-02-18:
- Can you confirm v1.0.24 also works with the external converter after re-pairing?
- Regarding the battery: can you stop z2m, remove battery for this device from data/state.json file and provide me the debug logging when the device publishes it battery?
- I integrated the fix for 2.4.5 (OnOff, LevelCtrl and Scenes bindings)

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Daviey, 2023-02-18:
Confirming that the docker image `latest-dev` with the above fix solves the issue for me, thanks!

rdu70, 2023-02-19:
Confirm also that `latest-dev` fix issues except battery.
Tested with 1.0.0024 and 2.4.5.  Removed and repaired, they work now correctly and the bindings are ok by default.
Thanks.

With 2.4.5, the battery now report 127.5

v1.0.24 -> Battery 90%
debug 2023-02-19 22:28:52: Received Zigbee message from '0x84ba20fffef7d3b6', type 'readResponse', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":90}' from endpoint 1 with groupID 0

v2.4.5 -> Battery 127.5%
debug 2023-02-19 22:26:27: Received Zigbee message from '0x385b44fffe38ba50', type 'readResponse', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":255}' from endpoint 1 with groupID 0

Full log below.  0x38xxx device is 2.4.5 and 0x84xxx is 1.0.24

info  2023-02-19 22:20:13: Starting Zigbee2MQTT version 1.30.1-dev (commit #7fa2b5f)
iinfo  2023-02-19 22:20:13: Starting zigbee-herdsman (0.14.93)

info  2023-02-19 22:20:15: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210708,"transportrev":2},"type":"zStack3x0"}'

info  2023-02-19 22:21:04: Zigbee: allowing new devices to join.

info  2023-02-19 22:25:11: Device '0x385b44fffe38ba50' joined
info  2023-02-19 22:25:11: Starting interview of '0x385b44fffe38ba50'
debug 2023-02-19 22:25:11: Device '0x385b44fffe38ba50' announced itself
debug 2023-02-19 22:25:13: Received Zigbee message from '0x385b44fffe38ba50', type 'readResponse', cluster 'genBasic', data '{"modelId":"Remote Control N2"}' from endpoint 1 with groupID 0
debug 2023-02-19 22:25:13: Skipping message, definition is undefined and still interviewing
debug 2023-02-19 22:25:13: Saving state to file /app/data/state.json
debug 2023-02-19 22:25:15: Received Zigbee message from '0x385b44fffe38ba50', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"IKEA of Sweden"}' from endpoint 1 with groupID 0
debug 2023-02-19 22:25:15: Skipping message, definition is undefined and still interviewing
debug 2023-02-19 22:25:16: Received Zigbee message from '0x385b44fffe38ba50', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"IKEA of Sweden"}' from endpoint 1 with groupID 0
debug 2023-02-19 22:25:16: Skipping message, definition is undefined and still interviewing
debug 2023-02-19 22:25:17: Received Zigbee message from '0x385b44fffe38ba50', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"IKEA of Sweden"}' from endpoint 1 with groupID 0
debug 2023-02-19 22:25:17: Skipping message, definition is undefined and still interviewing
debug 2023-02-19 22:25:18: Received Zigbee message from '0x385b44fffe38ba50', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"IKEA of Sweden"}' from endpoint 1 with groupID 0
debug 2023-02-19 22:25:18: Skipping message, definition is undefined and still interviewing
debug 2023-02-19 22:25:19: Received Zigbee message from '0x385b44fffe38ba50', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
debug 2023-02-19 22:25:19: Skipping message, definition is undefined and still interviewing
debug 2023-02-19 22:25:21: Received Zigbee message from '0x385b44fffe38ba50', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
debug 2023-02-19 22:25:21: Skipping message, definition is undefined and still interviewing
debug 2023-02-19 22:25:27: Received Zigbee message from '0x385b44fffe38ba50', type 'readResponse', cluster 'genBasic', data '{"zclVersion":8}' from endpoint 1 with groupID 0
debug 2023-02-19 22:25:27: Skipping message, definition is undefined and still interviewing
warn  2023-02-19 22:25:28: Device '0x385b44fffe38ba50' left the network
info  2023-02-19 22:25:47: Device '0x385b44fffe38ba50' joined
info  2023-02-19 22:25:48: Starting interview of '0x385b44fffe38ba50'
debug 2023-02-19 22:25:48: Device '0x385b44fffe38ba50' announced itself
debug 2023-02-19 22:25:49: Received Zigbee message from '0x385b44fffe38ba50', type 'readResponse', cluster 'genBasic', data '{"appVersion":1}' from endpoint 1 with groupID 0
debug 2023-02-19 22:25:50: Received Zigbee message from '0x385b44fffe38ba50', type 'readResponse', cluster 'genBasic', data '{"stackVersion":106}' from endpoint 1 with groupID 0
debug 2023-02-19 22:25:52: Received Zigbee message from '0x385b44fffe38ba50', type 'readResponse', cluster 'genBasic', data '{"stackVersion":106}' from endpoint 1 with groupID 0
debug 2023-02-19 22:25:58: Received Zigbee message from '0x385b44fffe38ba50', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
debug 2023-02-19 22:26:00: Received Zigbee message from '0x385b44fffe38ba50', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20221028"}' from endpoint 1 with groupID 0
debug 2023-02-19 22:26:02: Received Zigbee message from '0x385b44fffe38ba50', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"2.4.5"}' from endpoint 1 with groupID 0
debug 2023-02-19 22:26:08: Received Zigbee message from '0x385b44fffe38ba50', type 'readResponse', cluster 'genPollCtrl', data '{"checkinInterval":172800}' from endpoint 1 with groupID 0
info  2023-02-19 22:26:08: Successfully interviewed '0x385b44fffe38ba50', device has successfully been paired
info  2023-02-19 22:26:08: Device '0x385b44fffe38ba50' is supported, identified as: IKEA STYRBAR remote control (E2001/E2002)
info  2023-02-19 22:26:09: Configuring '0x385b44fffe38ba50'
[... cut ...]
debug 2023-02-19 22:29:24: Is new image available for '0x385b44fffe38ba50', current '{"fieldControl":1,"manufacturerCode":4476,"imageType":4555,"fileVersion":33816581}', latest meta '{"fileVersion":581,"url":"http://fw.ota.homesmart.ikea.net/global/GW1.0/01.19.046/bin/10078031-zingo_kt_styrbar_remote-2.4.5.ota.ota.signed"}'
debug 2023-02-19 22:29:24: Update available for '0x385b44fffe38ba50': NO
warn  2023-02-19 22:29:24: Firmware on '0x385b44fffe38ba50' is newer than latest firmware online.
debug 2023-02-19 22:29:25: Responded to OTA request of '0x385b44fffe38ba50' with 'NO_IMAGE_AVAILABLE'
`

sjorge, 2023-02-20:
@Koenkk 

```
debug 2023-02-19 22:26:27: Received Zigbee message from '0x385b44fffe38ba50', type 'readResponse', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":255}' from endpoint 1 with groupID 0
```

Since we already have a separate IKEA convertor for battery, should we just reject invalid 255 battery readings or maybe map them to 100% , I guess we need a bit more data to see if there are any batteryPercentageRemaining values > 200 except 255 which is probably invalid as it's the MAX value that field type supports.

MattWestb, 2023-02-20:
@sjorge I have the same problem and is using alkaline batteries in my Styrbar and its recommend using rechargeable batteries that is having little lower voltage so it can being way its reporting to high battery % left.
Also reading battery voltage is giving 255 so its broken.

sjorge, 2023-02-20:
> @sjorge I have the same problem and is using alkaline batteries in my Styrbar and its recommend using rechargeable batteries that is having little lower voltage so it can being way its reporting to high battery % left. Also reading battery voltage is giving 255 so its broken.

Interesting, batteryVoltage also returning 255 does indeed seem to indicate 255 is just a 'oops bad value' on the firmware part. So the current converter that does `fwVer <= 2.3 ? 0-100 : 0-200;` is probably fine.

MattWestb, 2023-02-20:
With Tradfri they is testing the direct parring with lights (ala ZLL) and the bridge is using the same working model then its one` Zigbee Controller Bridge` and is not doing so mush work on the normal working mode (with coordinator).

Dirigera is working in normal `coordinator` working mode and but is not using all functions that we is using and they is not using group binding all is going thru the GW. But they still  must have the direct binding with ZLL for remotes controlling lights with GW or hubs.
And Dirigera is doing one very nice pairing of devices using `find and bind`  that i have not sen other using and is getting all possible functions from the device.

Styrbar was the first 2.4 / 24 version controller and its look being fixed for all other is reporting 200% as Zigbee standard (nearly all controller is getting update to V24 then being added in Dirigera also the blinds).

Im very sure its one firmware bug and shall being fixed but its likely IKEA is not  being aware of it.

In the end IKEA is one of the most Zigbee 3 compatible devices on the market and we cant comparing with Aqara and tuya that only have reading the half of the specification before shipping there devices.

By the way Synfonisk 2 and new  OnOff, OpenClose and Shortcutt buttons ver 2 have getting FCC cert so is coming in some mounts (If CE can delivering them).

Koenkk, 2023-02-20:
I think we should map any value above 200 to 100%, agree?

sjorge, 2023-02-20:
Above @MattWestb said voltage also returns 255 in those case, with that info I's be more inclined to discard 255 entirely.

MattWestb, 2023-02-21:
Reported voltage of 255 is in 100mV = 25.5 V by ZCL R8  and is unrealistic for 98%  of all devices we is using and also its saying:
`3997 The value 0xff indicates an invalid or unknown reading.` 

For BatteryPercentageRemaining Attribute is ZCL R8 saying:
`4002 The value 0xff indicates an invalid or unknown reading.`

So in the end its somthing wrong  in the device firmware and the question is doing workaround or letting it until IKEA have fixing it ?

rdu70, 2023-02-21:
Indeed, the 255 is probably invalid reading and should be ignored.  After some time, my device where now at 85% (using rechargeable batteries fully charged, so about 2.7v).
[image]

I've trying to discharge a bit the batteries on another remote yesterday evening and after some time the battery is showing 70%, so it seems to update correctly.
[image]

closed 2023-04-01 by github-actions[bot]

victorhooi, 2023-08-16:
I have a E2002 remote as well - and for some reason, the OTA doesn't seem to be able to show the upgrade to the latest version (2.4.5).

My firmware is stuck at 1.0.024.

Do you know if the latest version (2.5.4) was somehow blacklisted on the Z2M side?

alienatedsec, 2023-08-16:
@victorhooi if you check logs you will find that the new firmware value is seen as lower value than the installed one and that prevents Z2M from proceeding. 

@Koenkk is there a workaround for this behaviour with IKEA Styrbar switches?

sjorge, 2023-08-16:
> The feeded OTA is the right version inside but the metadata is having wrong version tagged.
> ZHA have patching one OTA file with the right metadata that can being used for local update in Z2M  or patching Z2M for forcing it updating the device.

Hmm does that mean it was dumped wrong originally? Or that the ikea hubs do some weird stuff to push them?~ sjorge

alienatedsec, 2023-08-16:
> ZHA have patching one OTA file with the right metadata that can being used for local update in Z2M or patching Z2M for forcing it updating the device.

@MattWestb I am trying to understand the above - please can you explain better?

MattWestb, 2023-08-16:
@sjorge IKEA TF is not using the meta data tags in the OTA file only the list its downloading from the OTA feed (they have rolling back some light that way that was one bad update). 
And last i was looking in one 24 hour sniff Dirigera is trying rolling on the update agen every 24 hours on devices that is up to date so its looks it using the tagged metadata, but i have moving all Styrbar to one of my RCP test setups so i dont knowing if its doing now and is being updated by ZHA to current production version.

So patching Z2M or ZHA for overriding the version or using the ZHA posted metadata patched OTA file in one local OTA upgrading is working.

ZHA IKEA OTA discussion https://github.com/zigpy/zigpy/discussions/660 for more information.

MattWestb, 2023-08-16:
All IKEA controllers is blocked for firmware downgrading so its not possible flashing one older file then its running (not the meta data its the version inside the OTA file that is signed and cant being changed without braking it => the device is not flashing it).
I think all lights is not blocked then before it was possible doing downgrading for getting the "default scenes" IKEA is using for there remotes, but i have not testing with  the latest updated.

flechaig, 2023-08-19:
> * I integrated the fix for 2.4.5 (OnOff, LevelCtrl and Scenes bindings)

To make it clear for everyone - it was not obvious to me: all you need to do is to enable LevelCtrl, OnOff and Scenes on the first line of the bind tab of the 2.4.5 Styrbar device. To push theses settings to the device, remove the batteries, quickly click the bind button and put the batteries in.

evilmumi, 2023-11-03:
> That is exactly what I did @Freeswiss. After this procedure the remote should start working as usual, so you are able to unbind it from the bulb and configure everything through Z2MQTT.

How did you bind it with a bulb and Z2MQTT at the same time?

MattWestb, 2023-11-03:
@evilmumi It shall being enough deleting it restart the system and adding it new so its being new added and not the old configuration then the system must getting the new cluster / endpoints OK then its one functional firmware upgrade.

evilmumi, 2023-11-03:
I del

> @evilmumi It shall being enough deleting it restart the system and adding it new so its being new added and not the old configuration then the system must getting the new cluster / endpoints OK then its one functional firmware upgrade.

I deleted my_index.json, the  ext_converter.js and the firmware AND also undone the configuration.yaml change.

After a lot of pressing 4 times in 5 sec and 10 times fast i got the styrbar back in z2mqtt BUT no button press in LOG

adekloet, 2023-11-14:
I am able to have the Styrbar Ikea Remote device join with Z2MQTT, but the reporting of the button press etc. is not recorded and Z2MQTT reports "Null" as status. The HA status of the device is Unknown. Also binding of LevelCtrl, OnOff and Scenes is not working (also not with the batt out and quickly in after pressing bind. So although it seems supported the converter is not accepting all devices with the new firmware 2.4.5

It does work in the Ikea version 1 and the new bridge. It also works when connected straight to an Ikea lightbuld with touchlink, but not in HA.

Zigbee2MQTT version
    [1.33.2]

Coordinator type
    zStack3x0

Coordinator revision
    20221226

HawkanForce, 2023-12-23:
This is at all not obvious to me :) but I dont have z2m either :) a pain to get this styrbar to work