Koenkk/zigbee2mqtt #15729: No converter available for WSD500A (modelName: _TZE200_yjjdcqsq) despite current merge request intending to add it

TheRealMephisto, 2022-12-23:
### What happened?

Tried to set up a temperature sensor that just got delivered to me.
ModelId: TS0601 , manufacturerName: _TZE200_yjjdcqsq

Saw yesterday that it seemed to be supported in the dev branch of zigbee2mqtt, as that one uses the zigbee-herdsman-converters v15.0.4, where those details are included in devices/tuja.js

Still, the sensor does not report values that can be interpreted right by zigbee2mqtt.

I turned on debugging of my bare metal zigbee2mqtt version and tracked all the information that the sensor sends over a period of around 15 minutes (will attach file in this report).
The log output still states "No converter available for WSD500A [...]"
Figured out that some of the dp values correspond to some possible matches of the dp values with meaningful interpretations (temperature, humdity...) in the lib/tuya.js. Stopped there yesterday night.

Today I fund this merged pull request:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/5165

The first comment of the author states:
"Please don't merge this at the moment. Something is going wrong with that sensor. We get no Data from. I'll sniff the traffic for different datapoints."
A few days after, it was merged. Not sure if the issue was resolved - I think not, based on the problems I am facing right now.

I want to help solve this issue, but I just started looking into this awesome project.
So if some help / hints can be given, I'd appreciate that very much!

I have some similar sensor flying around, which work just fine.
modelID: TS0201 , manufacturerName: _TZ3000_bguser20
One idea I have is to start understand what the converter does with the information delivered from those and then later see what goes wrong with the data of the other sensor.

If this is a useless approach, please let me know!

If you know already how to solve this issue based on the information I've given, I'm looking forward to see the resolving commit!

### How to reproduce it (minimal and precise)

Use the described hardware via zigbee2mqtt

### Zigbee2MQTT version

1.28.4-dev commit: 5afda96c

### Adapter firmware version

20201127

### Adapter

Texas Instruments, Inc. CC2531 ZigBee

### Debug log

[ZigBeeTemperatureSensor_log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10297638/ZigBeeTemperatureSensor_log.txt)

Burki24, 2022-12-24:
@TheRealMephisto 

Hi, I'm waiting for that device to check the DPs. AS you said, it's not sending any data to Zigbee2MQTT. That's the reason, why my PR don't has to be merged now. 

If I get the device from the dealer, I will make an update to that. So we have to look on TuYa-Developer, which DPs we need.
I made the PR to fast, as a customer called me and said, that this device is the same as the others. After I made the PR, he called me back and said, the device doesn't send data. So please wait, or try to find the right DPs (if you have an original tuya coordinator).

Koenkk, 2022-12-24:
@Burki24 maybe this device requires the gateway connection status to be confirmed before it starts sending data. You can find an example [here](https://gist.github.com/Koenkk/8fc146f998e2bd9986df783164f4f3df). Make sure to update to the latest dev first, otherwise the gateway connection status command is not supported.

TheRealMephisto, 2022-12-24:
@Burki24 Thanks for the fast reply!

My device is sending data, I get values for dp (1, 2, 3, 9). 1 and 2 come with data values that look like temperature*10 and the value of humidity. (Well, not exactly, it comes formatted in an array with four elements, where the first 3 are always 0 and the last one is the value i mentioned. I could imagine this is already after some conversion of the values has happened)
Example quote from my attached log file: 
`Received Zigbee message from '0xa4c1381b13d63ffb', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,58],"type":"Buffer"},"datatype":2,"dp":2}],"seq":30208}' from endpoint 1 with groupID 0
`

But maybe I do have a different device? On my device is printed Model: ZTH01, instead of WSD500A. I do not find WSD500A anywhere on my device, the package or the manual.,
But the manufacturerName is the same like in your PR.
Could also be that I got something wrong here, so not so sure. To be honest, I am still confused about what all the different modelNames, modelIds, etc.

@Koenkk  Thanks for your answer as well! 
How can I determine wether my device needs the gateway connection status to be confirmed?

TheRealMephisto, 2022-12-24:
Adding to my Example quote from my attached log file: 
`Received Zigbee message from '0xa4c1381b13d63ffb', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,58],"type":"Buffer"},"datatype":2,"dp":2}],"seq":30208}' from endpoint 1 with groupID 0
`
Here, dp: 2 with a value of 58 (%) makes sense - I have another sensor closeby that is working fine with zigbee2mqtt and reporting around 58% humidity as well.

Koenkk, 2022-12-25:
Could you check if the issue is fixed with the following external converter:  https://gist.github.com/Koenkk/a5ef6a151f1cc63a01183b47e3404541

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

DrDuk, 2022-12-26:
I have the same device but marked ZTH01 and having the same manufacturer id. I've added the external converter (to the edge version running in supervisor and installed today).
I don't see any data returned. I've tried re-pairing but no joy.  Logs look like:

Zigbee2MQTT:info  2022-12-26 17:52:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"FreezerLowTemp"},"type":"device_announced"}'
Zigbee2MQTT:debug 2022-12-26 17:52:31: Retrieving state of 'FreezerLowTemp' after reconnect
Zigbee2MQTT:debug 2022-12-26 17:52:33: Received Zigbee message from 'FreezerLowTemp', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-12-26 17:52:34: Received Zigbee message from 'FreezerLowTemp', type 'attributeReport', cluster 'genBasic', data '{"65506":56,"65508":1,"appVersion":72}' from endpoint 1 with groupID 0

Burki24, 2022-12-27:
> Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/a5ef6a151f1cc63a01183b47e3404541
>
> * save this as file next to `configuration.yaml` as [...]

@Koenkk 

sorry, I had to wait for the device to ckeck. Your external converter isn't running. 
[image]
DataPoints are: 1=Temperature, 2=Humidity, 3=Battery_state, 9=Unit of temperature.

```json
{
  "result": {
    "category": "wsdcg",
    "functions": [
      {
        "code": "temp_unit_convert",
        "dp_id": 9,
        "type": "Enum",
        "values": "{\"range\":[\"c\",\"f\"]}"
      }
    ],
    "status": [
      {
        "code": "va_temperature",
        "dp_id": 1,
        "type": "Integer",
        "values": "{\"unit\":\"℃\",\"min\":-200,\"max\":600,\"scale\":1,\"step\":1}"
      },
      {
        "code": "va_humidity",
        "dp_id": 2,
        "type": "Integer",
        "values": "{\"unit\":\"%\",\"min\":0,\"max\":100,\"scale\":0,\"step\":1}"
      },
      {
        "code": "battery_state",
        "dp_id": 3,
        "type": "Enum",
        "values": "{\"range\":[\"low\",\"middle\",\"high\"]}"
      },
      {
        "code": "temp_unit_convert",
        "dp_id": 9,
        "type": "Enum",
        "values": "{\"range\":[\"c\",\"f\"]}"
      }
    ]
  },
  "success": true,
  "t": 1672180565372,
  "tid": "d97a1eb1863611edb1f19690e083ab74"
}
```
```json
{
  "result": {
    "active_time": 1672179339,
    "biz_type": 0,
    "category": "wsdcg",
    "create_time": 1672179339,
    "icon": "smart/icon/ay1525749833414yotNt/4bcc19327130246668c27a0a039d1670.png",
    "id": "bfc1f8579a0f9ee528nks6",
    "ip": "",
    "lat": "52.4127",
    "local_key": "9548d1080787ef4c",
    "lon": "13.5069",
    "model": " ZTH01ZTU温湿度传感器",
    "name": "T & H Sensor",
    "node_id": "a4c1384ab034b502",
    "online": false,
    "owner_id": "36291891",
    "product_id": "yjjdcqsq",
    "product_name": "T & H Sensor",
    "status": [
      {
        "code": "va_temperature",
        "value": 263
      },
      {
        "code": "va_humidity",
        "value": 47
      },
      {
        "code": "battery_state",
        "value": "high"
      },
      {
        "code": "temp_unit_convert",
        "value": "c"
      }
    ],
    "sub": true,
    "time_zone": "+02:00",
    "uid": "eu1629960823758Zk6eX",
    "update_time": 1672179339,
    "uuid": "a4c1384ab034b502"
  },
  "success": true,
  "t": 1672181838668,
  "tid": "d0624a6d863911edb1f19690e083ab74"
}
```
I'm not sure, if this information will help. If you need more information, please tell me.

Greetings
Burkhard

Koenkk, 2022-12-28:
Are you sure you added the converter to your z2m `configuration.yaml`? The device at least should be detected as supported.

Burki24, 2022-12-28:
```php
 cache_state: true
  cache_state_persistent: true
  cache_state_send_on_startup: true
  output: json
  last_seen: epoch
device_options:
  legacy: false
devices: devices.yaml
groups: groups.yaml
automations: automations.yaml
external_converters:
  - ext_converter.js
blocklist:
  - '0x0015bc003900293d'
  - '0xa4c1387e1e0dd63d'
  - '0xa4c1385beeafec7f'
  - '0xa4c13813a3fe7dd3'
  - ```
  
Yes, I'm sure. tried in two installations (Z-Stack 3.0 and EZSP v8) on both systems the same reaction.

Koenkk, 2022-12-28:
Updated https://gist.github.com/Koenkk/a5ef6a151f1cc63a01183b47e3404541, can you try again?

Burki24, 2022-12-28:
Now it's accepted, but didn't send any data, only transmission and last seen:
```txt
Info 2022-12-28 09:24:05MQTT publish: topic 'z2m-og/bridge/event', payload '{"data":{"friendly_name":"0xa4c1384ab034b502","ieee_address":"0xa4c1384ab034b502"},"type":"device_joined"}'
Info 2022-12-28 09:24:05MQTT publish: topic 'z2m-og/0xa4c1384ab034b502', payload '{"last_seen":1672215845087,"linkquality":127}'
Info 2022-12-28 09:24:05Starting interview of '0xa4c1384ab034b502'
Info 2022-12-28 09:24:05MQTT publish: topic 'z2m-og/bridge/event', payload '{"data":{"friendly_name":"0xa4c1384ab034b502","ieee_address":"0xa4c1384ab034b502","status":"started"},"type":"device_interview"}'
Info 2022-12-28 09:24:05MQTT publish: topic 'z2m-og/0xa4c1384ab034b502', payload '{"last_seen":1672215845327,"linkquality":127}'
Debug 2022-12-28 09:24:05Device '0xa4c1384ab034b502' announced itself
Info 2022-12-28 09:24:05MQTT publish: topic 'z2m-og/bridge/event', payload '{"data":{"friendly_name":"0xa4c1384ab034b502","ieee_address":"0xa4c1384ab034b502"},"type":"device_announce"}'
Debug 2022-12-28 09:24:05Received Zigbee message from '0xa4c1384ab034b502', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
Info 2022-12-28 09:24:05MQTT publish: topic 'z2m-og/0xa4c1384ab034b502', payload '{"last_seen":1672215845827,"linkquality":69}'
Debug 2022-12-28 09:24:06Received Zigbee message from '0xa4c1384ab034b502', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
Info 2022-12-28 09:24:06MQTT publish: topic 'z2m-og/0xa4c1384ab034b502', payload '{"last_seen":1672215846041,"linkquality":61}'
Debug 2022-12-28 09:24:07Retrieving state of '0xa4c1384ab034b502' after reconnect
Debug 2022-12-28 09:24:08Received Zigbee message from '0xa4c1384ab034b502', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Info 2022-12-28 09:24:08MQTT publish: topic 'z2m-og/0xa4c1384ab034b502', payload '{"last_seen":1672215848413,"linkquality":61}'
Info 2022-12-28 09:24:08Successfully interviewed '0xa4c1384ab034b502', device has successfully been paired
Info 2022-12-28 09:24:08Device '0xa4c1384ab034b502' is supported, identified as: TuYa Temperature & humidity sensor (TS0601_temperature_humidity_sensor)
Info 2022-12-28 09:24:08MQTT publish: topic 'z2m-og/bridge/event', payload '{"data":{"definition":{"description":"Temperature & humidity sensor","exposes":[{"access":1,"description":"Measured temperature value","name":"temperature","property":"temperature","type":"numeric","unit":"°C"},{"access":1,"description":"Measured relative humidity","name":"humidity","property":"humidity","type":"numeric","unit":"%"},{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0601_temperature_humidity_sensor","options":[{"access":2,"description":"Number of digits after decimal point for temperature, takes into effect on next report of device.","name":"temperature_precision","property":"temperature_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the temperature value (absolute offset), takes into effect on next report of device.","name":"temperature_calibration","property":"temperature_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for humidity, takes into effect on next report of device.","name":"humidity_precision","property":"humidity_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the humidity value (absolute offset), takes into effect on next report of device.","name":"humidity_calibration","property":"humidity_calibration","type":"numeric"}],"supports_ota":false,"vendor":"TuYa"},"friendly_name":"0xa4c1384ab034b502","ieee_address":"0xa4c1384ab034b502","status":"successful","supported":true},"type":"device_interview"}'
Info 2022-12-28 09:24:12MQTT publish: topic 'z2m-og/0xa4c1384ab034b502', payload '{"last_seen":1672215851874,"linkquality":58}'
Debug 2022-12-28 09:24:12Received Zigbee message from '0xa4c1384ab034b502', type 'attributeReport', cluster 'genBasic', data '{"65506":56,"65508":1,"appVersion":72}' from endpoint 1 with groupID 0
Info 2022-12-28 09:24:12MQTT publish: topic 'z2m-og/0xa4c1384ab034b502', payload '{"last_seen":1672215852001,"linkquality":54}'
Debug 2022-12-28 09:24:12Received Zigbee message from '0xa4c1384ab034b502', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
Info 2022-12-28 09:24:12MQTT publish: topic 'z2m-og/0xa4c1384ab034b502', payload '{"last_seen":1672215852555,"linkquality":58}'
Info 2022-12-28 09:25:07Accepting joining not in blocklist device '0xa4c1384ab034b502'
Info 2022-12-28 09:25:07MQTT publish: topic 'z2m-og/0xa4c1384ab034b502', payload '{"last_seen":1672215907571,"linkquality":58}'
Info 2022-12-28 09:25:07MQTT publish: topic 'z2m-og/0xa4c1384ab034b502', payload '{"last_seen":1672215907598,"linkquality":58}'
Debug 2022-12-28 09:25:07Device '0xa4c1384ab034b502' announced itself
Info 2022-12-28 09:25:08MQTT publish: topic 'z2m-og/bridge/event', payload '{"data":{"friendly_name":"0xa4c1384ab034b502","ieee_address":"0xa4c1384ab034b502"},"type":"device_announce"}'
[... cut ...]
Info 2022-12-28 09:27:35MQTT publish: topic 'z2m-og/0xa4c1384ab034b502', payload '{"last_seen":1672216054956,"linkquality":72}'
Debug 2022-12-28 09:27:35Received Zigbee message from '0xa4c1384ab034b502', type 'attributeReport', cluster 'genBasic', data '{"65506":56,"65508":1,"appVersion":72}' from endpoint 1 with groupID 0
Info 2022-12-28 09:27:35MQTT publish: topic 'z2m-og/0xa4c1384ab034b502', payload '{"last_seen":1672216055089,"linkquality":72}'
Info 2022-12-28 09:27:36MQTT publish: topic 'z2m-og/0xa4c1384ab034b502', payload '{"last_seen":1672216056154,"linkquality":69}'
```

Burki24, 2022-12-28:
Several times it loose connection and pair again from itself.

Burki24, 2022-12-28:
[image]
No information is comming through the device.

Burki24, 2022-12-28:
Getting some news:
[image]

But it isn't sending battery but battery_low and battery_level, so I added this device to the last part in your external converter.

Burki24, 2022-12-28:
For me, this device isn'z worth the money I spend for.

It's not stable and temp. and humidity not updated. I put it on my heated bed from the 3D-Printer (50°C) no effect, blowing inside with a hairdryer also brings no update of the measurement. Only linkquality and last seen will be updated.

DrDuk, 2022-12-28:
I've updated the external converter file but I still don't get data:

Zigbee2MQTT:debug 2022-12-28 10:32:16: Received Zigbee message from 'FreezerLowTemp', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-12-28 10:32:16: Received Zigbee message from 'FreezerLowTemp', type 'attributeReport', cluster 'genBasic', data '{"65506":56,"65508":1,"appVersion":72}' from endpoint 1 with groupID 0

Do the logs show that an external converter file is being loaded properly?

Burki24, 2022-12-28:
> I've updated the external converter file but I still don't get data:
>
> Zigbee2MQTT:debug 2022-12-28 10:32:16: Received Zigbee message from 'FreezerLowTemp', type 'read', cluster 'genTime', data [...]

No, all I did was to eleminate the entry in tuya.js "_TZE200_yjjdcqsq". But normally, if I remember, an external converter overwright the internal tuya.js.

Koenkk, 2022-12-29:
Updated https://gist.github.com/Koenkk/a5ef6a151f1cc63a01183b47e3404541, can you try to re-pair and see if it works better now?

DrDuk, 2022-12-29:
Still no joy.  I still only get link quality returned from the device.

Burki24, 2022-12-29:
I think, this evening, I have finished the Update for this device. I'm working on it and get the data from that device now. So hold on. :-)

Burki24, 2022-12-29:
Hi,

can you please test the following code as external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const {
    precisionRound, mapNumberRange, isLegacyEnabled, toLocalISOString, numberWithinRange, hasAlreadyProcessedMessage,
    calibrateAndPrecisionRoundOptions, addActionGroup, postfixWithEndpointName, getKey,
    batteryVoltageToPercentage, getMetaValue,
} = require('zigbee-herdsman-converters/lib/utils');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    gateway_connection_status: {
        cluster: 'manuSpecificTuya',
        type: ['commandMcuGatewayConnectionStatus'],
        convert: async (model, msg, publish, options, meta) => {
            // "payload" can have the following values:
            //  0x00: The gateway is not connected to the internet.
            // 0x01: The gateway is connected to the internet.
            // 0x02: The request timed out after three seconds.
            const payload = {payloadSize: 1,payload: 1}
            await msg.endpoint.command('manuSpecificTuya', 'mcuGatewayConnectionStatus', payload, {});
        },
    },
    tuya_temperature_humidity_sensor: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataReport', 'commandDataResponse'],
        options: [exposes.options.precision('temperature'), exposes.options.calibration('temperature'),
            exposes.options.precision('humidity'), exposes.options.calibration('humidity')],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            const dpValue = tuya.firstDpValue(msg, meta, 'tuya_temperature_humidity_sensor');
            const dp = dpValue.dp;
            const value = tuya.getDataValue(dpValue);
            switch (dp) {
            case tuya.dataPoints.tthTemperature:
                return {temperature: calibrateAndPrecisionRoundOptions(value / 10, options, 'temperature')};
            case tuya.dataPoints.tthHumidity:
                return {humidity: calibrateAndPrecisionRoundOptions(value, options, 'humidity')};
            case tuya.dataPoints.tthBatteryLevel:
                return {
                    battery_level: {0: 'low', 1: 'middle', 2: 'high'}[value],
                    battery_low: value === 0 ? true : false,
                };
            case tuya.datapoints.tthTemperatureUnit:
            //    return {temperature_unit_convert: {0x00: 'celsius', 0x01: 'fahrenheit'}[value]};
            default:
                meta.logger.warn(`zigbee-herdsman-converters:maa_tuya_temp_sensor: NOT RECOGNIZED ` +
                    `DP #${dp} with data ${JSON.stringify(dpValue)}`);
            }
        },
    },
}

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_yjjdcqsq']),
    model: 'TS0601_humidity_temperature_1',
    vendor: 'TuYa',
    description: 'Temperature and humidity',
    fromZigbee: [fzLocal.tuya_temperature_humidity_sensor, fzLocal.gateway_connection_status],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [e.temperature(), e.humidity(), e.battery_low(),
        exposes.enum('battery_level', ea.STATE, ['low', 'middle', 'high']).withDescription('Battery level state'), e.linkquality()],
    meta: {
        tuyaDatapoints: [
    //        [1, 'temperature', tuya.valueConverter.divideBy10],
    //        [2, 'humidity', tuya.valueConverter.raw],
    //        [3, 'battery', tuya.valueConverter.raw],
            [9, 'temperature_unit', tuya.valueConverter.raw],
        ],
    },
}

module.exports = definition;
```
For me, it runs. There are some difficults to pair this device with Zigbee2MQTT. If the pairing was successfull, please wait for a while (1-5 minutes) after that the data for temperature, humidity, battery_low and battery_level will be shown.

@Koenkk Can you check the code, I'm not the best coder. So I'm sure, there are some possibilities to optimize the code. Also the DP9 is an open issue. But it has no affect on the device. Even not with the original tuya-bridge.

Thanks and greetz

Burkhard

TheRealMephisto, 2022-12-30:
@Burki24 Thanks a lot for the code! I just added it as external converter and will test it. 

I uncommented the lines of the first the array elements of tuya_datapoints in definition and the return statement regarding the temperature unit, as I assume those should actually be in place as well.

After a few minutes of waiting, this is what I get: A constant announcement of reconnecting, accompanied with those messages:

- `Received Zigbee message from 'xxx', type 'attributeReport', cluster 'genBasic', data '{"65503":"\t\rA+\u0019\t\rA+\u0007\u0000\u0000\u0000\u0000\u0011�\rA+\u0019�\rA+\u0007\u0000\u0000\u0000\u0000\u0011�\rA+\u0019�\rA+\u0007\u0000\u0000\u0000\u0000\u0011"}' from endpoint 1 with groupID 0`

-`Received Zigbee message from 'xxx', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0`

-`Received Zigbee message from 'xxx', type 'attributeReport', cluster 'genBasic', data '{"65506":56,"65508":1,"appVersion":72}' from endpoint 1 with groupID 0`

Maybe my device and coordinator combination just takes longer to establish a connection that allows a meaningful value transmission.

I'll have it running over the next hours and check back in the frontend and the logs afterwards.

Burki24, 2022-12-30:
@TheRealMephisto 

this device sends only, when a parameter is changing. So in my automation software, I looked into the archive and found this:
[image]
It is connected to my TI CC2652P Stick.
I was not able to get it paired with my EFR32MG21 Stick. This device is really difficult to handle. And also, if zigbee2mqtt will say, the device is paired successfully (I tried it several times) sometimes it send data and sometimes, no data were send to zigbee2mqtt. Only linkquality was send everytime, I paired it.

Greetings
Burkhard

Koenkk, 2022-12-30:
@Burki24 I've improved your converter, please check with https://gist.github.com/Koenkk/a5ef6a151f1cc63a01183b47e3404541

Burki24, 2022-12-30:
> @Burki24 I've improved your converter, please check with https://gist.github.com/Koenkk/a5ef6a151f1cc63a01183b47e3404541

Many thanks @Koenkk. This converter runs well. I added only the battery_low expose. If it's okay for you, I will bring it up to tuya.js as PR. So it's no external converter needed.

Again: Many, many thanks to you for help!!!

Greetings
Burkhard

rtorrente, 2022-12-30:
> @Burki24 I've improved your converter, please check with https://gist.github.com/Koenkk/a5ef6a151f1cc63a01183b47e3404541

Hello @Koenkk,

I have a strange behaviour that let me think we have do not have the same version of libs or zigbee2mqtt when I use your last converter

When docker container boots, there is a typeError

```
Using '/app/data' as data directory
/app/dist/util/externally-loaded.js:36
    exposes: [e.temperature(), e.humidity(), tuya.exposes.batteryState()],
                                                          ^
TypeError: tuya.exposes.batteryState is not a function
    at /app/dist/util/externally-loaded.js:36:59
    at Script.runInContext (node:vm:141:12)
    at Script.runInNewContext (node:vm:146:17)
    at Object.runInNewContext (node:vm:306:38)
    at loadModuleFromText (/app/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:165:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)
```

I'm using the latest stable Zigbee2Mqtt docker image **1.28.4 commit: 52e545f**

Do you have any idea ?

aleskr, 2022-12-30:
Many thanks for your hard work and patience!

DrDuk, 2022-12-30:
> Hello @Koenkk,
>
> I have a strange behaviour that let me think we have do not have the same version of libs or zigbee2mqtt when I use your last converter
>
> When docker container boots, there is a [...]

@rtorrente
I had the same thing when running in the Master build.  I'm now in the dev version (Zigbee2MQTT Edge Addon in HomeAssistant) and I don't get this error and it works!  You need to rebind and wait a few minutes.

Thanks all of the above for their help.

closed 2022-12-31 by Koenkk

OneCreek, 2023-01-09:
for me it works as well with the ioBroker.zigbee: [image] with setup according to https://github.com/ioBroker/ioBroker.zigbee/issues/1649 except the voltage bug.