Koenkk/zigbee2mqtt #15730: [New device support]: Zigbee model 'TS004F' and manufacturer name '_TZ3000_ja5osu5g'

AlexSG92, 2022-12-24:
### Link

https://es.aliexpress.com/item/1005004774449897.html?spm=a2g0o.order_list.order_list_main.17.2962194dbvaiyF&gatewayAdapt=glo2esp

### Database entry

immediate"} {"id":17,"type":"EndDevice","ieeeAddr":"0xdc8e95fffece2f90","nwkAddr":54020,"manufId":4098,"manufName":"_TZ3000_ja5osu5g","powerSource":"Battery","modelId":"TS004F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,1,3,4,6,4096,57345],"outClusterList":[25,10,3,4,6,8,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TS004F","manufacturerName":"_TZ3000_ja5osu5g","stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1671286373879,"defaultSendRequestWhen":"immediate"}

### Comments

I’m trying to find a solution for this device but I was not lucky. Would it be possible to add it to the next update?

Thanks in advance.

Koenkk, 2022-12-24:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/3d982aef1f138ebae6021d517621e482

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

AlexSG92, 2022-12-24:
> Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/3d982aef1f138ebae6021d517621e482
>
> * save this as file next to `configuration.yaml` as [...]

I have followed your instructions and now I have this problem:

Failed to configure '0xdc8e95fffece2f90', attempt 1 (TypeError: Cannot read properties of undefined (reading 'bind') at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:40:24) at Object.configure (/app/dist/util/externally-loaded.js:33:25) at Configure.configure (/app/lib/extension/configure.ts:117:13))

Koenkk, 2022-12-24:
we can ignore that for now, does the button emit events when clicking?

AlexSG92, 2022-12-24:
Yes, I saw the events coming in until I tried the "hold" event and it stopped receiving any events.

AlexSG92, 2022-12-24:
I have re-paired the device and all events seem to come through without any problems.

Koenkk, 2022-12-25:
Great, can you check if everything is fine with: https://gist.github.com/Koenkk/3d982aef1f138ebae6021d517621e482 ?

AlexSG92, 2022-12-25:
Yes, everything seems to work correctly.

I migrated my zigbee system from Deconz to Zigbee2Mqtt recently and I couldn't be happier.

Thank you for everything.

Koenkk, 2022-12-26:
Great, happy xmas! Integrated the changes.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-12-26 by Koenkk

srett, 2024-02-11:
Just moved from ZHA to Z2M as device support in general appears to be better, but this one is a little wonky.

In ZHA, this button appeared as having two physical buttons which was a little weird (maybe the pairing button under the cover can actually emit events, dunno), but most importantly, for the main button it supported all three events as advertised by the manufacturer: press, double-press, long-press. Long-press is missing with Z2M. [This](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts004f.py#L385) seems to be there ZHA is handling this.

Also, and this might be more related to HA (but maybe someone here using this thing knows): Triggers in HA only work if the button is in command mode, and I create triggers for "on" (single-press) or "off" (double-press). When in event mode, I can *see* HA receiving "single" and "double" events in the button's log, but any triggers listening for these won't get triggered. This wouldn't be that bad, but unfortunately, a triple-press of the button toggles between command and event mode, "breaking" the button for the mere mortals in my household. :-) I guess I can make a trigger that would toggle back to command mode for the time being, but that seems like a bodge.