Koenkk/zigbee2mqtt #15736: Nous SZ-T04 - reporting interval

kiekerjan18, 2022-12-24:
### What happened?

Recently I convinced my father to move away from Tuya platform and join the power of Zigbee2Mqtt. He is comparing quite a lot and want at least what he gets on Tuya (and hopefully more).

Now we encounter with the following device: Nous SZ-T04 (https://www.zigbee2mqtt.io/devices/SZ-T04.html) an issue.

In Smartlife the minimal reporing interval is 1 minutes while in Zigbee2Mqtt this is only 5 minutes. If he sets the reporting interval in Smartfile to 1 minute and then pairs to Z2M the logs files get flooded with errors, since 1 minutes is not allowed, but the device do support it. 

### What did you expect to happen?

Any chance to adjust to a minimal reporting interval to 1 minute?

### Zigbee2MQTT version

1.28.4-dev commit: 58445a2

### Adapter firmware version

20221102

### Adapter

Sonoff Zigbee Dongle Plus

closed 2023-01-31 by github-actions[bot]

kiekerjan18, 2023-02-11:
I've solved it by creating an external converter and adjusting some values in the already available nous.js 

https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/nous.js

I've adjusted:

`            e.temperature(), e.humidity(), e.battery(),
            exposes.numeric('temperature_report_interval', ea.STATE_SET).withUnit('min').withValueMin(5).withValueMax(120).withValueStep(5)
                .withDescription('Temperature Report interval'),
            exposes.numeric('humidity_report_interval', ea.STATE_SET).withUnit('min').withValueMin(5).withValueMax(120).withValueStep(5)
                .withDescription('Humidity Report interval'),`

TO:

`            e.temperature(), e.humidity(), e.battery(),
            exposes.numeric('temperature_report_interval', ea.STATE_SET).withUnit('min').withValueMin(**1**).withValueMax(120).withValueStep(**1**)
                .withDescription('Temperature Report interval'),
            exposes.numeric('humidity_report_interval', ea.STATE_SET).withUnit('min').withValueMin(**1**).withValueMax(120).withValueStep(**1**)
                .withDescription('Humidity Report interval'),`

mati1988r, 2023-05-05:
> I've solved it by creating an external converter and adjusting some values in the already available nous.js
> 
> https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/nous.js
> 
> I've adjusted:
> 
> ` e.temperature(), e.humidity(), e.battery(), exposes.numeric('temperature_report_interval', ea.STATE_SET).withUnit('min').withValueMin(5).withValueMax(120).withValueStep(5) .withDescription('Temperature Report interval'), exposes.numeric('humidity_report_interval', ea.STATE_SET).withUnit('min').withValueMin(5).withValueMax(120).withValueStep(5) .withDescription('Humidity Report interval'),`
> 
> TO:
> 
> ` e.temperature(), e.humidity(), e.battery(), exposes.numeric('temperature_report_interval', ea.STATE_SET).withUnit('min').withValueMin(1).withValueMax(120).withValueStep(1) .withDescription('Temperature Report interval'), exposes.numeric('humidity_report_interval', ea.STATE_SET).withUnit('min').withValueMin(1).withValueMax(120).withValueStep(1) .withDescription('Humidity Report interval'),`
> 
> Is it possible to make this adjustment in the file of zigbee-herdsman-converters as well? @Koenkk

hi, can you send me file js with this change ?

kiekerjan18, 2023-05-05:
[nous.zip](https://github.com/Koenkk/zigbee2mqtt/files/11410190/nous.zip)