Koenkk/zigbee2mqtt #15738: Interview/Pairing Failure for Frient (Develco) Smoke Sensor

arwed01, 2022-12-24:
### What happened?

When trying to pair 4 new smoke sensors from Frient (Develco), the sensor fails to pair with a interview failure.

### What did you expect to happen?

I have bought before one sensor of the same type to check if it works with zigbee2mqtt and Home Assistant. Pairing was working fine without any issue. I expected the same for the additional 4 sensors of the same type (SMSZB-120)

### How to reproduce it (minimal and precise)

It fails 100% if I try to pair the new sensors.
New battery has been used und pairing have been tried several times in a series.
Coordinator is used with USB extension cable.

### Zigbee2MQTT version

1.28.4-1

### Adapter firmware version

see Dongle-Info in Debug-Log.

### Adapter

Sonoff Dongle-E (/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220816161418-if00)

### Debug log

Sonoff-Dongle:

DEVLINKS: >-
  /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220816161418-if00
  /dev/serial/by-path/pci-0000:00:05.0-usb-0:2:1.0
DEVNAME: /dev/ttyACM0
DEVPATH: /devices/pci0000:00/0000:00:05.0/usb2/2-2/2-2:1.0/tty/ttyACM0
ID_BUS: usb
ID_MODEL: SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2
ID_MODEL_ENC: SONOFF\x20Zigbee\x203.0\x20USB\x20Dongle\x20Plus\x20V2
ID_MODEL_ID: 55d4
ID_PATH: pci-0000:00:05.0-usb-0:2:1.0
ID_PATH_TAG: pci-0000_00_05_0-usb-0_2_1_0
ID_REVISION: '0442'
ID_SERIAL: ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220816161418
ID_SERIAL_SHORT: '20220816161418'
ID_TYPE: generic
ID_USB_DRIVER: cdc_acm
ID_USB_INTERFACES: ':020201:0a0000:'
ID_USB_INTERFACE_NUM: '00'
ID_VENDOR: ITEAD
ID_VENDOR_ENC: ITEAD
ID_VENDOR_ID: 1a86
MAJOR: '166'
MINOR: '0'
SUBSYSTEM: tty
TAGS: ':systemd:'
USEC_INITIALIZED: '568857523815'

Herdsman Debug - failed device: [0x0015bc0031011aa8] :

2022-12-24T13:52:04.341Z zigbee-herdsman:adapter:ezsp:uart --> [87009f7e]
2022-12-24T13:52:04.341Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: e780017500ff
2022-12-24T13:52:04.341Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x75: {"_id_":117,"_cls_":"findKeyTableEntry","_isRequest_":false,"index":255}
2022-12-24T13:52:04.342Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (0) success
2022-12-24T13:52:05.312Z zigbee-herdsman:adapter:ezsp:uart <-- [70a5b1a9702af1c7f18e4b14aae987499d4e27abec4a7e]
2022-12-24T13:52:05.312Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (7,0,0): 70a5b1a9702af1c7f18e4b14aae987499d4e27abec4a7e
2022-12-24T13:52:05.312Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (0)
2022-12-24T13:52:05.312Z zigbee-herdsman:adapter:ezsp:uart --> [8070787e]
2022-12-24T13:52:05.312Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: e790012400e475a81a013100bc150001000000
2022-12-24T13:52:05.313Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x24: {"_id_":36,"_cls_":"trustCenterJoinHandler","_isRequest_":false,"newNodeId":30180,"newNodeEui64":{"type":"Buffer","data":[0,21,188,0,49,1,26,168]},"status":1,"policyDecision":0,"parentOfNewNodeId":0}
2022-12-24T13:52:05.313Z zigbee-herdsman:adapter:ezsp:ezsp ==> findKeyTableEntry: {"address":{"type":"Buffer","data":[0,21,188,0,49,1,26,168]},"linkKey":true}
[... 64 lines cut ...]
2022-12-24T13:52:10.213Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x5: {"_id_":5,"_cls_":"nop","_isRequest_":false}
2022-12-24T13:52:10.213Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (3) success
2022-12-24T13:52:11.180Z zigbee-herdsman:adapter:ezsp:uart <-- [63a8b1a97d312a15b658964d24ab5593499c7d3aaf15e78a9874f1dec283fc7d5e7d3a05ebcdde6f8ffd7d38217e]
2022-12-24T13:52:11.180Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (6,3,0): 63a8b1a9112a15b658964d24ab5593499c1aaf15e78a9874f1dec283fc7e1a05ebcdde6f8ffd18217e
2022-12-24T13:52:11.181Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (7)
2022-12-24T13:52:11.181Z zigbee-herdsman:adapter:ezsp:uart --> [87009f7e]
2022-12-24T13:52:11.181Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: ea9001450000040102070101000100005488be0a44ffff0c18a10a000025a2000000000002
2022-12-24T13:52:11.181Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x45: {"_id_":69,"_cls_":"incomingMessageHandler","_isRequest_":false,"type":0,"apsFrame":{"profileId":260,"clusterId":1794,"sourceEndpoint":1,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":84},"lastHopLqi":136,"lastHopRssi":-66,"sender":17418,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[24,161,10,0,0,37,162,0,0,0,0,0]}}
2022-12-24T13:52:11.181Z zigbee-herdsman:adapter:ezsp:debg processMessage: {"messageType":0,"apsFrame":{"profileId":260,"clusterId":1794,"sourceEndpoint":1,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":84},"lqi":136,"rssi":-66,"sender":17418,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[24,161,10,0,0,37,162,0,0,0,0,0]}}
2022-12-24T13:52:11.184Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":161,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":37,"attrData":[0,162]}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":17418,"endpoint":1,"linkquality":136,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
Zigbee2MQTT:debug 2022-12-24 14:52:11: Received Zigbee message from '0x54ef4410005feb6b', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,162]}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-12-24 14:52:11: MQTT publish: topic 'zigbee2mqtt/0x54ef4410005feb6b', payload '{"current":null,"device_temperature":20,"energy":0.16,"linkquality":136,"power":0,"power_outage_memory":false,"state":"OFF","update":{"installed_version":"unknown","state":"idle"},"update_available":null,"voltage":null}'
2022-12-24T13:52:12.748Z zigbee-herdsman:adapter:ezsp:debg Node descriptor request for '30180' failed (Error: {"address":30180,"clusterId":32770,"sequence":34} after 10000ms), retry
2022-12-24T13:52:12.748Z zigbee-herdsman:controller:device:log Interview - quirks check for 'undefined'-'undefined'-'Unknown'
2022-12-24T13:52:12.748Z zigbee-herdsman:controller:device:log Interview - quirks did not match
2022-12-24T13:52:12.748Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x0015bc0031011aa8', attempt 6
2022-12-24T13:52:12.749Z zigbee-herdsman:controller:device:log Interview - quirks check for 'undefined'-'undefined'-'Unknown'
2022-12-24T13:52:12.749Z zigbee-herdsman:controller:device:log Interview - quirks did not match
2022-12-24T13:52:12.750Z zigbee-herdsman:controller:device:log Interview - failed for device '0x0015bc0031011aa8' with error 'Error: Interview failed because can not get node descriptor ('0x0015bc0031011aa8')
    at Device.interviewInternal (/app/node_modules/zigbee-herdsman/src/controller/model/device.ts:558:19)
    at Device.interview (/app/node_modules/zigbee-herdsman/src/controller/model/device.ts:445:13)
    at Controller.onDeviceJoined (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:567:17)'
2022-12-24T13:52:12.752Z zigbee-herdsman:controller:error Interview failed for '0x0015bc0031011aa8 with error 'Error: Interview failed because can not get node descriptor ('0x0015bc0031011aa8')'
Zigbee2MQTT:error 2022-12-24 14:52:12: Failed to interview '0x0015bc0031011aa8', device has not successfully been paired
Zigbee2MQTT:info  2022-12-24 14:52:12: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc0031011aa8","ieee_address":"0x0015bc0031011aa8","status":"failed"},"type":"device_interview"}'

arwed01, 2023-01-01:
I have ordered the latest Sonoff Dongle-Plus-P, if this makes a different and solves the problem.
Unfortunately I do not know because of the failed paring, what firmware is on the the sensors which do not work, if this is the reason. The working sensor is still working fine even after the update to the latest firmware 4.0.5 .

janogre, 2023-01-02:
OK, I ordered the SkyConnect dongle a couple of weeks ago. Hopefully this works with SMSZB-120.

Bascht74, 2023-01-17:
I have the same problem with my SMSZB-120

scontomio, 2023-01-24:
I have the same problem with that adapter and the Frient Develco AQSZB-110

janogre, 2023-01-29:
I revceived my SkyConnect this week, and exchanged the Sonoff Dongle-E with the new SkyConnect dongle.
I had to reinstall my devices, but the SMSZB-120 registered right away after this.
So there's something not working with the Dongle-E pairing to SMSZB-120 (and looks like other Frient-devices too).

arwed01, 2023-01-29:
That is good news, maybe I have to change to SkyConnect as well. 
In the meantime I also have received my Sonoff Dongle-Plus-P, but still same problem. Because with Dongle-E the Zigbee communication ist much better and more stable, I reinstalled my Dongle-E. After reinstalling suddenly at least one more SMSZB-120 have been paired. Strange!?!?

closed 2023-03-08 by github-actions[bot]

KoalaWerewolf, 2023-03-17:
I'm experiencing the same problem with the SONOFF ZBDongle-E and the Frient/Develco SMSZB-120.

prohand, 2023-04-10:
Hello,

Same problem with ZHA.
Can you solved the problem please ?

Thank you

Nightraider89, 2023-07-05:
I have the same problem, with my conbee II it worked fine, but with my sonoff stick ( i think the E version? at least the experimental one newest) one of the smoke detectors works fine and is recognized in z2mqtt. the other 3 i tried pairing a 100 times, and every time they fail..

KoalaWerewolf, 2023-07-28:
Firmware upgrade fixed my problem. Upgraded to 7.1.1. (was on 6.10.3)

Nightraider89, 2023-07-28:
How did you do that @KoalaWerewolf ? I dont know where to start

KoalaWerewolf, 2023-07-28:
The webflasher available here makes it easy: https://github.com/darkxst/silabs-firmware-builder

The webflasher has firmware available, but I don’t think it has 7.1.1. If you would like to use that version you can find it towards the end of this thread: https://github.com/Koenkk/zigbee-herdsman/issues/319

sdalu, 2023-11-20:
Unplugging + plugging the dongle again. Fixed it on my side.

Frankster-NL, 2024-11-02:
Installed the update, removed the device from Zigbee2MQTT, factory reset the device and then paired it. After it becoming fully available I hit the "re-interview" button, but still get the following errors:

```
error 2024-11-02 09:28:01z2m: Failed to configure 'Rookmelder keuken', attempt 2 (Error: Bind 0x0015bc0031014c9f/35 genPowerCfg from '0x0ceff6fffece2fee/1' failed (Status 'TABLE_FULL') at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23) at processTicksAndRejections (node:internal/process/task_queues:95:5) at runNextTicks (node:internal/process/task_queues:64:3) at processImmediate (node:internal/timers:449:9) at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:160:13) at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:211:17) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:205:21) at Configure.configure (/app/lib/extension/configure.ts:130:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:206:17))
error 2024-11-02 09:29:29z2m: Failed to configure 'Rookmelder keuken', attempt 3 (Error: Bind 0x0015bc0031014c9f/35 ssIasWd from '0x0ceff6fffece2fee/1' failed ({"target":52243,"apsFrame":{"profileId":0,"clusterId":33,"sourceEndpoint":0,"destinationEndpoint":0,"options":4416,"groupId":0,"sequence":183},"zdoResponseClusterId":32801} timed out after 15000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:195:23) at EmberOneWaitress.startWaitingFor (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:219:50) at /app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1940:51 at processTicksAndRejections (node:internal/process/task_queues:95:5) at runNextTicks (node:internal/process/task_queues:64:3) at processImmediate (node:internal/timers:449:9) at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20) at EmberAdapter.sendZdo (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1867:16) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:563:30) at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9))
info 2024-11-02 09:29:29z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Rookmelder keuken"},"error":"Failed to configure (Bind 0x0015bc0031014c9f/35 ssIasWd from '0x0ceff6fffece2fee/1' failed ({\"target\":52243,\"apsFrame\":{\"profileId\":0,\"clusterId\":33,\"sourceEndpoint\":0,\"destinationEndpoint\":0,\"options\":4416,\"groupId\":0,\"sequence\":183},\"zdoResponseClusterId\":32801} timed out after 15000ms))","status":"error","transaction":"mm8ku-5"}'
```