Koenkk/zigbee2mqtt #15739: Smart plugs turn off at random

Rayxcer, 2022-12-24:
### What happened?

My Xiaomi Mi Smart Plugs turn off at random. The device log files are varying between nothing, turned on (after I turn them back on on the plugs themself) and turned off and then turned back on.
This is really worrying because I use them as zigbee power monitors for e.g. my fridge.

### What did you expect to happen?

The plugs shouldn't turn off power by themself.

### How to reproduce it (minimal and precise)

I don't know. I wasn't able to reproduce it (yet).

### Zigbee2MQTT version

1.28.4-1 (HA addon)

### Adapter firmware version

20220219

### Adapter

Slaesh's CC2652RB stick

Rayxcer, 2022-12-31:
To update this: It might be related to joining new devices to the network. But it will not always happen instantly, sometimes it seems to delay the turning off for a (seemingly) random time. And it doesn't happen to all my smart plugs, just one of them.

At least that's where I'm at now, I'm still not sure exactly what happens.

MattWestb, 2023-01-01:
The device is member of one default group so then joining device is sending off to the member group its turning your fridge off.
Read the groups of the device and deleting all groups that you dont need / like then shall working OK.

closed 2023-02-08 by github-actions[bot]