Koenkk/zigbee2mqtt #15742: [New device support]: Immax NEO 07503L - Smart Controller v2 ZigBee 3.0 (TS110F)

Asuanin, 2022-12-24:
### Link

https://www.beleuchtung.at/immax-neo-07503l-smart-controller-smart-zigbee-3-0-tuya/

### Database entry

{"id":26,"type":"Router","ieeeAddr":"0x14b457fffe0e4b8f","nwkAddr":52434,"manufId":4098,"manufName":"_TZ3210_vfrok4rm","powerSource":"Mains (single phase)","modelId":"TS110F","epList":[1,2,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,4,5,6,8,61184,64770],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005","65506":31,"65508":0,"65534":0,"modelId":"TS110F","manufacturerName":"_TZ3210_vfrok4rm","stackVersion":0,"dateCode":"","zclVersion":3,"appVersion":160,"powerSource":1}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff06a7eb","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00212effff06a7eb","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":1,"inClusterList":[],"outClusterList":[6,8],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":1,"inClusterList":[],"outClusterList":[6,8],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":160,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-31628648},"lastSeen":1671912191960,"defaultSendRequestWhen":"immediate"}

### Comments

[img=https://cdn.alza.cz/ImgW.ashx?fd=f4&cd=IMXX19323&i=1.jpg]

I create a custom converter for Immax NEO 07503L - Smart Controller v2 using the one already exisiting for the model Fantem ZB006-X, looks like a copy.
Works fine but before pair need reset device through 5x on-off cycle. (**Hint for others**)

Tested:
Zigbee2MQTT: 1.28.4 commit: 52e545f
Coordinator: ConBee2
Revision of coordinator: 0x26780700

On/Off: OK
Brightness: OK
Triggered action: Not tested
External switch type: Not tested
External switch status: Not tested

### External converter

```shell
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

/* 
Copy of
    model: 'ZB006-X',
    vendor: 'Fantem',
*/
module.exports = [
    {
        fingerprint: [{modelID: 'TS110F', manufacturerName: '_TZ3210_vfrok4rm'}],
        model: '07503L',
        vendor: 'Immax',
        description: 'Immax NEO Smart Controller SMART ZigBee 3.0 (before pair need reset through 5x on-off cycle)',
        extend: extend.light_onoff_brightness({noConfigure: true}),
        fromZigbee: [...extend.light_onoff_brightness({noConfigure: true}).fromZigbee,
            fz.command_on, fz.command_off, fz.command_move, fz.command_stop, fz.ZB006X_settings],
        toZigbee: [...extend.light_onoff_brightness({noConfigure: true}).toZigbee, tz.ZB006X_settings],
        exposes: [e.light_brightness(),
            e.action(['on', 'off', 'brightness_move_down', 'brightness_move_up', 'brightness_stop']),
            exposes.enum('control_mode', ea.STATE_SET, ['ext_switch', 'remote', 'both']).withDescription('Control mode'),
            exposes.enum('switch_type', ea.STATE_SET, ['unknown', 'toggle', 'momentary', 'rotary', 'auto_config'])
                .withDescription('External switch type'),
            exposes.numeric('switch_status', ea.STATE).withDescription('External switch status')
                .withValueMin(-10000).withValueMax(10000),
            exposes.enum('load_detection_mode', ea.STATE_SET, ['none', 'first_power_on', 'every_power_on'])
                .withDescription('Load detection mode'),
            // If you see load_type 'unknown', pls. check with Tuya gateway and app and update with label from Tuya app.
            exposes.enum('load_type', ea.STATE, ['unknown', 'resistive_capacitive', 'unknown', 'detecting'])
                .withDescription('Load type'),
            exposes.enum('load_dimmable', ea.STATE, ['unknown', 'dimmable', 'not_dimmable'])
                .withDescription('Load dimmable'),
            exposes.enum('power_supply_mode', ea.STATE, ['unknown', 'no_neutral', 'with_neutral'])
                .withDescription('Power supply mode'),
        ],
        meta: {disableActionGroup: true},
        onEvent: tuya.onEventSetLocalTime,
        configure: async (device, coordinatorEndpoint, logger) => {
            await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
            const endpoint = device.getEndpoint(1);
            // Enables reporting of physical state changes
            // https://github.com/Koenkk/zigbee2mqtt/issues/9057#issuecomment-1007742130
            await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        },
    },
];
```

Asuanin, 2023-01-24:
Revival task

Koenkk, 2023-01-25:
Since you got it working, please make a PR for [zigbee-herdsman-converters](https://github.com/koenkk/zigbee-herdsman-converters) to add support for this device out-of-the-box.

closed 2023-03-05 by github-actions[bot]