Koenkk/zigbee2mqtt #15745: Aqara curtain driver e1 OTA fails

mjvanderveen, 2022-12-24:
### What happened?

ZNCLBL01LM aqare curtain driver e1 OTA update announced in both zigbee2mqtt as well as home assistant updates. But the update is unable to complete.

### What did you expect to happen?

OTA to complete

### How to reproduce it (minimal and precise)

Current version 0.0.0_0027

Go to zigbee2mqtt ota and press update firmware for the aqara curtain driver e1 track version.

### Zigbee2MQTT version

1.28.4-1

### Adapter firmware version

0100

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Debug log

info  2022-12-24 21:36:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Updating 'Curtain parents right' to latest firmware","meta":{"device":"Curtain parents right","status":"update_in_progress"},"type":"ota_update"}'

debug 2022-12-24 21:36:16: Received Zigbee message from 'Curtain parents right', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":27,"imageType":4105,"manufacturerCode":4447}' from endpoint 1 with groupID 0
debug 2022-12-24 21:36:16: Got OTA request '{"fieldControl":0,"manufacturerCode":4447,"imageType":4105,"fileVersion":27}'
debug 2022-12-24 21:36:16: ZigbeeOTA: downloaded main index
debug 2022-12-24 21:36:16: getNewImage for '0x54ef441000558255', meta {"fileVersion":6168,"fileSize":288213,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/20220506181329_lumi.curtain.agl001_Multi_JN5189_FMSH_0.0.0_2424_20220422_1aa302.ota","sha512":"0d1bae25759420898d381b98e97858d2e47f48cd3495abd8754661d0e1957eb5b570a2ae7435acc435d5a18665409e07148abae6f1f4a190c9af786d94ec725d"}
debug 2022-12-24 21:36:16: ZigbeeOTA: downloading firmware image from https://github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/20220506181329_lumi.curtain.agl001_Multi_JN5189_FMSH_0.0.0_2424_20220422_1aa302.ota
debug 2022-12-24 21:36:17: OTA update checksum validation succeeded for '0x54ef441000558255'
debug 2022-12-24 21:36:17: getNewImage for '0x54ef441000558255', image header {"otaUpgradeFileIdentifier":{"type":"Buffer","data":[30,241,238,11]},"otaHeaderVersion":256,"otaHeaderLength":56,"otaHeaderFieldControl":0,"manufacturerCode":4447,"imageType":4105,"fileVersion":6168,"zigbeeStackVersion":2,"otaHeaderString":"CURTAIN-OCC-JN5189---ENCRYPTED00","totalImageSize":288213}
debug 2022-12-24 21:36:17: Got new image for '0x54ef441000558255'
debug 2022-12-24 21:36:17: Starting upgrade

error 2022-12-24 21:38:47: Update of 'Curtain parents right' failed (Timeout: device did not request any image blocks)
debug 2022-12-24 21:38:47: Error: Timeout: device did not request any image blocks
    at /app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:360:28
    at runNextTicks (node:internal/process/task_queues:61:5)
    at listOnTimeout (node:internal/timers:528:9)
    at processTimers (node:internal/timers:502:7)

tjxism, 2023-01-07:
I am also experiencing this issue. Same issue for QBKG31LM. Update reported and can be started, however it will show failed to update in the end.

lundyfpv, 2023-01-18:
Seeing the same on my end, but I'm using the TubeZB coordinator if that matters somehow.

Uschinator, 2023-01-26:
Same. Using a Conbee 2
Aqara curtain driver E1 (ZNCLBL01LM)
Firmware: 0.0.0_0024

The Update starts and after it goes to 100% it shows the same Firmware again and if i check i can update the Firmware again. On home Assistant it shows the Version as 6168

"update": {
        "installed_version": 24,
        "latest_version": 6168,
        "state": "available"
    },
    "update_available": true,

VeeTeeDev, 2023-01-27:
Same problem :-)...

I have 3 and the sonoff stick model E.

[image]

One of them shows a progress bar... but when it's finished, nothing changes. The other 2 don't do anything...

pushnoi, 2023-01-28:
Yes, same problem over here. The version 6168 does not make any sense. Latest one available is supposed to be [Version: 0.0.0_2424](url)https://www.aqara.com/en/version/curtain-driver-e1-track-version(/url).

LayerCakeMakes, 2023-01-28:
Same issue here, I'm stuck on 0.0.0_0024.

NfkPollo, 2023-02-01:
I have same issue
[image]

nathan-curtis, 2023-02-03:
Bump - same issue and it doesn't look like an owner has been assigned yet?

klevin92, 2023-02-10:
same issue

mrmartin95, 2023-02-26:
Same issue

ChLah, 2023-03-02:
Same issue here.

WorlockM, 2023-03-06:
Same issue using Zigbee2MQTT and Conbee 2 stick.

pushnoi, 2023-03-26:
Who do i have to buy a beer to get this to be picked up?

ChLah, 2023-03-30:
I'd join you spending a beer

> Who do i have to buy a beer to get this to be picked up?

pushnoi, 2023-04-30:
Bump, still same issue.

Sacr87, 2023-04-30:
Same issue.

LayerCakeMakes, 2023-04-30:
Yes, very much should not be automatically closed, this is still a problem.

cranxi, 2023-05-05:
Same issue.

AppleFlapp, 2023-05-06:
looks like the same issue here.

Hotelk52339, 2023-05-06:
Same issue 

Is someone going to fix it ?

gr1spy, 2023-05-24:
Same issue :(

Joakim-T, 2023-06-20:
Same issue on both of mine

Both devices report firmware upgrades are available..

    "update": {
        "installed_version": 27,
        "latest_version": 6168,
        "state": "available"
    },

When i try to update it fails after a while with..

Update of 'Curtain Bedroom' failed (Timeout: device did not request any image blocks)

Dimmers81, 2023-06-26:
Same for me

rcursaru, 2023-06-29:
It also happens while using a Home Assistant SkyConnect dongle.
[image]
I am also willing to contribute to the (case of good) beer fund :)

pesbob, 2023-07-11:
Same issue. But the Aqara curtain driver e1, works fine. So no need for an update.

fbooij, 2023-07-27:
Same here on both Conbee2 and Sonoff E-dongle. Keeps saying there's an update but states `Update of 'cover_one' failed (Failed to find endpoint which support OTA cluster)` upon attempting to update

prokher, 2023-07-29:
Sorry for spamming, but I faced this issue too. It would be nice to have a workaround.

Dimmers81, 2023-08-01:
Still happening with MQTT 1.32.2 :(

Zicmu69, 2023-08-04:
Same issue

peds, 2023-08-19:
Same issue

AUkkers, 2023-09-02:
OTA issue
device version 0.0.0_0027
INFO: 2023-09-02 18:30:43Updating 'bedroom_window_cover_left' to latest firmware

INFO: 2023-09-02 18:30:43MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Updating 'bedroom_window_cover_left' to latest firmware","meta":{"device":"bedroom_window_cover_left","status":"update_in_progress"},"type":"ota_update"}'

INFO: 2023-09-02 18:33:16MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"bedroom_window_cover_left"},"error":"Update of 'bedroom_window_cover_left' failed (Timeout: device did not request any image blocks)","status":"error","transaction":"ur8sb-4"}'

ERROR: 2023-09-02 18:33:16Update of 'bedroom_window_cover_left' failed (Timeout: device did not request any image blocks)

Zigbee2MQTT 1.32.2
SONOFF Zigbee 3.0 USB Dongle Plus 20221102

smGoblin, 2023-09-05:
same for me 
OTA issue
device version 0.0.0_0027

SchoutenHans, 2023-09-13:
Same issue here current device version 0.0.0_0027

activatorr, 2023-09-17:
Yep, this issue for me 2, gateway 3 w/z2m

luchtgitaar, 2023-09-17:
I bought a Aqara E1 Hub. There is no update available via the Aqara app. I don't know why Z2M 'thinks' an update is available (maybe Chinese ZNCLBL01LM  vs CM-M01), but I don't think an update is available at all.

mute2014, 2023-09-26:
Same here. Got Version 27 but it wants to update all 3 to 6168

dexonter, 2023-10-27:
Issue is definitely very much still present! I have 12 of these drivers and it's quite annoying.

nathan-curtis, 2023-11-02:
@Koenkk Did this get somehow patched in the latest Z2M?  I upgraded to 1.33.2-1 this morning.  I usually expect to see my two Curtain drivers pop with a repair issue (every time I update supervisor or Z2M)  I'm 'showing hidden updates' and - they're not there?

Did something else fixed in 1.33.2-1 clear this?

cpressland, 2023-11-02:
@nathan-curtis I’m still seeing this.

nathan-curtis, 2023-11-02:
Man - i was excited for 30 seconds...

Koenkk, 2023-11-03:
@nathan-curtis 

>  I'm 'showing hidden updates' and - they're not there?

I don't understand what you mean with *showing hidden update*, could you elaborate on this?

nathan-curtis, 2023-11-03:
No problem K

Short version I've had this since it was first reported (early this year) so I've basically got into the habit of every time I updated supervisor or the Z2M addon, the reported issue with this device would pop as a repair item on my 'repair list' in HA.

So process becomes:
Update, update, update,
Reboot
Hide the curtain driver repair item (x2)
... Profit

It was the same all the way through 2023.10.x.  When 2023.11.x came out. The repair item didn't pop in again (for me) and at least three days later it still wasn't in my repair list this morning. I was so suprised ot wasn't there I went into the repair list ellipsis menu and selected show hidden updates... And they weren't there as expected. 

Does that help?

highincolorado, 2023-11-03:
I had stopped even trying to get my Xiaomi ZNCLBL01LM updated. It’s been at version 0.0.0_0024 and always says there’s an update available which after 30 minutes or so completes and it’s at the same version as well as update available. Had really hoped this was fixed by now, anything I can supply to help solve this?

Koenkk, 2023-11-03:
@nathan-curtis a bit, so this situation has improved for you?

nathan-curtis, 2023-11-03:
Correct. Currently I am at a current build of HA/Z2M. I have the same device mentioned in this thread at the same firmware level and up until recently saw the same issue.  My normal behavior was to hide the repair item in core.

/config/repairs currently reporting no repairs available and option : show ignored repairs is CHECKED 

Let me cold boot everything and verify it stays that way

nathan-curtis, 2023-11-03:
AND nope - there they are...  REBOOT Home Assistant, 
[image]

Also - I'll pull the log if necessary, but I've restarted this install probably 3-4 times since 2023.11.x came out - and I BELIEVE I installed the addon a few hours after the build dropped just before 2023.11.x was available for public install. if that helps timelines on what's going on.  Like I said, it was initially a different enough experience for me to come back and check.

LayerCakeMakes, 2023-11-04:
I don't think anything changed with this issue. I'm still having the exact same thing happen I'm on 0.0.0_0024 and it's offering an update after which I'm still on the same version and after checking for updates again the same update gets offered again.
[image]

nathan-curtis, 2023-11-16:
NO its still there and with the spinning device update failure introduced with this build. (That bug has already been fixed for next build and is in Edge already)

@Koenkk is there ANYTHING we can do about this?  What do you need to debug it?

randommen96, 2023-11-16:
For me it is also still present, however since a couple weeks now it is saying "installing" endlessly and I didn't trigger the update.

Drivers work fine though, so not really bothered.

Koenkk, 2023-11-16:
The installing endless in Home Assistant should be solved in the dev branch already  (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

nathan-curtis, 2023-11-16:
I'm talking about the offer for invalid firmware this issue is about I know the other one is fixed in Edge

Sent via the Samsung Galaxy S22+ 5G, an AT&T 5G smartphone

nathan-curtis, 2023-11-16:
> The installing endless in Home Assistant should be solved in the dev branch already (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

I'm talking about THIS...

[image]

Koenkk, 2023-11-16:
In your screenshot it looked that the update started but does not finish, could you provide me the herdsman debug log when trying to update this device?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

markhaines, 2023-11-16:
I just tried flipping to the Edge release (stopped standard release, copied the config yaml, installed edge release, pasted the config and started it) and it seems to have made the two devices I had that have recently been stuck on the bogus update disappear - thanks!

markhaines, 2023-11-16:
...ah, spoke too soon - they're back, but now i have the option to go in and skip the upgrade rather than it trying to auto update and getting stuck (it used to do this previously, not sure why the behaviour changed in last couple of weeks), either way its no bother to just click skip.

nathan-curtis, 2023-11-17:
> In your screenshot it looked that the update started but does not finish, could you provide me the herdsman debug log when trying to update this device?
>

As soon as I can I'm just now recovering from the addon move weirdness of NR16 this week it may be a few days. But someone just above me nailed it. The problem is we're being offered something not there or invalid.

The spinning was just a side effect. Now that you fixed that one we still keep getting offered this invalid Ota.

Koenkk, 2023-11-18:
Can someone provide the debug log when trying to update this device? The OTA is pulled from https://github.com/koenkk/zigbee-ota

nathan-curtis, 2023-11-18:
Sorry got caught up in the changes to addons this week had to repair my install. Will be at least Mon before I can.

highincolorado, 2023-11-18:
Here is the beginning of the update still has many minutes to run:

debug 2023-11-18 14:16:36: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/update' with data '{"id":"library_curtain","transaction":"w98ax-19"}'
info  2023-11-18 14:16:36: Updating 'library_curtain' to latest firmware
info  2023-11-18 14:16:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Updating 'library_curtain' to latest firmware","meta":{"device":"library_curtain","status":"update_in_progress"},"type":"ota_update"}'
debug 2023-11-18 14:16:37: Received Zigbee message from 'library_curtain', type 'readResponse', cluster 'genBasic', data '{"dateCode":"Apr 22 2022","swBuildId":"0.0.0_0024"}' from endpoint 1 with groupID 0
debug 2023-11-18 14:16:37: Updating to latest '0x54ef4410004fe643' (lumi.curtain.agl001)
debug 2023-11-18 14:16:37: Using endpoint '1'
debug 2023-11-18 14:16:38: Received Zigbee message from 'library_curtain', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":24,"imageType":4105,"manufacturerCode":4447}' from endpoint 1 with groupID 0
debug 2023-11-18 14:16:38: Got OTA request '{"fieldControl":0,"manufacturerCode":4447,"imageType":4105,"fileVersion":24}'

highincolorado, 2023-11-18:
And here is the end of the update, If you want the whole 32 minutes worth of logs let me know.

debug 2023-11-18 14:47:48: Request offsets: fileOffset=288190 pageOffset=0 dataSize=48
debug 2023-11-18 14:47:48: Payload offsets: start=288190 end=288213 dataSize=48
debug 2023-11-18 14:47:50: Got upgrade end request for '0x54ef4410004fe643': {"status":0,"manufacturerCode":4447,"imageType":4105,"fileVersion":6168}
debug 2023-11-18 14:47:50: Update succeeded, waiting for device announce
info  2023-11-18 14:47:50: Update of 'library_curtain' at 100.00%
info  2023-11-18 14:47:50: MQTT publish: topic 'zigbee2mqtt/library_curtain', payload '{"action":null,"battery":36,"charging":false,"device_temperature":23,"hand_open":null,"hooks_lock":"UNLOCK","hooks_state":"unlocked","illuminance_lux":0,"linkquality":54,"motor_state":"stopped","position":100,"power_outage_count":3,"power_source":"battery","running":false,"state":"CLOSE","target_position":0,"update":{"installed_version":24,"latest_version":6168,"progress":100,"remaining":16,"state":"updating"},"update_available":false,"voltage":3000}'
info  2023-11-18 14:47:50: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'library_curtain' at 100.00%","meta":{"device":"library_curtain","progress":100,"status":"update_progress"},"type":"ota_update"}'
debug 2023-11-18 14:48:01: Got device announce or timed out, call resolve
debug 2023-11-18 14:48:01: Device 'library_curtain' announced itself
info  2023-11-18 14:48:01: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"library_curtain","ieee_address":"0x54ef4410004fe643"},"type":"device_announce"}'
info  2023-11-18 14:48:01: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"library_curtain"},"type":"device_announced"}'
info  2023-11-18 14:48:01: Finished update of 'library_curtain'
info  2023-11-18 14:48:01: MQTT publish: topic 'zigbee2mqtt/library_curtain', payload '{"action":null,"battery":36,"charging":false,"device_temperature":23,"hand_open":null,"hooks_lock":"UNLOCK","hooks_state":"unlocked","illuminance_lux":0,"linkquality":51,"motor_state":"stopped","position":100,"power_outage_count":3,"power_source":"battery","running":false,"state":"CLOSE","target_position":0,"update":{"installed_version":6168,"latest_version":6168,"state":"idle"},"update_available":false,"voltage":3000}'
debug 2023-11-18 14:48:02: Received Zigbee message from 'library_curtain', type 'readResponse', cluster 'genBasic', data '{"dateCode":"Apr 22 2022","swBuildId":"0.0.0_0024"}' from endpoint 1 with groupID 0
info  2023-11-18 14:48:02: Configuring 'library_curtain'
info  2023-11-18 14:48:02: Device 'library_curtain' was updated from '{"dateCode":"Apr 22 2022","softwareBuildID":"0.0.0_0024"}' to '{"dateCode":"Apr 22 2022","softwareBuildID":"0.0.0_0024"}'

[20 more replies]

closed 2024-01-31 by Koenkk