Koenkk/zigbee2mqtt #15747: TS011F plug cause warnings in HA log

kiekerjan18, 2022-12-25:
### What happened?

Hi,

In the log file of Home Assistant I see the following warnings:

**Home Assistant log**
`2022-12-25 14:17:35.454 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'installed_version' when rendering '{{ value_json['update']['installed_version'] }}'
2022-12-25 14:17:35.458 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'latest_version' when rendering '{{ value_json['update']['latest_version'] }}'
2022-12-25 14:17:37.624 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'installed_version' when rendering '{{ value_json['update']['installed_version'] }}'
2022-12-25 14:17:37.629 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'latest_version' when rendering '{{ value_json['update']['latest_version'] }}'`

I found a pattern with powerplugs TS011F received messages:

TS011F_plug_1 [TS011F_plug_1](https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html#tuya-ts011f_plug_1) 
- _TZ3000_hdopuwv6

TS011F_plug_3 [TS011F_plug_3](https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html#tuya-ts011f_plug_3)
- _TZ3000_cphmq0q7

**Z2M debug log**
`debug 2022-12-25 14:18:35: Received Zigbee message from 'Media', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":229}' from endpoint 1 with groupID 0
info  2022-12-25 14:18:35: MQTT publish: topic 'zigbee2mqtt/Media', payload '{"child_lock":"UNLOCK","current":0,"energy":6.63,"indicator_mode":"off/on","linkquality":105,"power":0,"power_outage_memory":"off","state":"OFF","update":{"state":"idle"},"update_available":null,"voltage":229}'
debug 2022-12-25 14:18:37: Received Zigbee message from 'Verwarmingsmat', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":229}' from endpoint 1 with groupID 0
info  2022-12-25 14:18:37: MQTT publish: topic 'zigbee2mqtt/Verwarmingsmat', payload '{"child_lock":"UNLOCK","current":0,"energy":5.82,"indicator_mode":"off/on","linkquality":94,"power":0,"power_outage_memory":"off","state":"OFF","update":{"state":"idle"},"update_available":null,"voltage":229}'`

After repairing all the devices the warnings were solved. But unfortunately after I rebooted Home Assistant these warnings all returned. 

It appears the ['update'].['latest_version'] and ['update'].['installed_version'] are expected somehow but not given in the message.

### What did you expect to happen?

No warnings in my log anymore

### Zigbee2MQTT version

1.28.4-dev commit: 6d08a2f

### Adapter firmware version

20221102

### Adapter

Sonoff Zigbee Dongle Plus

Gorouk, 2023-01-01:
It is happening on 1.29.0 commit dddce571 with TS011F_plug_1.

`{
    "child_lock": "UNLOCK",
    "current": 0,
    "energy": 0.27,
    "indicator_mode": "off/on",
    "linkquality": 36,
    "power": 0,
    "power_outage_memory": "restore",
    "state": "OFF",
    "update": {
        "installed_version": "unknown",
        "state": "idle"
    },
    "voltage": 238,
    "update_available": null
}`

only for `'{{ value_json['update']['latest_version'] }}'` tho

AbeltjeNL, 2023-01-01:
```Logger: homeassistant.helpers.template
Source: helpers/template.py:589 
First occurred: 12:28:53 (3015 occurrences) 
Last logged: 12:35:59

Template variable warning: 'dict object' has no attribute 'latest_version' when rendering '{{ value_json['update']['latest_version'] }}'```

Koenkk, 2023-01-02:
Should be fixed now in the latest-dev.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

rccoleman, 2023-01-02:
I didn't see any warnings restarting Z2M with that commit:

[image]

balbulator, 2023-01-02:
In a day, I have several thousand such errors. The entire magazine is full. I can't figure out what is causing this error
`Logger: homeassistant.helpers.template
Source: helpers/template.py:589
First occurred: 12:37:24 (46 occurrences)
Last logged: 12:47:52

`Template variable warning: 'dict object' has no attribute 'latest_version' when rendering '{{ value_json['update']['latest_version'] }}`

GerdRuetten, 2023-01-02:
See discussion under:

https://github.com/Koenkk/zigbee2mqtt/discussions/15039#discussioncomment-4570806

The problem isn't solve

balbulator, 2023-01-02:
> Дивіться обговорення під:
> 
> [#15039 (коментар)](https://github.com/Koenkk/zigbee2mqtt/discussions/15039#discussioncomment-4570806)
> 
> Проблема вирішена!

the problem is not solved

balbulator, 2023-01-02:
The problem is solved by disabling the update sensor on the device in Home Assistants

cinghialino, 2023-01-02:
I still have the error even if I disable the update sensor in home "assistants"

GerdRuetten, 2023-01-02:
> The problem is solved by disabling the update sensor on the device in Home Assistants

Thank you!
Nice workaround but i hope, it's not the final resolution.

AbeltjeNL, 2023-01-02:
Seems to do the trick. Thanks!

snp88, 2023-01-02:
same here 
i have only ywo models with OTA update 
[image]

I think both are impacted first i turn off update sensor for TS011F but i still get the issue 
now i turn off sensor for second device 
No issue so far

cinghialino, 2023-01-02:
Quick tip to find the entities: 

In your instance, navigate to `/config/entities`
search for `update.` and then click once on the top column `integration` (so to sort them alphabetically by integration)
then scroll down to the `MQTT` integration, you'll have all the entities ready to be disabled, by selecting them first and then clicking on `Disable Selected`

BoneheadFraggle, 2023-01-03:
Just to make sure I respond correctly:
I run Zigbee2MQTT as a HA Add-on. I stopped the ordinary version, installed Edge version, copied the config and started Edge version. Restarted HA just in case. **No logging issues, everything works fine.** 

Unless there is some problems I don't know about, it was really smooth to alternate between versions.

rccoleman, 2023-01-03:
I just restarted Z2M (container) and then HA and didn't see a single instance of the error that started this issue.  Currently running `1.29.0-dev commit: 15ccf94` and `Home Assistant 2023.1.0b5`.

Looks good to me.

Gorouk, 2023-01-03:
no more warning for me on `1.29.0-dev commit 15ccf945` with `TS011F_plug_1`

but a new problem with SONOFF 01MINIZB on `power_on_behavior` : 
`Publish 'get' 'power_on_behavior' to 'light1' failed: 'Error: Read 0x00124b0024c13809/1 genOnOff(["startUpOnOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
`

rccoleman, 2023-01-03:
Uh oh, maybe related to https://github.com/Koenkk/zigbee-herdsman-converters/pull/5253, which was a generalized version of https://github.com/Koenkk/zigbee-herdsman-converters/pull/5253.

GerdRuetten, 2023-01-03:
Sorry, but with Zigbee2MQTT Version
1.29.0-dev commit: [15ccf94] the problem is NOT solved

After a "Check all" on the OTA-Register from z2m the HA-log 

- install the latest edge-Version from z2m
- configuration and start z2m
- restart HA
- go to z2m -> OTA
- klick "Check all"
- go back to HA Settings -> System -> Logs

```
Logger: homeassistant.helpers.template
Source: helpers/template.py:589
First occurred: 21:43:28 (65 occurrences)
Last logged: 21:44:44

Template variable warning: 'dict object' has no attribute 'latest_version' when rendering '{{ value_json['update']['latest_version'] }}'

```

rccoleman, 2023-01-03:
For what it's worth, I just to the OTA section, did a Check All, and didn't see warnings in the log, so maybe it's related to the devices themselves or something that's baked in during the interview.

Koenkk, 2023-01-04:
@GerdRuetten thanks for your analysis, also fixed this please confirm.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Gorouk, 2023-01-04:
> * install the latest edge-Version from z2m
> * configuration and start z2m
> * restart HA
> * go to z2m -> OTA
> * klick "Check all"
> * go back to HA Settings -> System -> Logs

Following @GerdRuetten description but with 1.29.0-dev commit: [cf801344](https://github.com/Koenkk/zigbee2mqtt/commit/cf801344), i don't have log error with `TS011F_plug_1`, neither spamming, nor after 'Check all' click .

However, there is still the problem about SONOFF 01MINIZB `power_on_behavior` attribute.
if i take a look at its state, I have only this :
```
{
    "linkquality": 21,
    "state": "OFF"
}

Nezz, 2023-01-04:
The errors are gone for me after updating to the edge version.

GerdRuetten, 2023-01-05:
With edge-Version Zigbee2MQTT Version
1.29.0-dev commit: [cf80134](https://github.com/Koenkk/zigbee2mqtt/commit/cf80134) the error seems to have been fixed.

I no longer get any error messages in the log.

A big thank you to @Koenkk .

Great support!

Koenkk, 2023-01-05:
Awesome!

closed 2023-01-05 by Koenkk