Koenkk/zigbee2mqtt #15748: [New device support]: Inara Outdoor wall light

thohoj, 2022-12-25:
### Link

https://www.philips-hue.com/de-de/p/hue-white-inara-outdoor-wandleuchte/8719514388512

### Database entry

{"id":23,"type":"Router","ieeeAddr":"0x001788010b851c10","nwkAddr":52906,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"929003098901","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":257,"inClusterList":[0,3,4,5,6,8,4096,64514],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"929003098901","manufacturerName":"Philips","powerSource":1,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20190425","swBuildId":"1.55.8_r28815"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20190425","swBuildId":"1.55.8_r28815","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1671895556352,"defaultSendRequestWhen":"immediate"}

### Comments

The database entry is originally from an ioBroker zigbee adapter instance. I am not sure, but internally zigbee2mqtt seems to be used. 

I didn't know how to pair the bulb with zigbee2mqtt as described in the add new devices howto without running the risk of breaking my current zigbee environment. I hope the database entry is still sufficient to add the bulb as a new device.

Koenkk, 2022-12-25:
Please add the color temperature range as described on https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

thohoj, 2022-12-25:
I don't think this bulb supports color temperature.

Koenkk, 2022-12-26:
Great, added!

closed 2022-12-26 by Koenkk