Koenkk/zigbee2mqtt #15749: Cant pair my TuYa TS0044 switch

mouledla, 2022-12-25:
### What happened?

According to https://www.zigbee2mqtt.io/devices/TS0044.html the tuya 4gang switch based on zigbee should work with zigbee2mqtt. Unfortunately i cannot pair this device. Other zigbee products pair instantly with my zigbee2mqtt setup. I tried several things to solve this issue, like new battery, restarting zigbee2mqtt etc. 

To enter pairing mode I hold the bottom left button for 10 seconds until all 4 LEDs start flashing. Nothing happens. It keeps on flashing until it just stops flashing. Unfortunately, no debug logs or anything helpful in zigbee2mqtt logs.

Anyone else has this problem? 

### What did you expect to happen?

I expected successfull pairing of the device according to https://www.zigbee2mqtt.io/devices/TS0044.html

### How to reproduce it (minimal and precise)

Hold bottom left button for 10 seconds until all 4 LEDs start flashing.

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20211217

### Adapter

Sonoff Zigbee 3.0 USB Dongle

Silvenga, 2022-12-26:
FWIW, I have several of these - most of the units went smoothly, some took several tries (like a frustrating number of tries, not sure if I was just messing up the pair code).

Now I might be completely miss-remembering, but I was under the impression TuYa devices are wonky (they aren't standard Zigbee, so zigbee2mqtt and other open source coordinators) need a magic packet to be sent during interviewing (which, after re-reading your post, doesn't seem to be happening in your issue, since no logs). Race conditions are likely since timing is important. So there was a report that interviewing through a router was error prone, meshing directly with the coordinator was better due to possible retries over the network.

mouledla, 2022-12-26:
Hmm i actually have no idea about tuya devices being wonky, I simply just go zigbee2mqtt.io website and check if the vendor and specific device is pairable under devices.

marc-gist, 2023-01-06:
I just received a couple of these and can't pair either, updated to latest firmware on coordinator and z2m code via docker, and still can't pair these :(

marc-gist, 2023-01-07:
I was finally able to pair these @mouledla 

1. Ensure you have a good battery.  The ones I first used, couldn't handle the power draw, and voltage would drop and device would restart (used a voltage meter to see this).  So if when in pairing, you don't see all 4 leds flash for a minute at least, find a new battery.
2. Put the device right beside the coordinator, and select enable pairing for coordinator only.
3. If your coordinator is not on a USB ext cord and plugged directly in, remove it and put on an extension cord away from the computer you use.

Let us know if you get them working.

mouledla, 2023-01-07:
@marc-gist I have a strong suspicion it is indeed the battery. Will try a new battery tomorrow and let you know.

mouledla, 2023-01-07:
@marc-gist I tried a couple of new batteries... like 3 of them (CR2430). Somehow, when using 2 of the 3 batteries, pairing did not worked. For some unknown reason the last battery I tried, pairing went successfull. Case closed I guess.

closed 2023-01-07 by mouledla

marc-gist, 2023-01-08:
> 

Cheap batteries do not have the Amperage draw (so measure 3.2v "full", but as soon as you put a load on them, they drop under 2.8v, which causes the device to "brown out").

alexbenfica, 2023-01-10:
Battery was the problem. Just inserted new battery and it was immediately detected by the coordinator.

realalexandergeorgiev, 2023-12-18:
Can confirm, had cheap batteries and the device LEDs would work and indicate pairing start, but it would not pair. Checking with voltmeter showed dropping from 3.2V to 2.5V while pairing. Paired using external power supply @3.1V was without issues. Now ordering Varta batteries.

tclemos, 2023-12-30:
> I was finally able to pair these @mouledla
>
> 1. Ensure you have a good battery.  The ones I first used, couldn't handle the power draw, and voltage would drop and device would restart (used a voltage [...]

My device is a different model from the TS0044, but it is identified as a TS0044 when paired, this is my device:
[image]

This device has an internal reset button:
[image]

I was facing pairing issues until I placed the device closer to the coordinator and selected to pair only with the coordinator, 
[image]

it worked like a charm.

MacArtiom, 2024-10-14:
[image]
[image]

Why by default my 4 switch is not working as 4 key switch, can anyone help or describe what I do wrong?

quadrupole1988, 2024-10-28:
> ![Screenshot_1](https://private-user-images.githubusercontent.com/96257712/376288696-6701fffa-7f4a-443f-b27b-afec4c607d3d.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAxMDUzMTcsIm5iZiI6MTczMDEwNTAxNywicGF0aCI6Ii85NjI1NzcxMi8zNzYyODg2OTYtNjcwMWZmZmEtN2Y0YS00NDNmLWIyN2ItYWZlYzRjNjA3ZDNkLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDI4VDA4NDMzN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTViY2ZhODk1ZTlkMjQ4MjIyMThjYzhhMDU1YWE3YzJjMmIyYTRjYjE4YjYzMmUxNDllODM0ZTZkNzY4MTgyMDcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.CgoDyip7hdXck7rKF-BWl76J8glbs_Z1HS_-IHufje4) ![Screenshot_2](https://private-user-images.githubusercontent.com/96257712/376288710-6516994b-f991-4007-9618-326e717a79d6.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAxMDUzMTcsIm5iZiI6MTczMDEwNTAxNywicGF0aCI6Ii85NjI1NzcxMi8zNzYyODg3MTAtNjUxNjk5NGItZjk5MS00MDA3LTk2MTgtMzI2ZTcxN2E3OWQ2LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDI4VDA4NDMzN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE3ZDQ4MTkwYjBjZWJmYmFhMWZiOWM1NzAzNWI5NGQ5YWJmMmQzMjM3YmI2NjFlYTg0Y2YzZTJkYzlkNjU5ODgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.nYUpQojISzE8Ir2JFpCXaGkeh7S1OBL1-lNDjwi0OV8)
> 
> Why by default my 4 switch is not working as 4 key switch, can anyone help or describe what I do wrong?

Did you ever solve this? I just received the switch and I'm facing the same issue....

Whenever I click a button, I can see the link quality update but the action triggered just shows Empty String("")

XGCTHX9, 2025-07-11:
mine came today and i tried all day, 2mqtt saying its not supported yet, now after 10 hours i tried pairing again and its supported now but i cant see / press anything beside seeing battery life ? 

[image]

Siox0911, 2025-07-14:
Your post is out of scope:

> mine came today and i tried all day, 2mqtt saying its not supported yet, now after 10 hours i tried pairing again and its supported now but i cant see / press anything beside seeing battery life ?
> 
>

try this: [Zigbee2MQTT - Tuya 4-button Scene Switch (TS0044)](https://community.home-assistant.io/t/zigbee2mqtt-tuya-4-button-scene-switch-ts0044-rdeangel/663302)