Koenkk/zigbee2mqtt #15755: Zigbee2MQTT crash several times a day

jlpower68, 2022-12-26:
### What happened?

ZigbeeToMQTT crashes randomly 1 to 10 times a day.
After restarting the docker container, it works fine again for few hours.

I've attached the last log file, where last crashes appeared at  :
2022-12-26 02:09:41
2022-12-26 06:44:55

Txs

### Zigbee2MQTT version

 1.28.4 commit: 52e545f (also tested with 1.28.4-dev commit: 05cb140)

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus V2 model "ZBDongle-E"

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10303024/log.txt)

t112013, 2022-12-27:
I have the same issue. 
I regret so much i moved to this dondle :( 
No remotes work, IKEA remote drain battery,

tjarmann, 2022-12-29:
Same problem here, seems to be related to the ZBDongle-E.
I had a Conbee II and used groups and touchlink with remotes, but after changing to ZBD-E this does not work.

itchannel, 2022-12-29:
I also get the same issue since moving to the ZBDongle-E. I've tried both the docker version and the linux version master & Dev. But both result in random crashes 2-3 times a day. I've attached a full log so you can see where it's crashing, it's neary 24hrs worth but the crash appears when I start seeing the following errors:

```
Error while parsing to ZpiObject 'RangeError: Attempt to access memory outside buffer bounds
```

Shortly followed up by 
```
  [38;5;26;1mzigbee-herdsman:adapter:ezsp:uart [0mOpening SerialPort with /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220714103031-if00 and {"baudRate":115200,"rtscts":false,"autoOpen":false} [38;5;26m+6s[0m
  [38;5;42;1mzigbee-herdsman:adapter:ezsp:erro [0mReset error Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
  [38;5;42;1mzigbee-herdsman:adapter:ezsp:erro [0m    at SerialPort.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:90:28)
  [38;5;42;1mzigbee-herdsman:adapter:ezsp:erro [0m    at SerialPort._error (/opt/zigbee2mqtt/node_modules/@serialport/stream/lib/index.js:198:14)
  [38;5;42;1mzigbee-herdsman:adapter:ezsp:erro [0m    at /opt/zigbee2mqtt/node_modules/@serialport/stream/lib/index.js:242:12 [38;5;42m+6s[0m
  [38;5;163;1mzigbee-herdsman:adapter:ezsp:driv [0mPause 60sec before try 141 [38;5;163m+1s[0m
  [38;5;163;1mzigbee-herdsman:adapter:ezsp:driv [0mReset connection. Try 141 [38;5;163m+54s[0m
  [38;5;163;1mzigbee-herdsman:adapter:ezsp:driv [0mStop driver [38;5;163m+1ms[0m
  [38;5;178;1mzigbee-herdsman:adapter:ezsp:ezsp [0mStop ezsp [38;5;178m+55s[0m
  [38;5;178;1mzigbee-herdsman:adapter:ezsp:ezsp [0mClose ezsp [38;5;178m+0ms[0m
  [38;5;163;1mzigbee-herdsman:adapter:ezsp:driv [0mClose driver [38;5;163m+0ms[0m
```

This is on a fresh Raspbian build with nothing else using the adapter. 

[herdlog.txt](https://github.com/Koenkk/zigbee2mqtt/files/10322337/herdlog.txt)

Any help is greatly appreciated thanks.

panhans, 2022-12-30:
Same issue here. For me I see some devices leave network on disconnect, too. Have an automation if a divice is unknown z2m gets restarted.

t112013, 2022-12-30:
> Same issue here. For me I see some devices leave network on disconnect, too. Have an automation if a divice is unknown z2m gets restarted.

Oh could you share your automatation please ?

jlpower68, 2022-12-31:
Fyi, I have replaced my ZBDongle-E by a ZBDongle-P yesterday evening (thanks Amazon and the great return policy :-), and it still works this morning without any auto restart of the docker container.
Of course, it is not so much intelectually satisfaying than fixing the problem with this experimentally supported ZBDongle-E, but sometimes, invest 30 euros and save a lot of hours is wise :-).

Koenkk, 2022-12-31:
Switching to the ZBDongle-P dongle should indeed fix these issues. As states in the docs (https://www.zigbee2mqtt.io/guide/adapters/) support for the ZBDongle-E is still **experimental**

t112013, 2022-12-31:
Is it anyhow a way to not lose the pairing ?  or do i have to pair all the 60 devices i have ? 
What is to loose ? Zigbee 3.0 support ? why is even the E model better than the P ? 
What other dongle could i use ? what is the "best" recommended dongle ?

jlpower68, 2022-12-31:
Here is the list of recommended adapters:
https://www.zigbee2mqtt.io/guide/adapters/
It also explain the difference between the 2 families, due to the kind of
chipset. Initially I've chosen EFRxxxx because their are supposed to be
opened to Matter migration.
At the beginning, there is an explanation on how to migrate radio for 1
adapter to other, to skip redo pairing.
I don't know if it works when you change of chipset. I have less than 10
devices, so I repaired them.

andy-81, 2023-01-02:
I have had this exact issue, so purchased the zbdongle-p version to solve the problem.

on trying to switch I managed to mess up the setup and ended up having to recover to the backup I made of Z2m and put my old dongle back in temporarily to get back to a working state. I say temporarily but have since been running with it for the last couple of days, for some reason after repairing my devices everything seems stable again. Not sure what has changed but I haven’t had the system stop responding on me since Friday when I restored everything to my backup and reconfigured everything. 

If it helps solve anything I am running Z2m in HA supervisor (hence the issue with downgrading to the earlier version) and it all runs on an Rpi4 so I made a backup of the USB drive before starting the switch to the new dongle. After backing up the drive I tried to switch to the new/older zbdongle-p which showed up, however I had issues getting it to pair with any of the devices, it would enter pairing but then the front end never showed it connecting to the devices. So I had a go at using ZHA in HA which also failed even after resetting the Dongle. I did the same with the original zbdongle-e which reset the older dongle to factory settings (hence why I had to repair after restoring). I am guessing this is what may have solved the issue for me, after restoring the HA from the backup I had made beforehand, I decided to get back to a good working state and so put the original zbdongle-e into the setup and repaired the devices to make sure it was working, since then it has been stable. My issues started happening after Z2M upgraded from 1.27 to 1.28.

not useful to those with a lot of zigbee devices but hopefully it may point someone to where the issue is, to me it seems like something to do with zbdongle-e setup under an older version of Z2m before upgrading and maybe it puts something out of line. I’m not as technical as I used to be but hopefully that makes sense to someone. 

For now, I am sticking with the e version of the dongle after my failed attempt to switch to the p version, maybe I will look to switch again of my setup goes unstable again.

panhans, 2023-01-03:
The latest version (1.29.0-1) seems to run stable with the dongle-e. The issue didn't appear last two days since I updated. I think I will stick with that version until the newer dongle leave experimental state. I don't want to downgrade. The only option is the skyConnect dongle but that has the same issues. Just coming to z2m because of the bad trv support of zha.

xm4rcell0x, 2023-01-07:
> The latest version (1.29.0-1) seems to run stable with the dongle-e. The issue didn't appear last two days since I've updated. I think I will stick with that version until the newer dongle leave experimental state. I don't want to downgrade the hardware. The only option is the skyConnect dongle but that has the same issues with z2m. Just coming to z2m because of the bad trv support of zha.

Any news? Still working?

andy-81, 2023-01-09:
> Any news? Still working?

My crashes have come back, not as often but they do occur. 

I have created a sensor to detect the Z2M going offline and restart the server for me. It isn't ideal but it at least means my network stays running. 

I am sure there is a better way to do it, but I have set up a simple sensor which looks at the last_seen entity for one of my switches I always have on (collecting the power usage of our freezer) and if it hasn't checked in for 90 seconds I have a script which then restarts Z2M for me.

t112013, 2023-01-09:
> My crashes have come back, not as often but they do occur.
>
> I have created a sensor to detect the Z2M going offline and restart the server for me. It isn't ideal but it at least means my network [...]

Hello. could you share how do you made the sensor ? and eventually how yout automation looks like ? 
I have the same issue. on the last update it was fine for a few days and it crashed yesterday, for now its been a few times only.

andy-81, 2023-01-12:
> Hello. could you share how do you made the sensor ? and eventually how yout automation looks like ? I have the same issue. on the last update it was fine for a few days and it crashed yesterday, for [...]

My sensor in the config looks like this:
binary_sensor:
  - platform: template
    sensors: 
      z2m_health:
        friendly_name: 'Z2M Mesh health'
        value_template: >-
          {% if (as_timestamp(now()) - as_timestamp(states.sensor.front_room_freezer_last_seen.state)) > 90 %}Off
          {%- else -%}
            On
          {%- endif-%}

if you already have a binary_sensor section then place the code under it and remove the top line.

this will enable an on/off switch in HA under entities you can use in the automation. I found someone else doing similar but couldn’t get their way to work so ended up doing it this way. If you have a sensor which is always online and reports in at least once every minute then use that device, in my case it is a power socket which I use for metering my freezer (hence the name)

I also went into the device in HA and enabled the last seen option, if you don’t have the option you’ll need to enable it in Z2M as well, if you struggle ping me back and I’ll put a full documentation together. 

This is what the sensor looks like if working correctly
[image]

Once you have that sensor set up the automation was pretty straight forward. I created it in the GUI but you can also use the below yaml code which is a copy of what mine created:
alias: Z2M monitor
description: ""
trigger:
  - platform: state
    entity_id:
      - binary_sensor.z2m_health
    from: "on"
condition: []
action:
  - service: notify.mobile_app_ibone
    data:
      message: Z2m went offline
      title: Restarting
  - service: hassio.addon_restart
    data:
      addon: 45df7312_zigbee2mqtt
mode: single

The screen shots for the GUI are:
[image]

[image]

as you can see I also have an alert which goes to my iPhone (name is based on Cockney rhyming slang for phone) so I know the automation ran.

hope that helps, I am sure there are people on here who could come up with a better way of doing it as I am not a programmer but I found the above worked for me and has kept my network running with minimal downtime. 

It is worth remembering to turn the automation off when you upgrade Z2m, it doesn’t break it but will cause it to restart a second time after the update.

closed 2023-02-19 by github-actions[bot]

eatoff8, 2023-08-28:
> My sensor in the config looks like this: binary_sensor:
>
> * platform: template
>   sensors:
>   z2m_health:
>   friendly_name: 'Z2M Mesh health'
>   value_template: >-
>   {% if (as_timestamp(now()) - [...]
Can you screenshot your sensor template? Mine isnt working...