Koenkk/zigbee2mqtt #15756: [New device support]: _TZ3000_b28wrpvx BSEED Socket Power Monitoring

Ricox1975, 2022-12-26:
### Link

https://de.aliexpress.com/item/1005003595581660.html?spm=a2g0o.store_pc_groupList.8148356.3.615c722ae95liK&pdp_npi=2%40dis%21EUR%21%E2%82%AC%2028%2C02%21%E2%82%AC%2019%2C61%21%21%21%21%21%40210318d116720536385757402eb468%2112000026425315551%21sh

### Database entry

--->>> fromZ2M -> 0xa4c1389b5dcff8ea exposes: {"date_code":"","definition":{"description":"Smart plug (with power monitoring)","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":7,"description":"Recover state after power outage","name":"power_outage_memory","property":"power_outage_memory","type":"enum","values":["on","off","restore"]},{"access":7,"description":"LED indicator mode","name":"indicator_mode","property":"indicator_mode","type":"enum","values":["off","off/on","on/off","on"]},{"access":1,"description":"Instantaneous measured power","name":"power","property":"power","type":"numeric","unit":"W"},{"access":1,"description":"Instantaneous measured electrical current","name":"current","property":"current","type":"numeric","unit":"A"},{"access":1,"description":"Measured electrical potential value","name":"voltage","property":"voltage","type":"numeric","unit":"V"},{"access":1,"description":"Sum of consumed energy","name":"energy","property":"energy","type":"numeric","unit":"kWh"},{"features":[{"access":3,"description":"Enables/disables physical input on the device","name":"state","property":"child_lock","type":"binary","value_off":"UNLOCK","value_on":"LOCK"}],"type":"lock"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS011F_plug_1","options":[{"access":2,"description":"Calibrates the power value (percentual offset), takes into effect on next report of device.","name":"power_calibration","property":"power_calibration","type":"numeric"},{"access":2,"description":"Calibrates the current value (percentual offset), takes into effect on next report of device.","name":"current_calibration","property":"current_calibration","type":"numeric"},{"access":2,"description":"Calibrates the voltage value (percentual offset), takes into effect on next report of device.","name":"voltage_calibration","property":"voltage_calibration","type":"numeric"}],"supports_ota":true,"vendor":"TuYa"},"endpoints":{"1":{"bindings":[{"cluster":"genOnOff","target":{"endpoint":1,"ieee_address":"0x00124b0022811b60","type":"endpoint"}},{"cluster":"haElectricalMeasurement","target":{"endpoint":1,"ieee_address":"0x00124b0022811b60","type":"endpoint"}},{"cluster":"seMetering","target":{"endpoint":1,"ieee_address":"0x00124b0022811b60","type":"endpoint"}}],"clusters":{"input":["genIdentify","genGroups","genScenes","genOnOff","seMetering","haElectricalMeasurement","57344","manuSpecificTuya_3","genBasic"],"output":["genOta","genTime"]},"configured_reportings":[{"attribute":"rmsVoltage","cluster":"haElectricalMeasurement","maximum_report_interval":3600,"minimum_report_interval":5,"reportable_change":5},{"attribute":"rmsCurrent","cluster":"haElectricalMeasurement","maximum_report_interval":3600,"minimum_report_interval":5,"reportable_change":50},{"attribute":"activePower","cluster":"haElectricalMeasurement","maximum_report_interval":3600,"minimum_report_interval":5,"reportable_change":10},{"attribute":"currentSummDelivered","cluster":"seMetering","maximum_report_interval":3600,"minimum_report_interval":5,"reportable_change":[1,1]}],"scenes":[]},"242":{"bindings":[],"clusters":{"input":[],"output":["greenPower"]},"configured_reportings":[],"scenes":[]}},"friendly_name":"___blink_ 0xa4c1389b5dcff8ea","ieee_address":"0xa4c1389b5dcff8ea","interview_completed":true,"interviewing":false,"manufacturer":"_TZ3000_b28wrpvx","model_id":"TS011F","network_address":19760,"power_source":"Mains (single phase)","supported":true,"type":"Router"}

### Comments

Please add this Socket

### External converter

```shell
"ids": {
    "modelId": "TS011F",
    "manufacturerName": "_TZ3000_b28wrpvx"
  },
  "endpoints": {
    "endpointDescriptors": [
      {
        "endpointId": 1,
        "applicationProfileId": 260,
        "applicationDeviceId": 266,
        "applicationDeviceVersion": 0,
        "_reserved1": 1,
        "inputClusters": [
          3,
          4,
          5,
          6,
          1794,
          2820,
          57344,
          57345,
          0
        ],
        "outputClusters": [
          25,
          10
        ]
      },
      {
        "endpointId": 242,
        "applicationProfileId": 41440,
        "applicationDeviceId": 97,
        "applicationDeviceVersion": 0,
        "_reserved1": 0,
        "inputClusters": [],
        "outputClusters": [
          33
        ]
      }
    ],
    "endpoints": {
      "1": {
        "clusters": {
          "identify": {
            "attributes": []
          },
          "groups": {
            "attributes": [
              {
                "acl": [
                  "readable",
                  "reportable"
                ],
                "id": 0,
                "name": "nameSupport",
[... 520 lines cut ...]
                "value": 3
              }
            ]
          },
          "time": {
            "attributes": [
              {
                "acl": [
                  "readable"
                ],
                "id": 65533,
                "name": "clusterRevision",
                "value": 1
              }
            ]
          }
        }
      },
      "242": {
        "clusters": {},
        "bindings": {}
      }
    }
  }
```

Koenkk, 2022-12-26:
This device should work out of the box with z2m 1.28.4

Ricox1975, 2022-12-27:
Only briefly, the device can be taught and also switched for 2-5. After about 5 minutes it no longer reacts at all. If I briefly disconnect it from the power supply, it flashes red again and can be switched on for about 4 minutes. As soon as the flashing stops, I can no longer operate it. Sorry for my english, my german is much better ;-)
[image]

Koenkk, 2022-12-27:
This seems to be a problem in your zigbee network or coordinator. What coordinator/fw version are you using?

Ricox1975, 2022-12-27:
configuration.yaml:
`homeassistant: false
permit_join: true
external_converters:
  - lidlnew.js
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.1.51:1885
  keepalive: 60
  password: ‚xxxxx‘
  reject_unauthorized: true
  user: foo
  version: 4
serial:
  port: tcp://192.168.1.201:6638
  disable_led: true
frontend: {}
advanced:
  pan_id: 7000
  ext_pan_id:
  xxxxxxxxxxxxxxxxxx
  channel: 11
  xxxxxxxxxxxxxxxxxx
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
  output: json
  transmit_power: 20
  log_level: debug
device_options:
  legacy: false
availability:
  active:
    timeout: 20
  passive:
    timeout: 2000`
[image]

Koenkk, 2022-12-28:
Can you try the following fw: https://github.com/Koenkk/Z-Stack-firmware/tree/6.10.01.01/coordinator/Z-Stack_3.x.0/bin ?

Ricox1975, 2022-12-28:
It seems to be working now, great

closed 2022-12-30 by Ricox1975