Koenkk/zigbee2mqtt #15758: [New device support]: Develco ZHWR202

kristiankynk, 2022-12-26:
### Link

https://heimgard.com/support/brukerveiledninger/vare-eldre-produkter/home-control-smart-plug-develco-zhwr202

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0x0015bc001d020a19","nwkAddr":28066,"manufId":4117,"manufName":"Develco Products A/S","powerSource":"Mains (single phase)","modelId":"ZigBee Wallplug HA SP2","epList":[33,1],"endpoints":{"1":{"profId":49353,"epId":1,"devId":1,"inClusterList":[2,3],"outClusterList":[1],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"33":{"profId":260,"epId":33,"devId":9,"inClusterList":[0,3,4,5,6,1794,9],"outClusterList":[10,1030],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}}},"zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1672055188973,"defaultSendRequestWhen":"immediate"}

### Comments

Too noob to do that

ironboy84, 2023-04-26:
> ### Link
> https://heimgard.com/support/brukerveiledninger/vare-eldre-produkter/home-control-smart-plug-develco-zhwr202
>
> ### Database entry [...]

I also need support for this item :-) Anyone that can help?

Koenkk, 2023-04-27:
Could you check if the issue is fixed with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    zigbeeModel: ['ZigBee Wallplug HA SP2'],
    model: 'ZHWR202',
    vendor: 'Develco',
    description: 'Smart plug',
    extend: extend.switch(),
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint);
    },
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

ironboy84, 2023-05-01:
Failed to configure '0x0015bc001d023394', attempt 1 (Error: ConfigureReporting 0x0015bc001d023394/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 18666 - 1 - 18 - 6 - 7 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:559:17) at processTimers (node:internal/timers:502:7))

sjorge, 2023-05-01:
> Could you check if the issue is fixed with the following external converter:
>
>
>
>
>
> ```js
>
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
>
> const tz = [...]

You probably also want to read the sw version in the configure like we do for a lot of the other develco devices.

ironboy84, 2023-05-02:
Sorry, i am a noob. How do i do that, read the sw version? :-) Thanks!

sjorge, 2023-05-02:
That was more a reply to @Koenkk specifically it's just an extra line to the convertor he posted

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['ZigBee Wallplug HA SP2'],
    model: 'ZHWR202',
    vendor: 'Develco',
    description: 'Smart plug',
    extend: extend.switch(),
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(33);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint);
        await develco.configure.read_sw_hw_version(device, logger);
    },
};

module.exports = definition;
```

It's not magically gonna make the timeout go away, knowing develco the main endpoint is usually not 1 so I did update it above to be 33, which seems to be the first one in the 30+ range which is usually where develco puts their stuff. So try this one I guess.

kristiankynk, 2023-05-06:
Failed to configure 'Develco soc_1', attempt 2 (Error: Bind 0x0015bc001d020a19/33 genOnOff from '0xe0798dfffe3bd747/1' failed ({"address":28066,"clusterId":32801,"sequence":8} after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:559:17) at processTimers (node:internal/timers:502:7))

kristiankynk, 2023-06-06:
Any ideas:

Failed to configure 'Develco soc_1', attempt 2 (Error: Bind 0x0015bc001d020a19/33 genOnOff from '0xe0798dfffe3bd747/1' failed ({"address":28066,"clusterId":32801,"sequence":8} after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:559:17) at processTimers (node:internal/timers:502:7))

closed 2023-07-16 by github-actions[bot]

Koenkk, 2024-01-13:
Can anyone try with the new [generated external definition feature](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-creating-the-external-definition)?