Koenkk/zigbee2mqtt #15761: [Feature request]: Add additional events for Phillips Hue Tap Dial (e.g. number of steps you rotate the dial by)

victorhooi, 2022-12-26:
### Is your feature request related to a problem? Please describe

I'm trying to port this blueprint from ZHA to zigbee2mqtt.

[https://community.home-assistant.io/t/philips-hue-tap-dial-switch-zha/446636/](https://community.home-assistant.io/t/philips-hue-tap-dial-switch-zha/446636/)
https://github.com/apollo1220/blueprints/blob/main/philips_zigbee_dial.yaml

The blueprint allows you to use the Phillips Hue Tap Dial with the following functionality:
* Pressing buttons 1 through to 4 lets you toggle 4 different lights/devices on/off
* Turning the tap dial lets you dim the last toggled light (and there's a configurable scaling factor).

However, it seems like certain events are not exposed in zigbee2mqtt (or possibly I'm missing something very obvious), which is hampering my ability to port it.

### Describe the solution you'd like

The ZHA blueprint seems to pull things like `scene_id`, as well as `step_mode` (which I believe is either slow or fast) and `step_size`. This is the variables stanza:

```
action:
  - variables:
      command: "{{ trigger.event.data.command }}"
      args: "{{ trigger.event.data.args }}"
      params: "{{ trigger.event.data.params }}"
      scene: "{{ trigger.event.data.params.scene_id }}"
      step_mode: "{{ trigger.event.data.params.step_mode }}"
      step_size: "{{ trigger.event.data.params.step_size }}"
```

For zigbee2mqtt, it creates an entity for the tap dial called for example `sensor.hue_tap_dial_1_action` - and when you press buttons, it does things like `state: button_1_press` and `state: button_1_press_release`.

For turning the dial, it just does `state: dial_rotate_left_slow` or `state: dial_rotate_right_fast` when you turn it.

However, there doesn't seem to be any way to pull the number of steps you click on the dial.

### Describe alternatives you've considered

Use ZHA instead of zigbee2mqtt (but I'm already setup with zigbee2mqtt, and I like it, lol).

### Additional context

My Phillips Hue Tap DIal is on the latest firmware (2.59.25), if that helps. Updated by zigbee2mqtt, which was pretty neat.a

closed 2023-02-02 by github-actions[bot]

nareddyt, 2023-02-11:
Hi @victorhooi , how did you get the `sensor.hue_tap_dial_1_action` entities to show up in Home Assistant?

After pairing my tap dial via Zigbee2MQTT, I only get the `sensor.hue_tap_dial_battery` and `update.hue_tap_dial` entities. I don't see an `_action` entity. Was there anything special you had to do?

nareddyt, 2023-02-11:
BTW you don't need to rewrite the ZHA blueprint, there is already another blueprint for Z2M.

https://community.home-assistant.io/t/philips-hue-tap-dial-switch-zigbee2mqtt/461824

jafrouk, 2023-07-02:
I was looking at the same thing today and I notice that the zigbee messages are being picked up by zigbee2mqtt, but are not being converted:

`Received Zigbee message from 'Sitting TapDial 1', type 'commandStepWithOnOff', cluster 'genLevelCtrl', data '{"stepmode":0,"stepsize":87,"transtime":5}' from endpoint 1 with groupID 0
No converter available for '8719514440937/8719514440999' with cluster 'genLevelCtrl' and type 'commandStepWithOnOff' and data '{"stepmode":0,"stepsize":87,"transtime":5}'`

I suspect it is fairly easy to add these into the converter but I am having trouble figuring out how to do that, so thought I'd share here in case anyone can direct me.

My hope is we can get this added and can use the stepsize/transtime to build smoother dimming, with less zigbee messages (currently my automations are firing for every step, slow and fast and sometimes the zigbee network gets flooded)

joeblack2k, 2023-07-30:
@nareddyt did you ever get it to work? i tried to add it trough z2m but i can't see action on home assistant

jafrouk, 2023-09-26:
I have "legacy triggers" enabled, but I don't se stepsize or stepmode coming through on the events. I listened to all events (*), turned the dial, and the only event I see is state_changed, which doesn't include this. Am I missing something?