Koenkk/zigbee2mqtt #15771: Philips 929003017102  action empty string

TheIronRock95, 2022-12-27:
### What happened?

Philips 929003017102, action object is empty. When you use the wall switch is showing action voor 1 second and goes back to empty string. 

[image]

> {
    "action": "",
    "battery": 100,
    "device": {
        "applicationVersion": 9,
        "dateCode": "20210504",
        "friendlyName": "Philips 929003017102",
        "hardwareVersion": 3,
        "ieeeAddr": "0x001788010cc38f82",
        "manufacturerID": 4107,
        "manufacturerName": "Signify Netherlands B.V.",
        "model": "929003017102",
        "networkAddress": 17991,
        "powerSource": "Battery",
        "softwareBuildID": "1.0.5",
        "stackVersion": 2,
        "type": "EndDevice",
        "zclVersion": 2
    },
    "device_mode": "single_rocker",
    "last_seen": "2022-12-27T15:39:36.837Z",
    "linkquality": 255
}

### What did you expect to happen?

action holding status "left_press_release" or "left_press"

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

0x26780700

### Adapter

deconz

### Debug log

0
Zigbee2MQTT:info  2022-12-27 16:38:58: MQTT publish: topic 'zigbee2mqtt/Philips 929003017102', payload '{"action":"left_press","battery":100,"device":{"applicationVersion":9,"dateCode":"20210504","friendlyName":"Philips 929003017102","hardwareVersion":3,"ieeeAddr":"0x001788010cc38f82","manufacturerID":4107,"manufacturerName":"Signify Netherlands B.V.","model":"929003017102","networkAddress":17991,"powerSource":"Battery","softwareBuildID":"1.0.5","stackVersion":2,"type":"EndDevice","zclVersion":2},"device_mode":"single_rocker","last_seen":"2022-12-27T15:38:58.091Z","linkquality":255}'
29
Zigbee2MQTT:info  2022-12-27 16:38:58: MQTT publish: topic 'zigbee2mqtt/Philips 929003017102', payload '{"action":"","battery":100,"device":{"applicationVersion":9,"dateCode":"20210504","friendlyName":"Philips 929003017102","hardwareVersion":3,"ieeeAddr":"0x001788010cc38f82","manufacturerID":4107,"manufacturerName":"Signify Netherlands B.V.","model":"929003017102","networkAddress":17991,"powerSource":"Battery","softwareBuildID":"1.0.5","stackVersion":2,"type":"EndDevice","zclVersion":2},"device_mode":"single_rocker","last_seen":"2022-12-27T15:38:58.091Z","linkquality":255}'
28
Zigbee2MQTT:info  2022-12-27 16:38:58: MQTT publish: topic 'zigbee2mqtt/Philips 929003017102/action', payload 'left_press'
27
Zigbee2MQTT:info  2022-12-27 16:38:58: MQTT publish: topic 'zigbee2mqtt/Philips 929003017102', payload '{"action":"left_press_release","battery":100,"device":{"applicationVersion":9,"dateCode":"20210504","friendlyName":"Philips 929003017102","hardwareVersion":3,"ieeeAddr":"0x001788010cc38f82","manufacturerID":4107,"manufacturerName":"Signify Netherlands B.V.","model":"929003017102","networkAddress":17991,"powerSource":"Battery","softwareBuildID":"1.0.5","stackVersion":2,"type":"EndDevice","zclVersion":2},"device_mode":"single_rocker","last_seen":"2022-12-27T15:38:58.197Z","linkquality":255}'
26
Zigbee2MQTT:info  2022-12-27 16:38:58: MQTT publish: topic 'zigbee2mqtt/Philips 929003017102', payload '{"action":"","battery":100,"device":{"applicationVersion":9,"dateCode":"20210504","friendlyName":"Philips 929003017102","hardwareVersion":3,"ieeeAddr":"0x001788010cc38f82","manufacturerID":4107,"manufacturerName":"Signify Netherlands B.V.","model":"929003017102","networkAddress":17991,"powerSource":"Battery","softwareBuildID":"1.0.5","stackVersion":2,"type":"EndDevice","zclVersion":2},"device_mode":"single_rocker","last_seen":"2022-12-27T15:38:58.197Z","linkquality":255}'
25
Zigbee2MQTT:info  2022-12-27 16:38:58: MQTT publish: topic 'zigbee2mqtt/Philips 929003017102/action', payload 'left_press_release'
24
Zigbee2MQTT:info  2022-12-27 16:38:58: MQTT publish: topic 'zigbee2mqtt/Philips 929003017102', payload '{"action":"left_press","battery":100,"device":{"applicationVersion":9,"dateCode":"20210504","friendlyName":"Philips 929003017102","hardwareVersion":3,"ieeeAddr":"0x001788010cc38f82","manufacturerID":4107,"manufacturerName":"Signify Netherlands B.V.","model":"929003017102","networkAddress":17991,"powerSource":"Battery","softwareBuildID":"1.0.5","stackVersion":2,"type":"EndDevice","zclVersion":2},"device_mode":"single_rocker","last_seen":"2022-12-27T15:38:58.831Z","linkquality":255}'
23
Zigbee2MQTT:info  2022-12-27 16:38:58: MQTT publish: topic 'zigbee2mqtt/Philips 929003017102', payload '{"action":"","battery":100,"device":{"applicationVersion":9,"dateCode":"20210504","friendlyName":"Philips 929003017102","hardwareVersion":3,"ieeeAddr":"0x001788010cc38f82","manufacturerID":4107,"manufacturerName":"Signify Netherlands B.V.","model":"929003017102","networkAddress":17991,"powerSource":"Battery","softwareBuildID":"1.0.5","stackVersion":2,"type":"EndDevice","zclVersion":2},"device_mode":"single_rocker","last_seen":"2022-12-27T15:38:58.831Z","linkquality":255}'
22
Zigbee2MQTT:info  2022-12-27 16:38:58: MQTT publish: topic 'zigbee2mqtt/Philips 929003017102/action', payload 'left_press'
21
Zigbee2MQTT:info  2022-12-27 16:38:58: MQTT publish: topic 'zigbee2mqtt/Philips 929003017102', payload '{"action":"left_press_release","battery":100,"device":{"applicationVersion":9,"dateCode":"20210504","friendlyName":"Philips 929003017102","hardwareVersion":3,"ieeeAddr":"0x001788010cc38f82","manufacturerID":4107,"manufacturerName":"Signify Netherlands B.V.","model":"929003017102","networkAddress":17991,"powerSource":"Battery","softwareBuildID":"1.0.5","stackVersion":2,"type":"EndDevice","zclVersion":2},"device_mode":"single_rocker","last_seen":"2022-12-27T15:38:58.934Z","linkquality":220}'
20
Zigbee2MQTT:info  2022-12-27 16:38:58: MQTT publish: topic 'zigbee2mqtt/Philips 929003017102', payload '{"action":"","battery":100,"device":{"applicationVersion":9,"dateCode":"20210504","friendlyName":"Philips 929003017102","hardwareVersion":3,"ieeeAddr":"0x001788010cc38f82","manufacturerID":4107,"manufacturerName":"Signify Netherlands B.V.","model":"929003017102","networkAddress":17991,"powerSource":"Battery","softwareBuildID":"1.0.5","stackVersion":2,"type":"EndDevice","zclVersion":2},"device_mode":"single_rocker","last_seen":"2022-12-27T15:38:58.934Z","linkquality":220}'
[... 13 lines cut ...]
Zigbee2MQTT:info  2022-12-27 16:39:28: MQTT publish: topic 'zigbee2mqtt/Philips 929003017102/action', payload 'left_press_release'
12
Zigbee2MQTT:info  2022-12-27 16:39:29: MQTT publish: topic 'zigbee2mqtt/Philips 929003017102', payload '{"action":"left_press","battery":100,"device":{"applicationVersion":9,"dateCode":"20210504","friendlyName":"Philips 929003017102","hardwareVersion":3,"ieeeAddr":"0x001788010cc38f82","manufacturerID":4107,"manufacturerName":"Signify Netherlands B.V.","model":"929003017102","networkAddress":17991,"powerSource":"Battery","softwareBuildID":"1.0.5","stackVersion":2,"type":"EndDevice","zclVersion":2},"device_mode":"single_rocker","last_seen":"2022-12-27T15:39:29.758Z","linkquality":255}'
11
Zigbee2MQTT:info  2022-12-27 16:39:29: MQTT publish: topic 'zigbee2mqtt/Philips 929003017102', payload '{"action":"","battery":100,"device":{"applicationVersion":9,"dateCode":"20210504","friendlyName":"Philips 929003017102","hardwareVersion":3,"ieeeAddr":"0x001788010cc38f82","manufacturerID":4107,"manufacturerName":"Signify Netherlands B.V.","model":"929003017102","networkAddress":17991,"powerSource":"Battery","softwareBuildID":"1.0.5","stackVersion":2,"type":"EndDevice","zclVersion":2},"device_mode":"single_rocker","last_seen":"2022-12-27T15:39:29.758Z","linkquality":255}'
10
Zigbee2MQTT:info  2022-12-27 16:39:29: MQTT publish: topic 'zigbee2mqtt/Philips 929003017102/action', payload 'left_press'
9
Zigbee2MQTT:info  2022-12-27 16:39:29: MQTT publish: topic 'zigbee2mqtt/Philips 929003017102', payload '{"action":"left_press_release","battery":100,"device":{"applicationVersion":9,"dateCode":"20210504","friendlyName":"Philips 929003017102","hardwareVersion":3,"ieeeAddr":"0x001788010cc38f82","manufacturerID":4107,"manufacturerName":"Signify Netherlands B.V.","model":"929003017102","networkAddress":17991,"powerSource":"Battery","softwareBuildID":"1.0.5","stackVersion":2,"type":"EndDevice","zclVersion":2},"device_mode":"single_rocker","last_seen":"2022-12-27T15:39:29.861Z","linkquality":255}'
8
Zigbee2MQTT:info  2022-12-27 16:39:29: MQTT publish: topic 'zigbee2mqtt/Philips 929003017102', payload '{"action":"","battery":100,"device":{"applicationVersion":9,"dateCode":"20210504","friendlyName":"Philips 929003017102","hardwareVersion":3,"ieeeAddr":"0x001788010cc38f82","manufacturer [... cut ...] 003017102","hardwareVersion":3,"ieeeAddr":"0x001788010cc38f82","manufacturerID":4107,"manufacturerName":"Signify Netherlands B.V.","model":"929003017102","networkAddress":17991,"powerSource":"Battery","softwareBuildID":"1.0.5","stackVersion":2,"type":"EndDevice","zclVersion":2},"device_mode":"single_rocker","last_seen":"2022-12-27T15:39:36.837Z","linkquality":255}'
1
Zigbee2MQTT:info  2022-12-27 16:39:36: MQTT publish: topic 'zigbee2mqtt/Philips 929003017102/action', payload 'left_press_release'

danieledwardgeorgehitchcock, 2022-12-27:
This is correct - [See the FAQ's](https://www.zigbee2mqtt.io/guide/faq/#why-is-the-action-property-always-empty)

TheIronRock95, 2022-12-27:
A read over it.. Thank you

closed 2022-12-27 by TheIronRock95