Koenkk/zigbee2mqtt #15772: [New device support]: Marmitek Sense ME

Peterpc58, 2022-12-27:
### Link

https://marmitek.com/products/sense-me-motion-sensor-wireless-scene-activation-zigbee-3-0?scr=yes

### Database entry

{"id":83,"type":"EndDevice","ieeeAddr":"0x8cf681fffeba94eb","nwkAddr":40761,"manufId":4098,"manufName":"_TYZB01_yr95mpib","powerSource":"Battery","modelId":"SM0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280,61183],"outClusterList":[25],"clusters":{"genPowerCfg":{"attributes":{"batteryVoltage":32,"batteryPercentageRemaining":200}},"genBasic":{"attributes":{"modelId":"SM0202","manufacturerName":"_TYZB01_yr95mpib","zclVersion":3,"appVersion":73,"powerSource":3}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0022a54ffb","zoneState":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":73,"stackVersion":7,"hwVersion":1,"dateCode":"20190728","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1672152455032,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to follow the supporting new device page but got stuck on how to upload an image and the .js file.
[Marmitek_SenseME.zip](https://github.com/Koenkk/zigbee2mqtt/files/10309467/Marmitek_SenseME.zip)

Peterpc58, 2023-01-27:
Can this device be added?

Peterpc58, 2023-02-06:
Is this the right place to add a device, or must I place it somewhere else to add it?

Koenkk, 2023-05-22:
To confirm, does the external converter in the OP work?

Peterpc58, 2023-05-23:
Yes, the external converter is working.

Koenkk, 2023-05-24:
added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-05-24 by Koenkk