Koenkk/zigbee2mqtt #15774: [New device support]:  _TZE200_9yapgbuv   TS0601

derekangl, 2022-12-27:
### Link

https://www.aliexpress.com/item/1005004838346776.html?spm=a2g0o.order_list.order_list_main.11.4e3518022GTOrV

### Database entry

{"id":1,"type":"Coordinator","ieeeAddr":"0x04cd15fffee45691","nwkAddr":0,"manufId":0,"epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":48879,"inClusterList":[0,3,6,10,25,26,768],"outClusterList":[0,3,4,5,6,8,32,768,1024,1026,1029,1030,1280,2817,2819,2820,1794,4096,64513,64514],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":1672158325362,"defaultSendRequestWhen":"immediate"} {"id":2,"type":"EndDevice","ieeeAddr":"0xcc86ecfffe167699","nwkAddr":58948,"manufId":4098,"manufName":"_TZ3000_wkai4ga5","powerSource":"Battery","modelId":"TS0044","epList":[1,2,3,4],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[0,1,6],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�=+\u0011�=+\u0012","65506":31,"65508":0,"65534":0,"modelId":"TS0044","manufacturerName":"_TZ3000_wkai4ga5","stackVersion":0,"dateCode":"","appVersion":66,"zclVersion":3,"powerSource":3}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":166,"batteryVoltage":29}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":0,"inClusterList":[1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":0,"inClusterList":[1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":0,"inClusterList":[1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1672156871492,"defaultSendRequestWhen":"immediate"} {"id":3,"type":"Router","ieeeAddr":"0x00124b0022fe3591","nwkAddr":28706,"manufId":0,"manufName":"eWeLink","powerSource":"Mains (single phase)","modelId":"SA-003-Zigbee","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":9,"inClusterList":[0,3,4,5,6],"outClusterList":[0],"clusters":{"genOnOff":{"attributes":{"onOff":1}},"genBasic":{"attributes":{"modelId":"SA-003-Zigbee","manufacturerName":"eWeLink"}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x04cd15fffee45691","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":4,"hwVersion":1,"dateCode":"20201026","swBuildId":"1.0.3","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1672158388058,"defaultSendRequestWhen":"immediate"} {"id":4,"type":"Router","ieeeAddr":"0x00124b00246ca341","nwkAddr":24947,"manufId":0,"manufName":"eWeLink","powerSource":"Mains (single phase)","modelId":"SA-003-Zigbee","epList":[1],"endpoints":{"1":{"profId":49246,"epId":1,"devId":16,"inClusterList":[0,3,4,5,6],"outClusterList":[0],"clusters":{"genOnOff":{"attributes":{"onOff":1}},"genBasic":{"attributes":{"modelId":"SA-003-Zigbee","manufacturerName":"eWeLink"}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x04cd15fffee45691","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":8,"hwVersion":1,"dateCode":"20210703","swBuildId":"1.0.7","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1672158123840,"defaultSendRequestWhen":"immediate"} {"id":5,"type":"EndDevice","ieeeAddr":"0xa4c138c5e2218d1b","nwkAddr":32008,"manufId":4417,"manufName":"_TZ3000_fllyghyj","powerSource":"Battery","modelId":"TS0201","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[1,3,1026,1029,0],"outClusterList":[3,25,10],"clusters":{"genBasic":{"attributes":{"65503":"��=+\u0011G�=+\u0012","65506":51,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_fllyghyj","zclVersion":3,"appVersion":69,"modelId":"TS0201","powerSource":3}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2168}},"msRelativeHumidity":{"attributes":{"measuredValue":5843}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1672157766845,"defaultSendRequestWhen":"immediate"} {"id":6,"type":"EndDevice","ieeeAddr":"0xa4c13805ccf9d86a","nwkAddr":1578,"manufId":4417,"manufName":"_TZE200_9yapgbuv","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"L�=+\u0019L�=+\u0007\u0000\u0000\u0000\u0000\u0011","65506":56,"65508":1,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_9yapgbuv","powerSource":3,"zclVersion":3,"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1672157854030,"defaultSendRequestWhen":"immediate"} {"id":7,"type":"EndDevice","ieeeAddr":"0xa4c138669b05318d","nwkAddr":4129,"manufId":4417,"manufName":"_TZ3000_txpirhfq","powerSource":"Mains (single phase)","modelId":"TS0011","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"57344":{"attributes":{"53251":"AAAA"}},"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":0}},"manuSpecificTuya_3":{"attributes":{"switchType":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x04cd15fffee45691","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-999208499},"lastSeen":1672156882418,"defaultSendRequestWhen":"immediate"} {"id":8,"type":"EndDevice","ieeeAddr":"0xa4c13878c8e80064","nwkAddr":30209,"manufId":4417,"manufName":"_TZ3000_hhiodade","powerSource":"Mains (single phase)","modelId":"TS0011","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":2,"moesStartUpOnOff":1}},"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x04cd15fffee45691","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-999208499},"lastSeen":1672154539731,"defaultSendRequestWhen":"immediate"} {"id":9,"type":"Router","ieeeAddr":"0x00124b00251b83e2","nwkAddr":2617,"manufId":0,"manufName":"eWeLink","powerSource":"Mains (single phase)","modelId":"SA-003-Zigbee","epList":[1],"endpoints":{"1":{"profId":49246,"epId":1,"devId":16,"inClusterList":[0,3,4,5,6],"outClusterList":[0],"clusters":{"genOnOff":{"attributes":{"onOff":0}},"genBasic":{"attributes":{"modelId":"SA-003-Zigbee","manufacturerName":"eWeLink"}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x04cd15fffee45691","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":8,"hwVersion":1,"dateCode":"20210703","swBuildId":"1.0.7","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1672158123876,"defaultSendRequestWhen":"immediate"} {"id":10,"type":"EndDevice","ieeeAddr":"0xa4c138b7bb9c21b4","nwkAddr":57931,"manufId":4417,"manufName":"_TZ3000_mh9px7cq","powerSource":"Battery","modelId":"TS0044","epList":[1,2,3,4],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[1,6,57344,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0017�=+\u0012","65506":49,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_mh9px7cq","zclVersion":3,"appVersion":67,"modelId":"TS0044","powerSource":3}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":0,"inClusterList":[6,1,6],"outCluste [... cut ...] ":72,"manufacturerName":"_TZE200_9yapgbuv","zclVersion":3,"modelId":"TS0601","powerSource":3}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1672157794570,"defaultSendRequestWhen":"immediate"}

### Comments

hi, I have a problem with paring this i think is problem paring product is different. please see https://zigbee.blakadder.com/Tuya_ZTH01.html.

derekangl, 2022-12-27:
[image]
[image]

alexkiddde, 2023-01-02:
Same problem here with _TZE200_9yapgbuv - would be great to support this.

Koenkk, 2023-01-02:
`_TZE200_9yapgbuv` is supported in z2m 1.29.0

derekangl, 2023-01-02:
Hi, Happy New Year 2023.
thank you for all your support.
I upgraded to the new version of 1.29.0 and still does not fully work this sensor.

[image]
[image]
[image]
[image]

**I am adding log:**

Debug 2023-01-02 21:08:19Saving state to file /app/data/state.json
Debug 2023-01-02 21:08:36Received Zigbee message from '0xa4c13805ccf9d86a', type 'attributeReport', cluster 'genBasic', data '{"65503":"{\u0003F+f|\u0003F+\u0012|\u0003F+e�\u0003F+\u0019�\u0003F+\u0007\u0000\u0000\u0000\u0000\u0011"}' from endpoint 1 with groupID 0
Debug 2023-01-02 21:08:44Received Zigbee message from 'kichenHT', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
Debug 2023-01-02 21:09:03Received Zigbee message from '0xa4c13805ccf9d86a', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
Debug 2023-01-02 21:09:05Received Zigbee message from '0xa4c13805ccf9d86a', type 'attributeReport', cluster 'genBasic', data '{"65506":56,"65508":1,"appVersion":72}' from endpoint 1 with groupID 0
Debug 2023-01-02 21:10:58Received Zigbee message from 'kichenHT', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2168}' from endpoint 1 with groupID 0
Info 2023-01-02 21:10:58MQTT publish: topic 'z2mR22/kichenHT', payload '{"battery":100,"humidity":51,"linkquality":212,"temperature":21.68,"voltage":3000}'
Debug 2023-01-02 21:10:58Received Zigbee message from 'kichenHT', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":5128}' from endpoint 1 with groupID 0
Info 2023-01-02 21:10:58MQTT publish: topic 'z2mR22/kichenHT', payload '{"battery":100,"humidity":51,"linkquality":208,"temperature":21.68,"voltage":3000}'
Debug 2023-01-02 21:10:59Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"data":[25,11,2,10,0,0],"type":"Buffer"}' from endpoint 242 with groupID null, ignoring since it is from coordinator
Debug 2023-01-02 21:11:08Received Zigbee message from '0xa4c13805ccf9d86a', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
Debug 2023-01-02 21:11:09Received Zigbee message from '0xa4c13805ccf9d86a', type 'attributeReport', cluster 'genBasic', data '{"65506":56,"65508":1,"appVersion":72}' from endpoint 1 with groupID 0
Debug 2023-01-02 21:11:31Received Zigbee message from 'socket1', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info 2023-01-02 21:11:31MQTT publish: topic 'z2mR22/socket1', payload '{"linkquality":104,"state":"ON"}'
Debug 2023-01-02 21:12:06Received Zigbee message from 'socket2', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info 2023-01-02 21:12:06MQTT publish: topic 'z2mR22/socket2', payload '{"linkquality":144,"state":"ON"}'
Debug 2023-01-02 21:12:21Received Zigbee message from '0xa4c13805ccf9d86a', type 'attributeReport', cluster 'genBasic', data '{"65503":"l\u0004F+\u0019l\u0004F+\u0007\u0000\u0000\u0000\u0000\u0011�\u0004F+\u0019�\u0004F+\u0007\u0000\u0000\u0000\u0000\u0011"}' from endpoint 1 with groupID 0
Debug 2023-01-02 21:13:19Saving state to file /app/data/state.json

Thank you for help.

alexkiddde, 2023-01-02:
> `_TZE200_9yapgbuv` is supported in z2m 1.29.0

Updated to 1.29.0 (zigbee2mqtt in home assistant), reboot, reinstalled TZE200.

Still the same screen, no values reported.

[image]
[image]

Koenkk, 2023-01-03:
Can you try to:
- Force remove device through z2m frontend
- Restart z2m
- re-pair device.

alexkiddde, 2023-01-03:
> Can you try to:
> 
> * Force remove device through z2m frontend
> * Restart z2m
> * re-pair device.

I did, even twice. No changes....

Koenkk, 2023-01-04:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/6878634d9cd8ba56d00e19d4938ae7c1 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

alexkiddde, 2023-01-04:
> Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/6878634d9cd8ba56d00e19d4938ae7c1
>
> * save this as file next to `configuration.yaml` as [...]

Did what you said, the file is now listed as external converter in z2m. Removed device by force, restart, repairing. Same result:

[image]

derekangl, 2023-01-04:
hi, I would like to say thank you  for trying to issue fix , I am also Removed the device by force, restarting, repairing, and noting better.
[image]
[image]
[image]
[image]
[image]

Koenkk, 2023-01-05:
Can you try with https://gist.github.com/Koenkk/6878634d9cd8ba56d00e19d4938ae7c1 and provide me the debug logging of the following:

- force remove device
- restart z2m
- re-pair device
- blow into the sensor such that the humidity/temperature changes

alexkiddde, 2023-01-05:
> Can you try with https://gist.github.com/Koenkk/6878634d9cd8ba56d00e19d4938ae7c1 and provide me the debug logging of the following:
> 
> * force remove device
[image]
> * restart z2m
[image]
> * re-pair device
[image]
> * blow into the sensor such that the humidity/temperature changes
[image]

alexkiddde, 2023-01-05:
Okay, now I'm impressed. After 5 Minutes of nothing - it started to supply correct values. Only battery stats is missing...

[image]

Koenkk, 2023-01-06:
Seems this device supports `battery_state` instead of `battery` (%), updated https://gist.github.com/Koenkk/6878634d9cd8ba56d00e19d4938ae7c1

derekangl, 2023-01-06:
hi. i am sending this 
[image]
[image]
after updated

derekangl, 2023-01-06:
[image]

Koenkk, 2023-01-08:
Great, it will be addressed in https://github.com/Koenkk/zigbee-herdsman-converters/pull/5291

closed 2023-02-18 by github-actions[bot]