Koenkk/zigbee2mqtt #15776: SNZB-02 keeps disconnecting / freezing

JacekKac, 2022-12-28:
### What happened?

SNBZ-02 often keeps freezing - they are not reporting. Keeping 5sec gives "Device left" and then reconnecting. Devices is reporting again. 
Don't know exactly the reason.. nothing regular. Range is ok.

### What did you expect to happen?

constantly reporting

### How to reproduce it (minimal and precise)

not sure

### Zigbee2MQTT version

1.28.0-dev commit: f8d64b67 but happens on many

### Adapter firmware version

20220219

### Adapter

zStack3x0 /  usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_464f59958012ec11a5ec23c7bd930c07-if00-port0 

### Debug log

no logging available

Olomonster, 2022-12-28:
Did you try to force remove, restart z2m and then repair it?
Maybe adding a smaller report interval helps also.

JacekKac, 2022-12-28:
yes when i do this it's working fine, but not sure if it will be stable in the future. Recently my whole (small - 7 end devices network) was affected when i temporarly unpluged wall socket (repeater).  4 of SNZB-02 stopped reporting,pluging back the wall socket didn't help they where freazed

mimas71, 2022-12-29:
I confirm, after a power outage it is not possible to connect in the place where it worked normally before. When approaching the coordinator, after pressing the button it is possible to connect, after returning the snzb02 to its place, after about 28 hours it stops transmitting the status, but it is connected in zigbee2mqtt.

domoticz - plugin: v.3.2.0-beta, zigbee2mqtt: v.1.28.4 (zStack3x0 20210708)

JacekKac, 2022-12-30:
What is interesting that these sensor are not willing to use router wall plug as the Tuya thermostats do. numbers are rooms. Router wallplug is in room 1 - temp 1 should definitivly use wallplug instead of direct / poor connection to coordinator

[image]

JacekKac, 2022-12-31:
> I confirm, after a power outage it is not possible to connect in the place where it worked normally before. When approaching the coordinator, after pressing the button it is possible to connect, [...]

imho it's range issue and putting them back on place after reconfigure will not guarantee stable reporting

closed 2023-02-07 by github-actions[bot]