Koenkk/zigbee2mqtt #15779: [New device support]: _TZE200_eegnwoyw as TS0601_cover

hotchkj, 2022-12-28:
### Link

https://www.zemismart.com/products/bcm100db

### Database entry

{"id":54,"type":"EndDevice","ieeeAddr":"0x8cf681fffed1a98f","nwkAddr":57285,"manufId":4098,"manufName":"_TZE200_eegnwoyw","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1672217765740,"defaultSendRequestWhen":"immediate"}

### Comments

This appears to be a variant of the existing [TS0601_cover](https://www.zigbee2mqtt.io/devices/TS0601_cover.html), sold as Zenismart BCM100DB.

The only distinction I can spot vs the usual converter is that this is a fixed speed motor, i.e. `motor_speed` appears to do nothing & there are no settings or controls for changing the motor speed, merely open & close states.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_eegnwoyw'
        },
    ],
    model: 'TS0601_cover',
        vendor: 'TuYa',
        description: 'Curtain motor/roller blind motor/window pusher/tubular motor',
        whiteLabel: [
            {vendor: 'Zemismart', model: 'BCM100DB', description: 'Curtain motor'},
        ],
        fromZigbee: [fz.tuya_cover, fz.ignore_basic_report],
        toZigbee: [tz.tuya_cover_control, tz.tuya_cover_options],
        exposes: [
            e.cover_position().setAccess('position', ea.STATE_SET),
            exposes.composite('options', 'options')],
};

module.exports = definition;
```

hotchkj, 2023-01-30:
Still valid, github-actions, just awaiting any further requests for info, and/or enough data to make a PR.

Koenkk, 2023-01-30:
Make a pr with a copy of `TS0601_cover`, name it `TS0601_cover_1` and leave out the motor speed.

closed 2023-01-31 by Koenkk