Koenkk/zigbee2mqtt #15780: [Feature request]: Improve power_on_behavior by adding default brigtness level for bulbs

wiewiur, 2022-12-28:
### Is your feature request related to a problem? Please describe

I have few https://www.zigbee2mqtt.io/devices/LED2003G10.html bulbs, but i think problem exists not only for this Ikea LED's.

At the moment there is no way to set default brightness level on power on (when you turn on the electricity for this device). For example, after power on the bulb I want brightness to always be on level of 200.

 Z2M exposes two attributes related to this power on behaviour:

- `power_on_behavior` (`"off"`, `"on"`, `"toggle"`, `"previous"`)
- `color_temp_startup`

Setting `power_on_behavior` to `"on"` couses that the bulb's state after power on is always ON but with the last `brightness` level set before turning it off.

### Describe the solution you'd like

First solution that comes into my mind is to implement `brightness_startup` next to `color_temp_startup` . This `brightness_startup` value should be used when  `power_on_behavior` is set to `"on"`

### Describe alternatives you've considered

.

### Additional context

I don't know Is there any hardware/software limitation of bulbs cousing there is no way to set default brightness level. I have a lot of wifi bulbs of different brabds and all of them can do it and I thought this is basic thing. This IKEA model is only Zigbee bulb I have.

danieledwardgeorgehitchcock, 2022-12-29:
The `power_on_behaviour` that Z2M implements is as far as I am aware, what the device supports - Z2M cannot therefore extend that capability further. If the Ikea app allows brightness to be set as part of the power on behaviour when the device is linked to its native hub, please provide screenshot and zigbee sniffs and we can investigate how it can be implemented

wiewiur, 2022-12-29:
Unfortunately I dont have Ikea hub, so I can't check it. I am able to pair this bulb with Tuya hub, but Z2M have even more settings for this device than Tuya app :).

Can anyone with Ikea hub can confirm/deny?
Are there any bulbs which supports this feature in Z2M?

sjorge, 2022-12-29:
I was going to suggest to set level_config.on_level to the desired level, but it seems the tradfri bulbs don’t save this so once you cut tue power it forgets about it. So that makes it useless in combination with power_on_behavior. 😑

Tradfri bulbs forget a lot of things unfortunately (even all reporting)

closed 2023-02-05 by github-actions[bot]