Koenkk/zigbee2mqtt #15781: [New device support]: AwoX/EGLO 99106 Motion (PIR) sensor

elporsche, 2022-12-28:
### Link

https://www.eglo.com/nl/toebehoren-connect-z-sensor-99106.html

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0xa4c138fdfd4e52c8","nwkAddr":39368,"manufId":4417,"manufName":"AwoX","powerSource":"Battery","modelId":"TLSR82xx","epList":[1,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":263,"inClusterList":[0,3,4,4096],"outClusterList":[0,3,4,5,6,8,768,4096],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"epId":3,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":0,"zclVersion":3,"interviewCompleted":false,"meta":{"configured":88764544},"lastSeen":1672230133319,"defaultSendRequestWhen":"immediate"}

### Comments

I tried following the instructions to support a new device but I was unable to. On the debug log I found the 4 things  the device exposes:

`
Dec 28 12:44:47 raspberrypi npm[25179]: Zigbee2MQTT:debug 2022-12-28 12:44:47: Received Zigbee message from '0xa4c138fdfd4e52c8', type 'commandMoveToLevelWithOnOff', cluster 'genLevelCtrl', data '{"level":126,"transtime":0}' from endpoint 1 with groupID 32777
Dec 28 12:44:47 raspberrypi npm[25179]: Zigbee2MQTT:debug 2022-12-28 12:44:47: No converter available for '33951/33948' with cluster 'genLevelCtrl' and type 'commandMoveToLevelWithOnOff' and data '{"level":126,"transtime":0}'

Dec 28 12:44:47 raspberrypi npm[25179]: Zigbee2MQTT:debug 2022-12-28 12:44:47: Received Zigbee message from '0xa4c138fdfd4e52c8', type 'commandMoveToColorTemp', cluster 'lightingColorCtrl', data '{"colortemp":438,"transtime":0}' from endpoint 1 with groupID 32777
Dec 28 12:44:47 raspberrypi npm[25179]: Zigbee2MQTT:debug 2022-12-28 12:44:47: No converter available for '33951/33948' with cluster 'lightingColorCtrl' and type 'commandMoveToColorTemp' and data '{"colortemp":438,"transtime":0}'

Dec 28 12:44:48 raspberrypi npm[25179]: Zigbee2MQTT:debug 2022-12-28 12:44:48: Received Zigbee message from '0xa4c138fdfd4e52c8', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 32777
Dec 28 12:44:48 raspberrypi npm[25179]: Zigbee2MQTT:debug 2022-12-28 12:44:48: No converter available for '33951/33948' with cluster 'genOnOff' and type 'commandOn' and data '{}'

Dec 28 12:40:39 raspberrypi npm[25179]: Zigbee2MQTT:debug 2022-12-28 12:40:39: Received Zigbee message from '0xa4c138fdfd4e52c8', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 32777
Dec 28 12:40:39 raspberrypi npm[25179]: Zigbee2MQTT:debug 2022-12-28 12:40:39: No converter available for '33951/33948' with cluster 'genOnOff' and type 'commandOff' and data '{}'
`

During my attempts I noticed the following:

- The device is wrongly added as an AwoX 33951/33948
- 'CommandOff' is only sent if the brightness is too high, but when this command is sent it doesn't send a brightness measurement (as one would expect)
- 'commandMoveToColorTemp' looks like a color temp report but it's unclear to me if this is meant to represent a color in RGB or something else
- 'commandMoveToLevelWithOnOff' looks like a brightness report

### External converter

```shell
'
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['0xa4c138fdfd4e52c8'], // Couldn't find the modelID so this is the IEEE address (device is paired as a different device)
    model: '99106', // Vendor model number
    vendor: 'AwoX', // Vendor of the device (only used for documentation and startup logging)
    description: 'Eglo Connect-Z Motion (PIR) sensor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.brightness, fz.on_off, fz.color_colortemp], // External converters to be used
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.commandOn(), e.commandOff(), e.commandMoveToLevelWithOnOff(), e.commandMoveToColorTemp()], // Defines what this device exposes, note: device only sends a commandOff if the brightness gets way too high (and then it doesn't report the brigthness, only the Off command)
};

module.exports = definition;
'
```

Evililim, 2023-01-07:
I buy the same sensor for try, but I couldn't pair it on Z2M

elporsche, 2023-01-08:
Update from my side: this sensor ended up being only a couple euros cheaper than a Philips Hue Motion Sensor, so I ended up returning it and bought the Hue instead

closed 2023-02-18 by github-actions[bot]