Koenkk/zigbee2mqtt #15782: BOSCH RBSH-TRV0-ZB-EU --  not able to report pIHeatingDemand & default reporting should be expanded

Olomonster, 2022-12-28:
### What happened?

I would like to see the percentage of valve opened. 
I tried to setup a reporting to that attribute pIHeatingDemand from the cluster hvacThermostat. It says it's not supported.
Inside the bosch.js of zigbee-herdsman [https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/bosch.js#L413](Link to Bosch.js) it seems to be implemented.

Also I think it would be nice if the default reporting could be extended.

Current:

- HeatingSetpoint
- thermostatTemperature
- thermostatKeypadLockMode (aka childmode)
- batteryPercentageRemaining

Extend by:

- boost
- window_open
- display_orientation
- display_ontime
- display_brightness
- pIHeatingDemand

When I first set up the thermostate via installcode all the attributes from above were reported. 
After a restart of zigbee2mqtt/homebridge not all attributes were visible anymore and had to be manually set to be reported.

Thanks!

### What did you expect to happen?

I expected that all attributes from above are reportable.
After a restart I would like to see all reported attributes (like window_open or boost_mode) again.

### How to reproduce it (minimal and precise)

1. Add Bosch thermostate
2. Restart Zigbee2mqtt
3. Not all attributes available at reporting anymore

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-P // zStack3x0

### Debug log

Error 2022-12-28 13:56:02Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'ConfigureReporting 0x18fc260000048ef0/1 hvacThermostat([{"attribute":"pIHeatingDemand","minimumReportInterval":"0","maximumReportInterval":3600,"reportableChange":"10"}], {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Debug 2022-12-28 13:56:02Error: ConfigureReporting 0x18fc260000048ef0/1 hvacThermostat([{"attribute":"pIHeatingDemand","minimumReportInterval":"0","maximumReportInterval":3600,"reportableChange":"10"}], {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.configureReporting (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:694:22) at Bridge.deviceConfigureReporting (/opt/zigbee2mqtt/lib/extension/bridge.ts:448:9) at Bridge.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/bridge.ts:123:34)
Info 2022-12-28 13:56:02MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure_reporting', payload '{"data":{},"error":"ConfigureReporting 0x18fc260000048ef0/1 hvacThermostat([{\"attribute\":\"pIHeatingDemand\",\"minimumReportInterval\":\"0\",\"maximumReportInterval\":3600,\"reportableChange\":\"10\"}], {\"sendWhen\":\"active\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')","status":"error","transaction":"97wtn-4"}'

itkama, 2022-12-28:
I think this is already partly solved by @BBJake [here](https://github.com/Koenkk/zigbee-herdsman-converters/pull/5150). Should be in the next release I think.

BBJake, 2022-12-28:
Yes, in the next release pIHeatingDemand is supported and will be reported by the device.

[image]

What do you mean by?
"After a restart of zigbee2mqtt/homebridge not all attributes were visible anymore and had to be manually set to be reported."

can you add a screenshot?

The following are already in the actual version
- boost -> does not make sense to be reported, because this can't be changed on the device
- window_open - change on the device will be reported with the next version
- display_orientation -> does not make sense to be reported, because this can't be changed on the device
- display_ontime -> does not make sense to be reported, because this can't be changed on the device
- display_brightness -> does not make sense to be reported, because this can't be changed on the device
- child_lock  - change on the device will be reported with the next version
- system_mode - change on the device will be reported with the next version

[image]

I'm also looking at the moment for the option to display the actual temperature or the heating setpoint on the device display.
But this only works with newer firmware versions...

Olomonster, 2022-12-29:
With "After a restart of zigbee2mqtt/homebridge not all attributes were visible anymore and had to be manually set to be reported." I meant that inside the tab **status** only the following attributes were set:

battery, heatingSetpoint and localTemp. I had to manually recover the other attributes, but maybe I only was impatient for the automatic reports. I don't have a screenshot right now because I am not able to reproduce this..

Additonal question to @BBJake:
Do you also have problems with the localTemp and the heatingSetpoint?
For example I set the heatingSetpoint to 20°C. Then the Bosch thermostate heats to like 22-23°C and sometimes keeps heating to 24°C until I manually stop that by setting the heatingSetpoint to >20°C either physically or with Apple Home. Sometimes it just works...

The firmware: If I add the thermostats to the original Bosch Control Centre, will it be possible to upgrade the firmware of the thermostats or is it just the firmware of the control centre?
Is the functionality with zigbee2mqtt independent of the firmware or may it not work after an upgrade anymore?

Thanks!

BBJake, 2023-01-15:
> Additonal question to @BBJake: Do you also have problems with the localTemp and the heatingSetpoint? For example I set the heatingSetpoint to 20°C. Then the Bosch thermostate heats to like 22-23°C [...]

Yes, I had this some times. The firmware 3.04.03 should fix this...
It was mentioned (in german) here: https://www.bosch-smarthome.com/de/de/service/aktuelle-meldungen/release-infos/system-solutions/2022-10-05.html

> The firmware (currently 3.02.05): If I add the thermostats to the original Bosch Control Unit, will it be possible to upgrade the firmware of the thermostats or is it just the firmware of the control unit? Is the functionality with zigbee2mqtt independent of the firmware or may it not work after an upgrade anymore?

Yes, this should be independent, as the communication with Zigbee is a standard protocol.
I also updated my device with the Bosch SHC to 3.02.05

mipoat, 2023-01-28:
I'm not happy with my Popp 701721 because it simply will not detect window open by itself and does not react to window_open_external. The heating itself is also very strange and overshoots randomly. Thus I'd like to build my own PID controller, but to do that, the TRV needs to accept a pi_heating_demand set message so that I can control the % value directly.
Unfortunately the Popp does not accept a set but only a get (even if documentation states a set should be possible).

Long story short: Does the Bosch TRV really accept a set message for pi_heating_demand?
Anf if yes, will it stay forever in this remote controlled state once it received such a message or will it override the % value with its own after a while?

closed 2023-03-07 by github-actions[bot]