Koenkk/zigbee2mqtt #15783: [New device support]: Moes New Dimmer Switch 2 Gang with Dimmer based on TS0601

shawndamon, 2022-12-28:
### Link

https://moeshouse.com/products/new-zigbee-multi-gang-smart-light-dimmer-switch-relay-status-backlight-switch-off-us

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0xa4c1389d29ce441f","nwkAddr":51475,"manufId":4417,"manufName":"_TZE200_fjjbhx9d","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\t\u0000\u0000\u0000e\t\u0000\u0000\u0000\u0012\t\u0000\u0000\u0000e\t\u0000\u0000\u0000\u0012�\u0000\u0000\u0000e�\u0000\u0000\u0000\u0012�\u0001\u0000\u0000e�\u0001\u0000\u0000\u0012<\u0002\u0000\u0000e<\u0002\u0000\u0000\u0012","65506":32,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_fjjbhx9d","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1672233436175,"useImplicitCheckin":true}

### Comments

I did a search but i could not find this device being reported so far.

This seems to be a new switch that they launched and when i paired it i got a error message that this is not supported

warn  2022-12-28 21:36:28: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_fjjbhx9d'
warn  2022-12-28 21:36:28: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

I tried reading the support new device but got a little confused as the switch has on/off and dimming for the 2 gang and i wasnt sure how to construct the file.

### Supported color modes

None

### Color temperature range

None

pyrexfm, 2023-01-14:
I have the same device and is working (just added it). It had a hiccup, but after a while it was able to interview. I would retry pairing this.
However, my issue with it is that I'm not able to support changing the dimming range, which doesn't allow me to use this correctly with my led lights. Given TS0601_dimmer_1 supports I think it might be possible (I'll open another issue)

shawndamon, 2023-01-15:
Let me try it a few more times... Thank you for replying

closed 2023-02-22 by github-actions[bot]