Koenkk/zigbee2mqtt #15784: NODE_MODULE_VERSION 108 vs NODE_MODULE_VERSION 93

Dylantje, 2022-12-28:
### What happened?

After odroid/ armbian udpdate an error

### What did you expect to happen?

Z2m Still works

### How to reproduce it (minimal and precise)

Ligin with putty and see the error when do npm start

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

    20221102

### Adapter

sonoff zStack3x0

### Debug log

pi@odroidn2:~$ sudo systemctl stop  zigbee2mqtt
pi@odroidn2:~$ cd /opt/zigbee2mqtt
pi@odroidn2:/opt/zigbee2mqtt$ npm start

> zigbee2mqtt@1.28.4 start
> node index.js

Zigbee2MQTT:info  2022-12-28 15:00:42: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-12-28.15-00-41' filename: log.txt
Zigbee2MQTT:info  2022-12-28 15:00:42: Starting Zigbee2MQTT version 1.28.4 (commit #52e545f9)
Zigbee2MQTT:info  2022-12-28 15:00:42: Starting zigbee-herdsman (0.14.76)
Zigbee2MQTT:error 2022-12-28 15:00:42: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-12-28 15:00:42: Failed to start zigbee
Zigbee2MQTT:error 2022-12-28 15:00:42: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-12-28 15:00:42: Exiting...
Zigbee2MQTT:error 2022-12-28 15:00:42: Error: The module '/opt/zigbee2mqtt/node_modules/@serialport/bindings/build/Release/bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 93. This version of Node.js requires
NODE_MODULE_VERSION 108. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at Object.Module._extensions..node (node:internal/modules/cjs/loader:1243:18)
    at Module.load (node:internal/modules/cjs/loader:1037:32)
    at Function.Module._load (node:internal/modules/cjs/loader:878:12)
    at Module.require (node:internal/modules/cjs/loader:1061:19)
    at require (node:internal/modules/cjs/helpers:103:18)
    at bindings (/opt/zigbee2mqtt/node_modules/bindings/bindings.js:112:48)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/@serialport/bindings/lib/linux.js:2:36)
    at Module._compile (node:internal/modules/cjs/loader:1159:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1213:10)
    at Module.load (node:internal/modules/cjs/loader:1037:32)

lux73, 2022-12-28:
simple

`npm rebuild` should to - after that your problem is solved

Dylantje, 2022-12-29:
Simple yes, if you now this 👍 

Thanks this did the job!!!

Dylantje, 2022-12-29:
perhaps in the wiki?

closed 2022-12-29 by Dylantje

lux73, 2022-12-29:
you're welcome

feel free to add this to Wiki - but if you really read the Error Message carefully everything is crystal clear ^^ isn't it? ;-)

Dylantje, 2022-12-29:
> you're welcome
> 
> feel free to add this to Wiki - but if you really read the Error Message carefully everything is crystal clear ^^ isn't it? ;-)

Not for a windows noob.....
My english is not great, so when ik type something into te wiki :-(