Koenkk/zigbee2mqtt #15785: Xiaomi QBKG12LM not all actions are exposed

mikeS7, 2022-12-28:
### What happened?

According to [wiki](https://www.zigbee2mqtt.io/devices/QBKG12LM.html) QBKG12LM should exposed following actions:
single_left, single_right, single_both, double_left, double_right, double_both, hold_left, hold_right, hold_both, release_left, release_right, release_both.

The following are never rose:
double_both, hold_left, hold_right, hold_both, release_left, release_right, release_both

I've got only single_* actions instead.

### What did you expect to happen?

All actions should be caught and published.

### How to reproduce it (minimal and precise)

Nothing special, just try to perform physical actions on switch.
I can reproduce on CC2531 and EZSP v8.
Docker version on armv6 is in use.

### Zigbee2MQTT version

1.28.4 commit: 52e545f

### Adapter firmware version

6.7.10.0 build 423

### Adapter

EZSP v8

### Debug log

Issue with device 0x00158d0003052927 (SW-Bedroom-Light)
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10314554/log.txt)

mikeS7, 2022-12-28:
The same with QBKG11LM (no hold/release actions) but not with QBKG04LM.
Also I found the issue #6445

closed 2023-02-04 by github-actions[bot]

midlan, 2025-03-19:
@mikeS7 did you find any solution? having the same issue
https://github.com/Koenkk/zigbee2mqtt/issues/25861