Koenkk/zigbee2mqtt #15791: EZSP SkyConnect Fails to connect after update | Failure send version

Veldkornet, 2022-12-28:
### What happened?

I had everything working fine with EZSP and SkyConnect. 
However, the “Silicon Labs Multiprotocol” addon was updated to 0.11.1, and after their Z2M fails to start with the error below. 

Downgrading back to 0.11.0 did not resolve the problem. 
```
[20:57:55] INFO: Preparing to start...
[20:57:56] INFO: Socat not enabled
[20:57:58] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:error 2022-12-28 20:58:08: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-12-28 20:58:08: Failed to start zigbee
Zigbee2MQTT:error 2022-12-28 20:58:08: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-12-28 20:58:08: Exiting...
Zigbee2MQTT:error 2022-12-28 20:58:08: Error: Failure send version:{"type":"Buffer","data":[0,0,0,4]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:516:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
```

### Zigbee2MQTT version

1.28.4-1

### Adapter firmware version

?

### Adapter

HomeAssistant SkyConnect

### Debug log

[21:03:01] INFO: Preparing to start...
[21:03:02] INFO: Socat not enabled
[21:03:02] INFO: Zigbee Herdsman debug logging enabled
[21:03:04] INFO: Starting Zigbee2MQTT...
2022-12-28T20:03:10.912Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"rtscts":false,"path":"tcp://core-silabs-multiprotocol:9999","adapter":"ezsp"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2022-12-28T20:03:10.920Z zigbee-herdsman:adapter:ezsp:uart Opening TCP socket with core-silabs-multiprotocol:9999
2022-12-28T20:03:10.962Z zigbee-herdsman:adapter:ezsp:uart Socket connected
2022-12-28T20:03:10.964Z zigbee-herdsman:adapter:ezsp:uart Socket ready
2022-12-28T20:03:10.965Z zigbee-herdsman:adapter:ezsp:uart Uart reseting
2022-12-28T20:03:10.966Z zigbee-herdsman:adapter:ezsp:uart --> Write reset
2022-12-28T20:03:10.969Z zigbee-herdsman:adapter:ezsp:uart --> [1ac038bc7e]
2022-12-28T20:03:10.975Z zigbee-herdsman:adapter:ezsp:uart <-- [c1020b0a527e]
2022-12-28T20:03:10.978Z zigbee-herdsman:adapter:ezsp:uart <-- RSTACK: c1020b0a527e
2022-12-28T20:03:10.981Z zigbee-herdsman:adapter:ezsp:uart RSTACK Version: 2 Reason: RESET_SOFTWARE frame: c1020b0a527e
2022-12-28T20:03:10.984Z zigbee-herdsman:adapter:ezsp:ezsp ==> version: {"desiredProtocolVersion":4}
2022-12-28T20:03:10.988Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"version","_id_":0,"_isRequest_":true,"desiredProtocolVersion":4}
2022-12-28T20:03:10.992Z zigbee-herdsman:adapter:ezsp:uart --> DATA (0,0,0): 00000004
2022-12-28T20:03:10.996Z zigbee-herdsman:adapter:ezsp:uart --> [004221a850ed2c7e]
2022-12-28T20:03:10.997Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (1)
2022-12-28T20:03:12.999Z zigbee-herdsman:adapter:ezsp:uart --> Error: Error: {"sequence":1} after 2000ms
2022-12-28T20:03:12.999Z zigbee-herdsman:adapter:ezsp:uart -!- break waiting (1)
2022-12-28T20:03:12.999Z zigbee-herdsman:adapter:ezsp:uart Can't send DATA frame (0,0,0): 00000004
2022-12-28T20:03:13.000Z zigbee-herdsman:adapter:ezsp:uart ->> DATA (0,0,1): 00000004
2022-12-28T20:03:13.001Z zigbee-herdsman:adapter:ezsp:uart --> [084221a850ef017e]
2022-12-28T20:03:13.002Z zigbee-herdsman:adapter:ezsp:uart -?- rewaiting (1)
2022-12-28T20:03:15.003Z zigbee-herdsman:adapter:ezsp:uart --> Error: Error: {"sequence":1} after 2000ms
2022-12-28T20:03:15.003Z zigbee-herdsman:adapter:ezsp:uart -!- break rewaiting (1)
2022-12-28T20:03:15.003Z zigbee-herdsman:adapter:ezsp:uart Can't resend DATA frame (0,0,1): 00000004
2022-12-28T20:03:15.004Z zigbee-herdsman:adapter:ezsp:driv Reset connection. Try 0
2022-12-28T20:03:15.004Z zigbee-herdsman:adapter:ezsp:driv Stop driver
2022-12-28T20:03:15.004Z zigbee-herdsman:adapter:ezsp:ezsp Stop ezsp
Zigbee2MQTT:error 2022-12-28 21:03:15: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-12-28 21:03:15: Failed to start zigbee
Zigbee2MQTT:error 2022-12-28 21:03:15: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-12-28 21:03:15: Exiting...
Zigbee2MQTT:error 2022-12-28 21:03:15: Error: Failure send version:{"type":"Buffer","data":[0,0,0,4]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:516:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
2022-12-28T20:03:15.213Z zigbee-herdsman:adapter:ezsp:uart Port closed
2022-12-28T20:03:15.213Z zigbee-herdsman:adapter:ezsp:ezsp Close ezsp
2022-12-28T20:03:15.215Z zigbee-herdsman:adapter:ezsp:driv Close driver

MattWestb, 2022-12-28:
Did you having `Automatically flash firmware` in the addon enabled or disabled ?
If it was enabled you have getting RCP firmware flashed on you SC and must flashing it manual with one normal NCP for getting it working with Z2M.
Or you is trying if you can connecting over TCP port. You is setting the port under the last option `Network`.

Veldkornet, 2022-12-28:
Yup, I did have `Automatically flash firmware` enabled. 

Although I was already connecting via TCP. 
[image]

I’ve temporarily moved back to my SONOFF in the mean time.

MattWestb, 2022-12-29:
Was the IP:9999 as comport working with Z2M ?
It shall being the same functionality as one Sonoff ZBB.

I think the best / easiest was flashing your SC is with https://github.com/puddly/universal-silabs-flasher (its the flashed the addon have using) the it can talking with the RCP on the chip and rebooting it in to the bootloader and flashing new firmware.

Veldkornet, 2022-12-29:
> Was the IP:9999 as comport working with Z2M ?

Yup, was working perfectly actually until this mishap...

MattWestb, 2022-12-29:
So it was working with Silabs multi protocol until updating it in HA ??

If yes i think Z2M was not liking the port 9999 was not accessible then starting / running and was not able to (r)connecting to it.

That was / is happen with ZHA 2 but have getting better logic for fixing it but some time the addon is not starting the zigbeed OK and ZHA cant connect OK and its fixed by stopping and starting the addon (look in the addon log and you see if its getting connected to the right  handle in the protocol or disconnected).

Im running 3 RCPs and have moving most or my test devices to one ARM64 host for some weeks ago and its working very good with ZHA.
The problem is normally that the Zigbee system is sending one reset of the EZSP and the addon is loosing the connection to the SOC then the chip is rebooting and temporary is off line.

closed 2023-02-05 by github-actions[bot]