Koenkk/zigbee2mqtt #15794: SNZB-02 fails to pair to Conbee II

voltagex, 2022-12-29:
### What happened?

[snzb02.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/10317463/snzb02.pcapng.gz)

```
voltagex@debian:/opt/zigbee2mqtt$ node index.js
Zigbee2MQTT:debug 2022-12-29 14:36:37: Loaded state from file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2022-12-29 14:36:37: Logging to console only'
Zigbee2MQTT:info  2022-12-29 14:36:37: Starting Zigbee2MQTT version 1.28.4 (commit #52e545f)
Zigbee2MQTT:info  2022-12-29 14:36:37: Starting zigbee-herdsman (0.14.76)
Zigbee2MQTT:debug 2022-12-29 14:36:37: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"deconz","path":"/dev/ttyACM0"}}'
Zigbee2MQTT:info  2022-12-29 14:36:37: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2022-12-29 14:36:37: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":114,"product":0,"revision":"0x26720700","transportrev":0},"type":"ConBee2/RaspBee2"}'
Zigbee2MQTT:debug 2022-12-29 14:36:37: Zigbee network parameters: {"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754}
Zigbee2MQTT:info  2022-12-29 14:36:37: Currently 0 devices are joined:
Zigbee2MQTT:warn  2022-12-29 14:36:37: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2022-12-29 14:36:37: Allowing new devices to join.
Zigbee2MQTT:warn  2022-12-29 14:36:37: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2022-12-29 14:36:37: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2022-12-29 14:36:38: Connecting to MQTT server at mqtt://localhost
Zigbee2MQTT:debug 2022-12-29 14:36:38: Using MQTT anonymous login
Zigbee2MQTT:info  2022-12-29 14:36:38: Connected to MQTT server
Zigbee2MQTT:info  2022-12-29 14:36:38: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:info  2022-12-29 14:36:38: Started frontend on port 0.0.0.0:8080
Zigbee2MQTT:info  2022-12-29 14:36:38: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"52e545f","coordinator":{"meta":{"maintrel":0,"majorrel":38,"minorrel":114,"product":0,"revision":"0x26720700","transportrev":0},"type":"ConBee2/RaspBee2"},"log_level":"debug","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.28.4"}'
Zigbee2MQTT:info  2022-12-29 14:36:38: Zigbee2MQTT started!
Zigbee2MQTT:debug 2022-12-29 14:41:37: Saving state to file /opt/zigbee2mqtt/data/state.json
```

I was able to get the device into pairing mode but it did not seem to complete the connection.

### What did you expect to happen?

Device is added successfully

### How to reproduce it (minimal and precise)

Zigbee2MQTT version
    [1.28.4](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.28.4) commit: [52e545f](https://github.com/Koenkk/zigbee2mqtt/commit/52e545f)

Coordinator type
    ConBee2/RaspBee2

Coordinator revision
    0x26720700

Coordinator IEEE Address
    0x00212effff077051

Frontend version
    0.6.119

### Zigbee2MQTT version

    [1.28.4](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.28.4) commit: [52e545f](https://github.com/Koenkk/zigbee2mqtt/commit/52e545f)

### Adapter firmware version

0x26720700

### Adapter

Conbee II

voltagex, 2022-12-29:
pcap is from an nrf dongle.

I can also test with a zzh or skyconnect dongle.

Xeevis, 2022-12-29:
I have 13x SNZB-02 paired with ConBee II and they pair and work without any problems, make sure battery is alright and hold the button until it flashes and hold it near the coordinator. Also you should update CB firmware, solved some other problems for me, might help here as well. It's easy to miss because only middle digit is changing. 🤷‍♂️

0x267_2_0700 
0x267_8_0700

http://deconz.dresden-elektronik.de/deconz-firmware/

voltagex, 2023-01-01:
@Xeevis OK, with the updated firmware I'm able to get the SNZB-02 to turn up in deCONZ, but still not in z2m - any ideas?

[image]

closed 2023-02-08 by github-actions[bot]