Koenkk/zigbee2mqtt #15795: SMSZB-120 can't stop since 4.0.5 OTA upgrade

DamGranger, 2022-12-29:
### What happened?

After OTA update to 4.0.5 on SMSZB-120 the siren can't be turned off anymore from Z2M expose page, nor from Home Assistant. Tried repairing device but it doesn't change anything.

### What did you expect to happen?

The siren should stop

### How to reproduce it (minimal and precise)

Just start the siren, and try to stop it

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20220219

### Adapter

SONOFF cc2652p with zStack3x0

DamGranger, 2022-12-29:
In the debug log I can see :
```
Debug 2022-12-29 11:46:55Received MQTT message on 'zigbee2mqtt/AL BUR/set' with data '{"alarm":"OFF"}'
Debug 2022-12-29 11:46:55Publishing 'set' 'alarm' to 'AL BUR'
```

But it has no effect

TheStigh, 2022-12-29:
Are you using Conbee ?

DamGranger, 2022-12-29:
Hi, no I'm using SONOFF cc2652p with zStack3x0

TheStigh, 2022-12-29:
Weird, using the same. Mine are working as expected. Are you using the Frient version ?

DamGranger, 2022-12-29:
> Weird, using the same. Mine are working as expected. Are you using the Frient version ?

Yes they're the Frient ones.

TheStigh, 2022-12-29:
Seems to be an ongoing issue with Frient ones, don't know the reason.

DamGranger, 2022-12-29:
Wierdly they worked perfectly before the OTA upgrade, they were using 4.0.2 when I received them, but I'm not sure if it's original firmware from Develco or some kind of homebrew from Frient.

TheStigh, 2022-12-29:
I don't have access to Frient version. I know there are a different "version" of SMSZB-120 in Norway, I have the 4.0.6 for this one but haven't tested this fw on my original Develco. It might be an issue with Frient but without it, no clue at all and can't test.

DamGranger, 2022-12-29:
If you can share 4.0.6 and help me get the index.json file with the right informations, I'm willing to test on one of them if it'd help.

TheStigh, 2022-12-29:
> If you can share 4.0.6 and help me get the index.json file with the right informations, I'm willing to test on one of them if it'd help.

Currently, I was specifically told not to share this fw yet.

As a reference - all the ppl who has reported that Frient version doesn't find the OTA, just as an example. Dunno what it is with the Frient version - Original Develco doesn't nag, complain or fail, even with different coordinators. I have 3 different (but not the Conbee though).

DamGranger, 2022-12-29:
I have a conbee II, a sonoff cc2652p and a SkyConnect ezsp. I didn't try the latter yet but the Frient ones worked in 4.0.2 with the conbee II and they worked when I switched to cc2652p. When I say that they "worked", I obviously mean that they reacted to set {"alarm": "START"} and to {"alarm": "OFF"} now they react to the first one but never to the second command.

TheStigh, 2022-12-29:
> I have a conbee II, a sonoff cc2652p and a SkyConnect ezsp. I didn't try the latter yet but the Frient ones worked in 4.0.2 with the conbee II and they worked when I switched to cc2652p. When I say [...]

Ok, didn't test that, but I will

DamGranger, 2022-12-29:
Just to add some context, I'm actually using these as smoke detector, but also as sirens for Alarmo. And Alarmo triggers the sirens when detecting motion/opening without disarming the alarm using the entrance keypad.
I tracked in Z2M debug log the behavior and I caught that HomeAssistant and Z2M both issue these payloads when clicking on Start or Off button.
The payload is sent to topic /zigbee2mqtt/<DEVICE_NAME>/set and the data is either {"alarm": "START"} or {"alarm": "OFF"}

TheStigh, 2022-12-29:
I can confirm that {"alarm": "OFF"} also applies here, need to investigate. They might done some changes.

TheStigh, 2022-12-29:
Seems it is the `tz.simple_warning` that are broken / they've changed something.

DamGranger, 2022-12-29:
Ok si if you can reproduce on Develco that means it’s definitely the firmware and not some weird behavior in the white labeled ones.

DamGranger, 2022-12-29:
In the meantime the downgrade just finished and I can confirm that they're working as expected with 4.0.2

TheStigh, 2022-12-29:
> In the meantime the downgrade just finished and I can confirm that they're working as expected with 4.0.2

Are u on Telegram ?

DamGranger, 2022-12-29:
No sorry

TheStigh, 2022-12-29:
Please tell me - how did you proceed to downgrade ? Just for learning - never tried it :)

DamGranger, 2022-12-29:
Oh ! 
I created a file called my_index.json in /config/zigbee2mqtt containing this (just used the old content of index.json from zigbee-ota) :
```
[
    {
        "fileVersion": 262146,
        "fileSize": 198738,
        "manufacturerCode": 4117,
        "imageType": 593,
        "sha512": "6ff88b27a44d0762e62d70d278313803a241ace28433135be765bb3e3fe34ec848ce71d645e28c78a88005fac03e039d507553b2fc16e1fd1b024d571a9c84c9",
        "url": "https://github.com/Koenkk/zigbee-OTA/raw/master/images/Develco/ED_Smoke_Sensor_SSIG_4.0.2.zigbee",
        "path": "images/Develco/ED_Smoke_Sensor_SSIG_4.0.2.zigbee",
        "force": true
    }
]
```
Then in Z2M I went into the OTA settings and set :
[image]

I then restarted Z2M, went in OTA and voila!
Just note that "force": true is needed for downgrading.

TheStigh, 2022-12-29:
Also, done some more investigation - temp fix are to use the FULL command (until I've resolved the `warning_simple`)

Full command;
START:
`{"warning":{"duration":60,"level":"low","mode":"burglar","strobe":false,"strobe_duty_cycle":0}}`

STOP:
`{"warning":{"duration":60,"level":"low","mode":"stop","strobe":false,"strobe_duty_cycle":0}}`

Can you please try and report back if it worked for you also? Try on both 4.0.2 and 4.0.5 please.

DamGranger, 2022-12-29:
I can confirm this is working I already tried it. On both firmware version, unless I'm mistaken. Unfortunately, while this is usable in MQTT, Alarmo requires an entity to trigger the sirens so it won't work there. I'd have to setup an automation. While I reckon this is doable, I'd rather stick for now to 4.0.2 and wait for a definitive fix in the firmware.

TheStigh, 2022-12-29:
Okie - uploading 4.0.6 now and will do some tests. If OK, I'll ask for permission to publish this fw. I'll keep you updated.

TheStigh, 2022-12-29:
Still on 4.0.5 - found a solution - need to see how it affect other devices uses this converter

Claudio1L, 2022-12-29:
exactly same situation for me. Let me know if I can help

TheStigh, 2022-12-29:
@DamGranger @Claudio1L 

Thing is, I've fixed it for 4.0.5 but I don't know if it is backward-compatible.
Should I just publish it or should we test it on < 4.0.5 first?

DamGranger, 2022-12-29:
@TheStigh I'm on 4.0.2 right now on 3 devices, if you have an easy way to test your fix, I'm willing to do it. I can also upgrade one device to 4.0.5 to confirm that everything is working even on Frient. Alas it has to be done tomorrow as I'll be pretty occupied saturday and sunday due to new year's preparations and cleaning. I use Z2M as a module in HA, so it's all in a docker container and only /config is exposed I don't know if you can provide instructions that only involves this folder, but I'll change things inside the docker container if needed.

I'd say try to at least submit a PR quickly, as it would be nice if it could be in next Z2M version which I believe is due in 3 days.

Claudio1L, 2022-12-29:
[image]
[image]

that's my setup, on ESXi  VM with Z2M as addon. Tomorrow no way to test for me, I could do something after tomorrow (31th).  Have 5x Frient in 4.0.5 and 1x downgraded to 4.0.2

TheStigh, 2022-12-29:
Just had a guy to test my fix on 4.0.4 and worked perfect. I will make the PR so @Koenkk can add it to the January release

Claudio1L, 2022-12-30:
Guys,

I'm not able to remove the custom .jason file (my_index.jason) to allow again upgrade instead of a forced downgrade to 4.0.2

what I'm doing wrong?

[image]

DamGranger, 2022-12-30:
@TheStigh you can remove the line altogether in /config/zigbee2mqtt/configuration.yaml
[image]

TheStigh, 2022-12-30:
> @TheStigh you can remove the line altogether in /config/zigbee2mqtt/configuration.yaml [image]

Wan't me :) It was @Claudio1L

scxs535, 2022-12-30:
since yesterday after upgrading to 4.0.5, my SMSZB-120 doesn't work anymore, siren can't be turned off.  i removed the device from z2m and added a new device with 4.0.2, but the new device with 4.0.2 doesn't work either, siren can't be turned off.

DamGranger, 2022-12-30:
@scxs535 Hi, if you read the previous comments you’ll notice the fix is already merged in the next release of zigbee2mqtt (Thanks @TheStigh !). 
Now you just have to wait until January 1st for the release. 
For the second part of your message that’s weird because the sirens I downgraded to 4.0.2 works perfectly.

scxs535, 2022-12-30:
Thank you for your note.
Why it doesn't work for me with 4.0.2 yet, I don't know, I've tested that not only the turning off the siren doesn't work, but the turning it on also doesn't work with 4.0.2 on the new device.

scxs535, 2022-12-30:
just for your Information, I have now tried several times to delete and re-add the device, now it works as usual with 4.0.2.

Claudio1L, 2022-12-31:
> > @TheStigh you can remove the line altogether in /config/zigbee2mqtt/configuration.yaml [image]
> 
> Wasn't me :) It was @Claudio1L

Thank you, everything ok. I'm sorry for the dumb question

Claudio1L, 2022-12-31:
it's only me that the current version is 4.0.2 and the installed version 4.0.5?

[image]

Veldkornet, 2022-12-31:
> @TheStigh @DamGranger it's only me that the current version is 4.0.2 and the installed version 4.0.5?
> 
> [image]

I don’t have an entity like that in my HA. Mine (for the update part anyway) are are of class “update” not “binary_sensor”. And these show the correct information.

TheStigh, 2023-01-04:
> instead of 4.0.5 there is now an upgrade to 262149, should we upgrade it or should we rather stay at 4.0.2?

Up to you. The original bug from the 4.0.5 firmware update are already fixed in the January release of Z2M.

TheStigh, 2023-01-04:
> @TheStigh @DamGranger it's only me that the current version is 4.0.2 and the installed version 4.0.5?
> 
> [image]

[image]

If you remove and re-pair, it will look differently.

TheStigh, 2023-01-04:
NOTE TO ALL:

The original message in this topic has been fixed in the January release 1.29.0 of Z2M.
If you have a Conbee II coordinator, it is known to cause issues with lot more devices than SMSZB-120.

As always, please be sure to have the latest update of all software before publishing "something are broken". There might be fixes already in later releases.

@DamGranger, can you please close this thread as your initial issue has been fixed :)

closed 2023-01-06 by DamGranger

Tozzi89, 2023-08-04:
I have the same issue. All the latest updates. The alarm even goes on when pressing OFF. And if alarm is on I can’t turn it off..