Koenkk/zigbee2mqtt #15796: Latest Zigbee2MQTT Docker image does not support 386 architecture although it's stated in documentation.

hakapeszi76, 2022-12-29:
### What happened?

I tried to pull the docker image on an old Atom-based netbook and I got
```
$ docker pull koenkk/zigbee2mqtt --platform "linux/386"
Using default tag: latest
latest: Pulling from koenkk/zigbee2mqtt
no matching manifest for linux/386 in the manifest list entries

```
The manifest list only contains amd64, arm/v6, arm/v7, arm64. Apparently it worked up until 1.15.
(I'm a newbie in zigbee/docker and I haven't checked yet if 1.15 is good enough...)

### What did you expect to happen?

I expected that... It would work :)

### How to reproduce it (minimal and precise)

docker pull koenkk/zigbee2mqtt --platform "linux/386"

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

x

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

Koenkk, 2022-12-29:
I just published a dev image (if it works a release version will be available on 1 January). Can you check? `docker pull koenkk/zigbee2mqtt:latest-dev --platform "linux/386"`

hakapeszi76, 2022-12-29:
Yep, it pulls like charm. I'll continue the config/test steps later.
Many thanks!

closed 2022-12-29 by hakapeszi76