Koenkk/zigbee2mqtt #15798: [New device support]: H1 magnetic track light

ChengyiP, 2022-12-29:
### Link

https://www.aqara.com/cn/Track-light-H1_overview

### Database entry

{"id":7,"type":"Router","ieeeAddr":"0x54ef4410003772ad","nwkAddr":14779,"manufId":4447,"manufName":"Aqara","powerSource":"DC Source","modelId":"lumi.light.acn006","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[5,4,3,0,768,8,6,64704],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.light.acn006","manufacturerName":"Aqara","powerSource":4,"zclVersion":3,"appVersion":35,"stackVersion":27,"hwVersion":1,"dateCode":"20221008"}},"aqaraOpple":{"attributes":{"247":{"type":"Buffer","data":[3,40,45,5,33,37,0,8,33,35,1,9,33,0,26,10,33,0,0,11,32,2,12,32,10,160,32,1,154,32,0,100,32,1,101,32,254,102,33,27,1,106,35,140,10,112,23,105,32,0]},"1299":6000,"1300":2700}},"genOnOff":{"attributes":{"245":3783591,"onOff":1}},"genLevelCtrl":{"attributes":{"245":3783595,"currentLevel":254,"onLevel":254}},"lightingColorCtrl":{"attributes":{"245":3783603,"colorTemperature":283,"colorCapabilities":16,"colorTempPhysicalMin":166,"colorTempPhysicalMax":370,"colorMode":2,"currentX":0,"currentY":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":35,"stackVersion":27,"hwVersion":1,"dateCode":"20221008","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1672304156437,"defaultSendRequestWhen":"immediate"} {"id":8,"type":"Router","ieeeAddr":"0x54ef441000370845","nwkAddr":479,"manufId":4447,"manufName":"Aqara","powerSource":"DC Source","modelId":"lumi.light.acn013","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[5,4,3,0,768,8,6,64704],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.light.acn013","manufacturerName":"Aqara","powerSource":4,"zclVersion":3,"appVersion":35,"stackVersion":27,"hwVersion":1,"dateCode":"20221008"}},"aqaraOpple":{"attributes":{"247":{"type":"Buffer","data":[3,40,48,5,33,87,0,8,33,35,1,9,33,1,26,10,33,0,0,11,32,0,12,32,10,160,32,1,154,32,0,100,32,1,101,32,254,102,33,27,1,106,35,140,10,112,23,105,32,0]},"1299":6000,"1300":2700}},"genOnOff":{"attributes":{"245":122792,"onOff":1}},"genLevelCtrl":{"attributes":{"245":122796,"currentLevel":254,"onLevel":254}},"lightingColorCtrl":{"attributes":{"245":122804,"colorTemperature":283,"colorCapabilities":16,"colorTempPhysicalMin":166,"colorTempPhysicalMax":370,"colorMode":2}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":35,"stackVersion":27,"hwVersion":1,"dateCode":"20221008","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1672303917666,"defaultSendRequestWhen":"immediate"} {"id":9,"type":"Router","ieeeAddr":"0x54ef4410004aff0a","nwkAddr":58178,"manufId":4447,"manufName":"Aqara","powerSource":"DC Source","modelId":"lumi.light.acn008","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[5,4,3,0,768,8,6,64704],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.light.acn008","manufacturerName":"Aqara","powerSource":4,"zclVersion":3,"appVersion":35,"stackVersion":27,"hwVersion":1,"dateCode":"20221008"}},"aqaraOpple":{"attributes":{"247":{"type":"Buffer","data":[3,40,51,5,33,16,0,8,33,35,1,9,33,2,26,10,33,0,0,11,32,0,12,32,10,160,32,1,154,32,0,100,32,1,101,32,254,102,33,27,1,106,35,140,10,112,23,105,32,0]},"1299":6000,"1300":2700}},"genOnOff":{"attributes":{"245":14893737,"onOff":1}},"genLevelCtrl":{"attributes":{"245":14893741,"currentLevel":254,"onLevel":254}},"lightingColorCtrl":{"attributes":{"245":14893750,"colorTemperature":283,"colorCapabilities":16,"colorTempPhysicalMin":166,"colorTempPhysicalMax":370,"colorMode":2}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":35,"stackVersion":27,"hwVersion":1,"dateCode":"20221008","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1672304262305,"defaultSendRequestWhen":"immediate"} {"id":10,"type":"Router","ieeeAddr":"0x54ef44100037b432","nwkAddr":65066,"manufId":4447,"manufName":"Aqara","powerSource":"DC Source","modelId":"lumi.light.acn009","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[5,4,3,0,768,8,6,64704],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.light.acn009","manufacturerName":"Aqara","powerSource":4,"zclVersion":3,"appVersion":35,"stackVersion":27,"hwVersion":1,"dateCode":"20221008"}},"aqaraOpple":{"attributes":{"247":{"type":"Buffer","data":[3,40,49,5,33,26,0,8,33,35,1,9,33,1,26,10,33,0,0,11,32,0,12,32,10,160,32,1,154,32,0,100,32,1,101,32,254,102,33,27,1,106,35,140,10,112,23,105,32,0]},"1299":6000,"1300":2700}},"genOnOff":{"attributes":{"245":16657066,"onOff":1}},"genLevelCtrl":{"attributes":{"245":16657070,"currentLevel":254,"onLevel":254}},"lightingColorCtrl":{"attributes":{"245":16657077,"colorTemperature":283,"colorCapabilities":16,"colorTempPhysicalMin":166,"colorTempPhysicalMax":370,"colorMode":2}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":35,"stackVersion":27,"hwVersion":1,"dateCode":"20221008","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1672304035015,"defaultSendRequestWhen":"immediate"} {"id":11,"type":"Router","ieeeAddr":"0x54ef4410005b8572","nwkAddr":41559,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.light.acn004","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,2,3,4,5,6,8,9,768],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.light.acn004","manufacturerName":"LUMI","powerSource":1,"zclVersion":3,"appVersion":25,"stackVersion":2,"hwVersion":1,"dateCode":"02-12-2022"}},"aqaraOpple":{"attributes":{"223":{"type":"Buffer","data":[22,33,0,0,23,33,0,0,12,33,2,0,24,33,253,1,30,33,6,0,31,33,181,87,32,33,184,39,34,33,62,42,35,33,0,0,36,32,0,37,35,55,35,83,102,254,35,1,0,0,0]},"229":{"type":"Buffer","data":[22,17,5,154,157,37,60,238,122,37,81,66,145,37,74,106,181,37,89,166,135,37,81]},"247":{"type":"Buffer","data":[3,40,39,5,33,5,0,154,32,0,8,33,25,1,9,33,1,21,11,32,0,160,32,1,100,32,0,101,32,1,102,33,66,1,105,32,0,10,33,154,157,12,32,1]}}},"genOnOff":{"attributes":{"245":10639104,"onOff":0}},"genLevelCtrl":{"attributes":{"245":10639104,"currentLevel":1}},"lightingColorCtrl":{"attributes":{"245":10639104,"colorTemperature":322,"colorCapabilities":31,"colorTempPhysicalMin":166,"colorTempPhysicalMax":370,"startUpColorTemperature":322}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":25,"stackVersion":2,"hwVersion":1,"dateCode":"02-12-2022","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1672307786654,"defaultSendRequestWhen":"immediate"} {"id":12,"type":"Router","ieeeAddr":"0x54ef4410005b87b1","nwkAddr":41666,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.light.acn004","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,2,3,4,5,6,8,768,64704],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.light.acn004","manufacturerName":"LUMI","powerSource":1,"zclVersion":3,"appVersion":25,"stackVersion":2,"hwVersion":1,"dateCode":"02-12-2022"}},"aqaraOpple":{"attributes":{"223":{"type":"Buffer","data":[22,33,0,0,23,33,0,0,12,33,1,0,24,33,105,23,30,33,13,0,31,33,75,86,32,33,42,229,34,33,50,42,35,33,0,0,36,32,0,37,35,31,220,163,104,254,35,1,0,0,0]},"229":{"type":"Buffer","data":[21,17,4,238,122,37,83,42,254,37,75,116,115,37,94,106,181,37,86]},"247":{"type":"Buffer","data":[3,40,40,5,33,6,0,154,32,0,8,33,25,1,9,33,0,21,11,32,0,160,32,1,100,32,0,101,32,1,102,33,66,1,105,32,0,10,33,0,0,12,32,1]}}},"genOnOff":{"attributes":{"245":10666496,"onOff":0}},"genLevelCtrl":{"attributes":{"245":10666496,"currentLevel":1}},"lightingColorCtrl":{"attributes":{"245":10666496,"colorTemperature":322,"colorCapabilities":31,"colorTempPhysicalMin":166,"colorTempPhysicalMax":370,"startUpColorTemperature":322}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":25,"stackVersion":2,"hwVersion":1,"dateCode":"02-12-2022","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1672304064033,"defaultSendRequestWhen":"immediate"} {"id":13,"type":"Router","ieeeAddr":"0x54ef4410005b8e83","nwkAddr":17796,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.light.acn004","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList [... cut ...]
        extend: extend.light_onoff_brightness_colortemp(),
    },
    {
        zigbeeModel: ['lumi.light.acn004'],
        model: 'SSWQD02LM',
        vendor: 'Aqara',
        description: 'T1Pro Led Controller',
        extend: extend.light_onoff_brightness_colortemp(),
    },
    {
        zigbeeModel: ['lumi.light.acn026'],
        model: 'LTSZNSD04LM',
        vendor: 'Aqara',
        description: 'T2 Downlight 60 Deg',
        extend: extend.light_onoff_brightness_colortemp(),
    },
];

module.exports = definition;
```

Koenkk, 2022-12-30:
Can you provide a 150x150 jpg picture for each? (preferably with white background)

closed 2023-02-06 by github-actions[bot]

pmagnemi20, 2024-09-09:
> Can you provide a 150x150 jpg picture for each? (preferably with white background)

Can you reopen this issue and add h1 pro in supported devices?

pmagnemi20, 2024-09-09:
> Can you provide a 150x150 jpg picture for each? (preferably with white background)

[image]