Koenkk/zigbee2mqtt #15801: [New device support]: innr RC 210

maartenhendrix, 2022-12-29:
### Link

https://www.innr.com/en/product/smart-button/

### Database entry

{"id":19,"type":"EndDevice","ieeeAddr":"0x003c84fffe1d2e22","nwkAddr":29753,"manufId":4454,"manufName":"innr","powerSource":"Battery","modelId":"RC 210","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":261,"inClusterList":[0,1,3,2821,4096,64769],"outClusterList":[3,4,6,8,25,768,4096],"clusters":{"genBasic":{"attributes":{"modelId":"RC 210","manufacturerName":"innr","powerSource":3,"zclVersion":3,"appVersion":33,"stackVersion":101,"hwVersion":1,"dateCode":"20211118","swBuildId":"2.01.05"}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":33,"stackVersion":101,"hwVersion":1,"dateCode":"20211118","swBuildId":"2.01.05","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1672250215184,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to create an external converter but am a bit stuck.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
    {
        zigbeeModel: ['RC 210'],
        model: 'RC 210',
        vendor: 'Innr',
        description: 'Remote control',
        fromZigbee: [fz.command_on, fz.command_off, fz.battery],
        toZigbee: [],
        exposes: [e.battery(), e.action(['on', 'off'])],
        meta: {disableActionGroup: true, battery: {dontDividePercentage: true}},
        configure: async (device, coordinatorEndpoint, logger) => {
            // Binding genOnOff is not required to make device send events.
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
            await reporting.batteryPercentageRemaining(endpoint);
        },
    }
];
```

echocrow, 2022-12-29:
also got one of these today - without noticing it's not yet on the list of supported devices.

basing a a converter off of [existing Innr entries](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/innr.js), this seems to be doing the trick so far:
```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;

module.exports = [
    {
        zigbeeModel: ['RC 210'],
        model: 'RC 210',
        vendor: 'Innr',
        description: 'Remote control',
        fromZigbee: [fz.command_on, fz.command_off, fz.command_move, fz.command_stop, fz.command_move_to_level, fz.command_move_to_color_temp],
        toZigbee: [],
        exposes: [e.action(['on', 'off', 'brightness_move_up', 'brightness_move_down', 'brightness_stop', 'brightness_move_to_level', 'color_temperature_move'])],
        configure: async (device, coordinatorEndpoint, logger) => {
            const ep = device.getEndpoint(1);
            await reporting.bind(ep, coordinatorEndpoint, ['genBasic', 'genOnOff', 'genLevelCtrl', 'lightingColorCtrl']);
        },
    },
];
```
give it a try! I'll be messing with it some more, too

echocrow, 2022-12-30:
I've submitted a PR for initial support in [here](https://github.com/Koenkk/zigbee-herdsman-converters/pull/5229)

maartenhendrix, 2022-12-30:
Thanks that works.
I changed the code a bit:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;

module.exports = [
    {
        zigbeeModel: ['RC 210'],
        model: 'RC 210',
        vendor: 'Innr',
        description: 'Remote control',
        fromZigbee: [fz.command_on, fz.command_off],
        toZigbee: [],
        exposes: [e.action(['on', 'off'])],
        configure: async (device, coordinatorEndpoint, logger) => {
            const ep = device.getEndpoint(1);
            await reporting.bind(ep, coordinatorEndpoint, ['genBasic', 'genOnOff']);
        },
    },
];
```

echocrow, 2022-12-30:
@maartenhendrix are you able to react to long-press and double-tab events with this setup?

I started with a setup like that too. However, at least with the default button configuration, I believe this will not forward double and long-press actions. The RC 210 seems to emit as `brightness_move_up`/`brightness_move_down` and `brightness_move_to_level`/`color_temperature_move` respectively, hence why I added them back in (and this is probably also why similar actions are listed for the RC 250)

maartenhendrix, 2022-12-31:
I did not use those so i did not test it.

echocrow, 2022-12-31:
Gotcha!

In other news, looks like the [PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/5229) to add the RC 210 was already accepted. Doesn't appear to be live in this repo yet, but I'm guessing it'll become available with the next release 🤞

Koenkk, 2023-01-01:
Support in z2m 1.29.0 😄 Happy new year!

closed 2023-01-01 by Koenkk