Koenkk/zigbee2mqtt #15804: Moes Smart dimmer module (MS-105Z)

Steini1984, 2022-12-29:
### What happened?

I installed some [MS-105Z](https://moeshouse.com/products/zigbee-dimmer-module) and have found three issues I wanted to report: 

**1: I cannot bind** the modules to my wall switches (ZG2833K8_EU05) and Z2M gives this error: "Nothing to bind". I guess it´s because the "genOnOff" is not mapped as a input cluster:
[image]

I maybe thought it was not supported from the manufacturer, but I also installed a Moes MS-104NZ Switch module which can bind without problems. 
[image]

**2: Transition.** I use this with Home Assistant and when I set transition as a parameter I get this error:
[image]
The problem for is that when this fails it also fails for the other parameters, like transition (which work fine on their own)

**3: Wrong identification:** It is presented as a TS0601_dimmer_1 from TuYa and shows a wrong icon for a wall switch:
[image]

This would be more fitting:
[image]

### What did you expect to happen?

1: That I can bind the dimmer to my wall switch for on/off and dimming functions without using Z2M and Home assistant

2: That I could either set the transition time, or Z2M would gracefully ignore that request but set the brightness from the request

3: That I would get the correct name and icon for the module 

### How to reproduce it (minimal and precise)

1: Go to a wall switch and try to bind it to the dimming module:
[image]

2: Turn on the light via Home Assistant with a brightness and transition parameteres:
```
service: light.turn_on
data:
  brightness: 255
  transition: 1
target:
  entity_id: light.gangur_kristall
```

3: Add a Moes MS-105Z device to Z2M and go to "Devices"

### Zigbee2MQTT version

1.28.4 commit: unknown

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

arveeproject, 2023-01-24:
Anybody already looked into it? I have the same problem see #15803

Steini1984, 2023-01-25:
I tried making a custom converter to see if I can fix some of the bugs.

I´m running Homeassistant and the zigbee2mqtt as an addon. I tried making a custom converter from the (What I think are) the relevant parts of https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js  and also added 
"tz.ignore_transition" which I hope fixes issue nr 2. 

After some adjustments the converter does not give any errors and z2m loads, but I dont know **how to make z2m prefer my custom converter** over the default one. Any ideas?

Here is my custom converter: https://pastebin.com/raw/sB52tMME

closed 2023-03-07 by github-actions[bot]

ambarusa, 2023-05-12:
@Steini1984 Hi, I am also having the same blocker. Could you progress with the converter?

Steini1984, 2023-08-24:
> @Steini1984 Hi, I am also having the same blocker. Could you progress with the converter?

Sorry did not see you comment until now, but no I kinda gave up :(

juagicre, 2024-01-19:
How do you put the device in pairing mode?? 
Manual says: "After the wiring of a switch module is done, press the return button switch for 10 times (keep time interval not too short) until the beep sound continuously and rapidly as Di-Di-Di.... for pairing mode". 
In my case I don't want to use any switch, so I did not connect any (they show a diagram for this case in the manual as well!). 
Should I connect it just for this purpose? Would a regular ON/OFF switch work and myself doing the return action manually?

chelming, 2024-03-09:
> How do you put the device in pairing mode?

damn this was annoying (and the first result on google)

To put the device into pairing mode you need to connect `L` to `S` ten times in quick succession and then give it a second after the tenth one. I just used a multimeter to do this instead of connecting a switch. You'll hear three loud beeps then rapid beeps. It took me three tries to get it in pairing mode.

EuthaNasi, 2025-02-21:
> damn this was annoying (and the first result on google)
>
> To put the device into pairing mode you need to connect `L` to `S` ten times in quick succession and then give it a second after the tenth [...]

Thanks for your insight, I connected a regular light switch to it and turned it on-off for 10 times and it worked like a charm. Based on the manual I thought you needed a smart switch but luckily not!