Koenkk/zigbee2mqtt #15805: Error updating Philips 9290022268

drjjr2, 2022-12-29:
### What happened?

I have notifications that my Philips 9290022268 have an update but when I click "Update device firmware" I get an error "Image contains padding bytes"
Devices are currently on 1.93.11
[image]

### What did you expect to happen?

Update to complete

### How to reproduce it (minimal and precise)

Attempt to update firmware

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20220219

### Adapter

Sonoff

FredericGuardia, 2022-12-29:
Hi,

I confirm same issue on my side with some other Philips HUE bulb models under same current firmware version : 1.93.11. 
I have an another bulb model (8718696695203) with firmware 1.88.1 updated today to 1.101.2 without any issue. I am on Zigbee2MQTT version 1.28.4 with SMLIGHT CC2652P adapter loaded with coordinator firmware  version 20221102.

dbbdark1990, 2022-12-29:
Same here 

Update of 'Lamp [Zolder]' failed (Request failed with status code 403)

Hue pulb 1.93.11

yayitazale, 2022-12-29:
Same here with model [915005996901](https://www.zigbee2mqtt.io/devices/915005996901.html#philips-915005996901) on firmware 1.93.7

mattbruman, 2022-12-29:
Getting same issue and HA keeps pestering me that there is a update but it fails.

Model:
9290022267
929002294101
9290022166

Scoff123, 2022-12-29:
Had this also. I have 2 Philips Hue bulbs. One of them I updated in the Philips Hue app yesterday, which went fine. 
Although Z2M then showed that it was running newer firmware than what was available. This message disappeared today.
The second bulb I tried updating in HA as I had a notification in Z2M that an update was available. 
I tried updating from the HA notification, but nothing seemed to happen. I then opened the Z2M web UI and saw the update was available under the OTA menu. 
I tried updating but received the same error message "image padding contains invalid bytes".

I though I would update via the Philips Hue app instead, but when I opened the app, it showed that the update was mid progress. It completed in the app successfully, and now shows in Z2M as the latest firmware.

TheJulianJES, 2022-12-29:
Please see: https://github.com/Koenkk/zigbee-OTA/issues/218

Some Hue images weren't parsed correctly by Z2M. The bugfix is already in the ``dev`` release of Z2M.
I think we should also get 1.29 in the next few days with the bugfix for this.

For now, just don't update the affected bulbs.

Koenkk, 2022-12-30:
this will be fixed in 1.29.0 indeed (available on 1 January)

closed 2022-12-30 by Koenkk

dbbdark1990, 2023-01-01:
Updated minut ago still get OTA error 403

danielsmith89, 2023-01-01:
> Updated minut ago still get OTA error 403

It's working for me, albeit very slowly!

dbbdark1990, 2023-01-01:
Strange ... all the hue bulbs giving update availble still give this error

yayitazale, 2023-01-01:
It is working for me. Thanks!!

-------- Mensaje original --------

hugalafutro, 2023-01-01:
For me after update Zigbee2MQTT claimed suddenly no new updates for my philips bulbs, but HA still pestered me there is, and HA now forced z2m to install it.
[image]

[image]

I was only able to start the install from HA not from Z2M.

dbbdark1990, 2023-01-01:
Lol i get the same now z2mqtt says no update any more HA did but can't update installed number stays the same

hugalafutro, 2023-01-01:
@dbbdark1990 have you pressed the install button in ha? It took about a minute till the progress bar appeared in Z2M, the number in HA hasn't changed and the Install button is still visible as the install is still going, I'm hoping it will disappear once the install finishes.

dbbdark1990, 2023-01-01:
@hugalafutro nope when i press it i directly get the same messages as before:

Update of 'Lamp [Gang]' failed (Request failed with status code 403) from HA update aswell as update inside zigbee2mqtt

dbbdark1990, 2023-01-01:
Zigbee2MQTT versie
1.29.0 Commit: Unknown

Coördinator type
ConBee2/RaspBee2

Coordinator revisie
0x26720700

Coordinator IEEE adres
0x00212effff080dee

Versie frontend
0.6.121

brhahlen, 2023-01-02:
@Koenkk Just a small ping, just to make you aware.

I've got this issue as well. Have updated two Philips Hue bulbs to 1.101.7 from 1.93.11

HA still shows these as having updates, but Z2M does not.

yayitazale, 2023-01-02:
One question, I have the model [4090330P9](https://www.zigbee2mqtt.io/devices/4090330P9.html#philips-4090330p9) on firm 1.93.7 and check for update says that I'm on latest and there is no update pending.

The other lamp model that I have [915005996901](https://www.zigbee2mqtt.io/devices/915005996901.html#philips-915005996901) also was on firmware 1.93.7 but I was able to update it to 1.101.5

Is this correct? I think 1.101.X update also applies to 4090330P9 model but I'm not sure.

dbbdark1990, 2023-01-02:
I have a work arround set my own index.json with self downloaded file on a location and now i can update ;)

Koenkk, 2023-01-03:
@brhahlen 1.29.0 currently has a bug regarding this, I will publish a hotfix release in the coming days.

brhahlen, 2023-01-03:
> @brhahlen 1.29.0 currently has a bug regarding this, I will publish a hotfix release in the coming days.

Awesome! Thank you!