Koenkk/zigbee2mqtt #15814: Spamming "device_announced" message

Bastiencc, 2022-12-30:
### What happened?

Hi

I have several devices (old one in the network) that sometimes send device_announced message.
Some are battery powered, some are bulbes.

Is that normal ?

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20220219

### Adapter

zzh

### Debug log

info  2022-12-29 21:07:27: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"wc_1_plafond","ieee_address":"0x00158d0002d6f577"},"type":"device_announce"}'
info  2022-12-29 21:07:27: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"wc_1_plafond"},"type":"device_announced"}'
info  2022-12-29 21:07:29: MQTT publish: topic 'zigbee2mqtt/wc_1_plafond', payload '{"brightness":121,"color_mode":"color_temp","color_temp":338,"linkquality":127,"power_on_behavior":null,"power_outage_memory":true,"state":"OFF","update":{"installed_version":"1.32\u0000","state":"idle"},"update_available":false}'
info  2022-12-29 21:07:35: MQTT publish: topic 'zigbee2mqtt/bureau_plafond', payload '{"brightness":254,"color_mode":"color_temp","color_temp":338,"linkquality":76,"power_on_behavior":null,"power_outage_memory":true,"state":"OFF","update":{"installed_version":"1.32\u0000","state":"idle"},"update_available":false}'
info  2022-12-29 21:07:45: Accepting joining not in blocklist device '0x0015bc0031015fe0'
info  2022-12-29 21:07:46: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"detecteur_fumee","ieee_address":"0x0015bc0031015fe0"},"type":"device_announce"}'
info  2022-12-29 21:07:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"detecteur_fumee"},"type":"device_announced"}'

Bastiencc, 2023-01-02:
no idea ? :(

Koenkk, 2023-01-03:
This message is usually send by a device when it is powered on.

ReneTode, 2023-01-04:
i got the same after updating from 1.25 to 1.28
more and more device announce themselves regularly.
on 1.25 i only had that for 1 device (a 4 gang tuya switch) but now its several different devices.

on discord i hear from others who experience the same.

somehow it seems that the network is weakened and devices lose connection to the network and reappear.

its possible that those devices lost the connection in the past and reconnected, without noticing, but someone i just spoke to noticed that when his bulbs re-announced they blinked (which would be hard not to notice before)

closed 2023-01-18 by Bastiencc