Koenkk/zigbee2mqtt #15819: E1812 - wrong payload

farao780, 2022-12-30:
### What happened?

i have e1812 as a  bell button outside my door connected to zigbee2mqtt.
Noticed that everybody were knocking at my door in last month, but nobody pushed the button. So I thought the battery will be the problem, although mqtt showed 100% (but it runs for over an year). So I changed CR2032 with a new one, but still no red led lightning up when I push the button.
Normally I removed the device and re-paired. Interview went succesfully, but now I cannot have action button and in logs I have 
when I push the button:
Zigbee2MQTT:info  2022-12-30 10:51:34: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x680ae2fffec1db8f","ieee_address":"0x680ae2fffec1db8f"},"type":"device_announce"}'
Zigbee2MQTT:info  2022-12-30 10:51:34: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x680ae2fffec1db8f"},"type":"device_announced"}' 

Firmware of the button is 2.3.015 

### What did you expect to happen?

I expect that sensor.(device)_action should be on when the button is pressed, but instead i have "device announced" . And I cannot use as expected button press

### Zigbee2MQTT version

1.28.4-1 

### Adapter firmware version

20201010

### Adapter

c2538 + 2592 zStack30x

Bastiencc, 2022-12-30:
I think I have the same problem [here](https://github.com/Koenkk/zigbee2mqtt/issues/15814)

RubenKelevra, 2022-12-30:
I got an E1812, after pretty rarely use (say two button presses per day 3 times a week) of it the battery died yesterday on it. So not sure but this may be related as I didn't expected that the battery dies after just 10 month of ownership.

I'm using Adapter Firmware Version 20220219 on a cc2652rb with Zigbee2MQTT 1.28.4.

closed 2023-02-06 by github-actions[bot]