Koenkk/zigbee2mqtt #15824: Error with TS0011 switch_modules

RafaelDuncan, 2022-12-30:
### What happened?

I have some Zigbee TS0011 switch_modules that I'm having trouble with.
I keep getting the following 2 errors:

ERROR 1:
Publish 'set' 'state' to '0xa4c138e99b4ac057' failed: 'Error: Command 0xa4c138e99b4ac057/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery ":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 35751 - 1 - 67 - 6 - 11 after 10000ms)'

ERROR 2:
Publish 'set' 'state' to '0xa4c138e99b4ac057' failed: 'Error: Command 0xa4c138e99b4ac057/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'

I can't operate the module either, the buttons don't work.
who could help me to solve this problem?
I have tried several modules of this model, and all have the same problem.

### What did you expect to happen?

I would expect the module to work as it is also supported by zigbee2mqtt.

### How to reproduce it (minimal and precise)

I get the message when I want to register the device and also when I turn the module on or off with the manual switch.

### Zigbee2MQTT version

1.25.0-dev commit: 7e61252

### Adapter firmware version

20200715

### Adapter

usb-Silicon_Labs_slae.sh_cc2652rb_stic

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10325465/log.txt)

RafaelDuncan, 2022-12-30:
[log (2).txt](https://github.com/Koenkk/zigbee2mqtt/files/10325911/log.2.txt)

RafaelDuncan, 2023-01-03:
@Koenkk I would like to report an update on the above issue. I have now performed a test with an alternative ZIgbee Gateway. Same Version of Home Assistant, Same Version of Zigbee2MQTT. The TS0011 Switch module does work here.
The only difference is the Coordinator version in the working test version is 20210708.
Coordinator version 20200715 thus gives the errors. Can I conclude that it is due to the coordinator version?

Koenkk, 2023-01-04:
@RafaelDuncan this could be the case. Since it works now, can this be closed?

closed 2023-02-11 by github-actions[bot]