Koenkk/zigbee2mqtt #15830: Zigbee2MQTT misses the first notification from the Tradfri motion sensor after startup

rccoleman, 2022-12-30:
### What happened?

I joined a Tradfri motion sensor (https://www.ikea.com/us/en/p/tradfri-wireless-motion-sensor-smart-white-60377655/) to my SkyConnect 1.0 using the EZSP v9 adapter and it consistently misses the first occupancy trigger whenever I start/restart Zigbee2MQTT.  If I trigger the motion sensor, which Z2M doesn't recognize, and wait a few mins, it will suddenly start recognizing occupancy triggers and will work fine until the next Z2M restart.  This is completely reproducible.

I enabled the herdsman log and I see some variant of this on the first trigger that Z2M doesn't see/recognize:

```
2022-12-30T16:49:09.005Z zigbee-herdsman:adapter:ezsp:uart <-- [4102b1a96b2a164fa6904b23aa546d499d4e272712ce675a3c7e]
2022-12-30T16:49:09.006Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (4,1,0): 4102b1a96b2a164fa6904b23aa546d499d4e272712ce675a3c7e
2022-12-30T16:49:09.006Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (5)
2022-12-30T16:49:09.007Z zigbee-herdsman:adapter:ezsp:uart --> [8520dd7e]
2022-12-30T16:49:09.009Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 4090013f0003fdff0401060001ff000100008cff0000
2022-12-30T16:49:09.010Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x3f: {"_id_":63,"_cls_":"messageSentHandler","_isRequest_":false,"type":3,"indexOrDestination":65533,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":255,"options":256,"groupId":0,"sequence":140},"messageTag":255,"status":0,"message":{"type":"Buffer","data":[]}}
2022-12-30T16:49:09.011Z zigbee-herdsman:adapter:ezsp:ezsp ==> setMulticastTableEntry: {"index":13,"value":{"endpoint":1,"multicastId":0,"networkIndex":0}}
2022-12-30T16:49:09.012Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"setMulticastTableEntry","_id_":100,"_isRequest_":true,"index":13,"value":{"endpoint":1,"multicastId":0,"networkIndex":0}}
2022-12-30T16:49:09.014Z zigbee-herdsman:adapter:ezsp:uart --> DATA (1,5,0): 41000164000d00000100
2022-12-30T16:49:09.015Z zigbee-herdsman:adapter:ezsp:uart --> [150321a9302a7d38b259954a144a7e]
2022-12-30T16:49:09.016Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (2)
2022-12-30T16:49:09.026Z zigbee-herdsman:adapter:ezsp:uart <-- [5203a1a9302a1542727e]
2022-12-30T16:49:09.026Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (5,2,0): 5203a1a9302a1542727e
2022-12-30T16:49:09.027Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (6)
2022-12-30T16:49:09.027Z zigbee-herdsman:adapter:ezsp:uart --> [8610be7e]
2022-12-30T16:49:09.029Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 418001640000
2022-12-30T16:49:09.030Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x64: {"_id_":100,"_cls_":"setMulticastTableEntry","_isRequest_":false,"status":0}
2022-12-30T16:49:09.031Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (2) success
2022-12-30T16:49:09.032Z zigbee-herdsman:adapter:ezsp:cast Set MulticastTableEntry #13 for 0 multicast id: 0
```

Here's another instance:

```
2022-12-30T21:13:48.040Z zigbee-herdsman:adapter:ezsp:uart <-- [5421b1a96b2a164fa6904b23aa546d499d4e270912ce67c29d7e]
2022-12-30T21:13:48.041Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (5,4,0): 5421b1a96b2a164fa6904b23aa546d499d4e270912ce67c29d7e
2022-12-30T21:13:48.041Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (6)
2022-12-30T21:13:48.042Z zigbee-herdsman:adapter:ezsp:uart --> [8610be7e]
2022-12-30T21:13:48.043Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 6390013f0003fdff0401060001ff00010000a2ff0000
2022-12-30T21:13:48.043Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x3f: {"_id_":63,"_cls_":"messageSentHandler","_isRequest_":false,"type":3,"indexOrDestination":65533,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":255,"options":256,"groupId":0,"sequence":162},"messageTag":255,"status":0,"message":{"type":"Buffer","data":[]}}
2022-12-30T21:13:48.044Z zigbee-herdsman:adapter:ezsp:ezsp ==> setMulticastTableEntry: {"index":13,"value":{"endpoint":1,"multicastId":0,"networkIndex":0}}
2022-12-30T21:13:48.045Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"setMulticastTableEntry","_id_":100,"_isRequest_":true,"index":13,"value":{"endpoint":1,"multicastId":0,"networkIndex":0}}
2022-12-30T21:13:48.046Z zigbee-herdsman:adapter:ezsp:uart --> DATA (4,6,0): 64000164000d00000100
2022-12-30T21:13:48.047Z zigbee-herdsman:adapter:ezsp:uart --> [462621a9302a7d38b259954a86257e]
2022-12-30T21:13:48.047Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (5)
2022-12-30T21:13:48.061Z zigbee-herdsman:adapter:ezsp:uart <-- [6526a1a9302a1574b27e]
2022-12-30T21:13:48.062Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (6,5,0): 6526a1a9302a1574b27e
2022-12-30T21:13:48.062Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (7)
2022-12-30T21:13:48.063Z zigbee-herdsman:adapter:ezsp:uart --> [87009f7e]
2022-12-30T21:13:48.065Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 648001640000
2022-12-30T21:13:48.066Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x64: {"_id_":100,"_cls_":"setMulticastTableEntry","_isRequest_":false,"status":0}
2022-12-30T21:13:48.067Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (5) success
2022-12-30T21:13:48.069Z zigbee-herdsman:adapter:ezsp:cast Set MulticastTableEntry #13 for 0 multicast id: 0
```

And a third:

```
2022-12-30T21:35:10.178Z zigbee-herdsman:adapter:ezsp:uart <-- [447d31b1a96b2a164fa6904b23aa546d499d4e270f12ce6702997e]
2022-12-30T21:35:10.179Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (4,4,0): 4411b1a96b2a164fa6904b23aa546d499d4e270f12ce6702997e
2022-12-30T21:35:10.179Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (5)
2022-12-30T21:35:10.180Z zigbee-herdsman:adapter:ezsp:uart --> [8520dd7e]
2022-12-30T21:35:10.181Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 5390013f0003fdff0401060001ff00010000a4ff0000
2022-12-30T21:35:10.182Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x3f: {"_id_":63,"_cls_":"messageSentHandler","_isRequest_":false,"type":3,"indexOrDestination":65533,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":255,"options":256,"groupId":0,"sequence":164},"messageTag":255,"status":0,"message":{"type":"Buffer","data":[]}}
2022-12-30T21:35:10.183Z zigbee-herdsman:adapter:ezsp:ezsp ==> setMulticastTableEntry: {"index":12,"value":{"endpoint":1,"multicastId":0,"networkIndex":0}}
2022-12-30T21:35:10.184Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"setMulticastTableEntry","_id_":100,"_isRequest_":true,"index":12,"value":{"endpoint":1,"multicastId":0,"networkIndex":0}}
2022-12-30T21:35:10.186Z zigbee-herdsman:adapter:ezsp:uart --> DATA (4,5,0): 54000164000c00000100
2022-12-30T21:35:10.187Z zigbee-herdsman:adapter:ezsp:uart --> [451621a9302a19b259954af9157e]
2022-12-30T21:35:10.188Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (5)
2022-12-30T21:35:10.197Z zigbee-herdsman:adapter:ezsp:uart <-- [5516a1a9302a1502b37e]
2022-12-30T21:35:10.198Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (5,5,0): 5516a1a9302a1502b37e
2022-12-30T21:35:10.198Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (6)
2022-12-30T21:35:10.199Z zigbee-herdsman:adapter:ezsp:uart --> [8610be7e]
[... 75 lines cut ...]

### What did you expect to happen?

I expect that the Tradfri motion sensor will respond correctly after restarting Z2M, and not ignore the first trigger.

### How to reproduce it (minimal and precise)

Restart Z2M and observe that the Tradfri motion sensor state doesn't reflect "occupied" in response to the first motion trigger, but eventually starts responding.  This behavior repeats every time I restart Z2M.

### Zigbee2MQTT version

1.28.4 & 1.28.4-dev

### Adapter firmware version

7.1.1.0 build 273

### Adapter

SkyConnect 1.0 using EZSP v9

### Debug log

See the herdsman logs above, and I'm attaching the latest debug log here.
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10326336/log.txt)

rccoleman, 2022-12-31:
I don't know if it will help understand what's going on, but I built a packet sniffer using a CC2531 and captured the first and second motion triggers.  When applying a filter for the motion sensor as source or destination (0xdb06), I see the following for the first trigger that doesn't reflect properly in Z2M:

```
18	6.816631	0x2d06	Broadcast		ZigBee HA	54	ZCL OnOff: On with timed off, Seq: 58
20	6.828319	0x2d06	Broadcast		ZigBee HA	54	ZCL OnOff: On with timed off, Seq: 58
37	7.302141	0x2d06	Broadcast		ZigBee HA	54	ZCL OnOff: On with timed off, Seq: 58
44	7.775980	0x2d06	Broadcast		ZigBee HA	54	ZCL OnOff: On with timed off, Seq: 58
46	7.849198	0x2d06	0xdb06		IEEE 802.15.4	12	Data Request
49	8.881497	0x2d06	0xdb06		IEEE 802.15.4	12	Data Request
51	9.913465	0x2d06	0xdb06		IEEE 802.15.4	12	Data Request
58	10.333200	0xdb06	0xafff		IEEE 802.15.4	105	Data, Dst: 0xafff, Src: 0xdb06
```

And for the second trigger that does reflect properly:

```
4	1.038755	0xdb06	Broadcast		IEEE 802.15.4	104	Beacon, Dst: Broadcast, Src: 0xdb06, Bad FCS
26	10.763168	0x2d06	Broadcast		ZigBee HA	54	ZCL OnOff: On with timed off, Seq: 59
28	10.772341	0x2d06	Broadcast		ZigBee HA	54	ZCL OnOff: On with timed off, Seq: 59
39	10.894555	0xdb06	0x0000		ZigBee	77	Route Reply, Responder: 0x2d06, Originator: 0x0000
40	10.899097	0xdb06	0x0000		ZigBee	77	Route Reply, Responder: 0x2d06, Originator: 0x0000
41	10.908427	0xdb06	0x0000		ZigBee	77	Route Reply, Responder: 0x2d06, Originator: 0x0000
51	11.082966	0x0000	0x2d06		ZigBee HA	50	ZCL: Default Response, Seq: 59
67	11.275291	0x2d06	Broadcast		ZigBee HA	54	ZCL OnOff: On with timed off, Seq: 59
71	11.320957	0xdb06	0x0000		ZigBee	77	Route Reply, Responder: 0x2d06, Originator: 0x0000
100	11.734189	0xdb06	0x0000		ZigBee	77	Route Reply, Responder: 0x2d06, Originator: 0x0000
101	11.741580	0xdb06	0x0000		ZigBee	77	Route Reply, Responder: 0x2d06, Originator: 0x0000
102	11.749204	0xdb06	0x0000		ZigBee	77	Route Reply, Responder: 0x2d06, Originator: 0x0000
103	11.758483	0xdb06	0x0000		ZigBee	77	Route Reply, Responder: 0x2d06, Originator: 0x0000
106	11.767780	0x2d06	Broadcast		ZigBee HA	54	ZCL OnOff: On with timed off, Seq: 59
107	11.796896	0x2d06	0xdb06		IEEE 802.15.4	12	Data Request
110	11.804750	0x0000	0x2d06		ZigBee HA	50	ZCL: Default Response, Seq: 59
112	11.810568	0x2d06	0x0000		ZigBee	45	APS: Ack, Dst Endpt: 1, Src Endpt: 1
114	11.815251	0x2d06	0x0000		ZigBee	57	Route Record, Dst: 0x0000
115	11.819793	0x2d06	0x0000		ZigBee	57	Route Record, Dst: 0x0000
117	11.867898	0x2d06	0x0000		ZigBee	45	APS: Ack, Dst Endpt: 1, Src Endpt: 1
122	12.831282	0x2d06	0xdb06		IEEE 802.15.4	12	Data Request
124	13.864128	0x2d06	0xdb06		IEEE 802.15.4	12	Data Request
```

I'm not sure yet what to make of that, other than there's a lot more traffic when it works.  I'll continue to poke around with it, but pcap files are attached if you want to look closer.

[pcap.tgz](https://github.com/Koenkk/zigbee2mqtt/files/10328078/pcap.tgz)

MattWestb, 2023-01-01:
Cant see what is inside the package then have no network key but can see enough.
The sleeping end device is sending one on with off command to its parent, the parent sending one 15.4 ACK, the parent is broadcasting the command to the network and all (what i have seen but with some bad received package) routers is relaying the broadcast (to the group).
The device is then pulling its parent some seconds and it acking all 15.4 commands.

Th second is the coordinator sending one default response (cant see the command network key is failing) and the device is acking it in both 15.4 and Zigbee layer.
This not not needed / wanted / shall not being then the sleeper was sending one broadcast and and is shall not being acked with unicast and is "pasive acked" of the network (the parent is sending it and then listening if all neighbors is relaying it OK = OK and if not its repeating the command).

So the first sniff is 110% OK in the network level but it can being that the coordinator / system have not doing all the setting OK so its not doing all OK.

I can looking deeper in the network trafic if having the network key but is not needed then all looks OK

You is having many bad frames received but the network looks working OK no bad reseeding of frames and and it can being the sniffing CC-2531 that is having bad signal but if you is getting problem look if you is having WiFi on the same channel.
And SkyConnect is having much better radio retriever so very likely not having problems.

rccoleman, 2023-01-01:
Thanks for the comments!  I was going to add that based on the network map, the motion sensor (0x2D06) is routed through 0xDB06, so that's why it keeps popping up in the traffic.  When I was collecting these pcap dumps, I was walking into sight of the motion sensor holding my laptop with the sniffer dangling from a USB port, so it was directly in sight of the motion sensor & router device and the coordinator was almost directly above me on the 2nd floor on its provided USB extension cable.  I suppose there could be interference - I'm using the default of channel 11, and I have several Unifi APs around my house that are strategically placed on Wifi channels 1, 6, and 11.  I'm happy to provide the network key if it would help decoding the pcap trace, but it sounds like maybe it's not necessary.  Do let me know if you think it would help.

For what it's worth, I swapped out the Tradfri motion sensor for an Aquara, which joined with a direct connection to the coordinator, and it has no issues working immediately after starting Z2M.  I don't know if taking the router out of the equation helped, or if there's some other difference in how the Tradfri and Aquara sensors behave.

I'm having trouble parsing this language:
> Th second is the coordinator sending one default response (cant see the command network key is failing) and the device is acking it in both 15.4 and Zigbee layer.
> This not not needed / wanted / [...]

If I understand correctly, there's some excess/unnecessary traffic in the second trace that isn't relevant to this issue?

In any case, what I took away from the capture is that:
- The only difference in configuration setup between the two traces is restarting Z2M
- In the failing case, I see no evidence in the herdsman UART log that the coordinator saw the initial command from the motion sensor.  Since it sounds like you're saying that the network traffic looks good, and the antenna in the SkyConnect should be good enough to see the same traffic that I see with the sniffer, there's maybe something in the coordinator firmware or the configuration of the SkyConnect that's getting in the way.

RubenKelevra, 2023-01-14:
I also see issues with my TuYa [TS0202](https://www.zigbee2mqtt.io/devices/TS0202.html#tuya-ts0202) sensors, as explained [here](https://github.com/Koenkk/zigbee2mqtt/issues/15707#issuecomment-1368089862). The Aqara sensor and the Ikea sensors work … **better** … but sometimes I see a delay of 5–10 seconds, after they should see and report motion, until it's shown in Z2M as detected.

So either they can't get a package routed through the network, or the coordinator ignores it – as the information seems to be repeated after a timeout occurs.

The TS0202 seems to not repeat the package and the information just gets lost. So only once the state is “clear” and then “occupied” again this state will properly show up in Z2M.

So I guess the issues could be related? — as in both cases Z2M seems to not receive or process the sent information properly.

Adapter: Slaesh's CC2652RB stick
Adapter Firmware: 20220219
Z2M with this problem: at least 1.28.x, 1.29.0, 1.29.1

closed 2023-02-21 by github-actions[bot]