Koenkk/zigbee2mqtt #15839: TuYa TS0601_thermostat: firmware update shows no current version – which is confusing

RubenKelevra, 2022-12-31:
### What happened?

My TuYa TS0601 thermostats are showing up in the OTA page, so I can check/update their firmware. But the check does nothing – there's no version currently installed reported/shown by these devices (as well as no build date):

[image]

The info page of those devices does not show this info either:

[image]

---

Without a currently installed version it's completely unclear to me how the firmware update check is even done. It's also unclear if it's a bug that it shows “no update available”, as it cannot compare the currently installed version to the available versions?!

So I dug into the source code of Zigbee2MQTT and the debug logs and found, that in the end the value of `genBasic.appVersion` is used to check for firmware updates. This value is logged in the debug log as `fileVersion` – but never shown in the GUI.

The currently ([#5b3d218](https://github.com/Koenkk/zigbee-OTA/commit/5b3d218de38d7270c6baf1fb81a23bc78540a1e4)) available firmware version is `fileVersion` 87 and that's the same value the device reports if asked for it's `genBasic.appVersion` value.

So in the end the update check process is working correctly, as the available image is the same as the installed one. Just the GUI is non-optimal here.

### What did you expect to happen?

To follow the principle of least astonishment it is important to show this internal `fileVersion` value to the users as primary version number, instead of the only descriptive 'Firmware version' and 'Firmware build date' values.

#### Device's About page

On the device page the reported `fileVersion` should be added.

#### OTA page

The value of 'Firmware version' and 'Firmware build date' could compacted into a single row, by showing it as “$firmware_version ($firmware_build_date)”. This way there's enough room for a new row for the `fileVersion` value.

### How to reproduce it (minimal and precise)

- Add a TuYa TS0601_thermostat to the network
- Go to the OTA page
- See an empty row for the firmware version and the firmware build date for this device

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20220219

### Adapter

Slaesh's CC2652RB stick

Koenkk, 2023-01-01:
@nurikk this may be a possible improvement for this page. Starting with z2m the update object will have an `installed_version` and `latest_version`. This is used in Zigbee to determine wether a new fw is available (not the swbuildid or datecode).

Scoff123, 2023-01-02:
Different device but the same is happening with one of my zigbee devices - Aqara smart wall switch H1 EU (with neutral, single rocker).

In the OTA page there is no data displayed for this device in the Firmware version column. In the About page for the device, it shows a Firmware build date 11-10-2021, but no Firmware version field like my other devices do.
If I go into the State page for the device, under Update section, it shows installed_version: 2839 and latest_version: 2839.
The Dev console shows genBasic: appVersion: 23

ovizii, 2023-01-15:
I just had the same problem (no installed version, no latest version available) with the same TRV: _TZE200_cpmgn2cf although a firmware check said: update available, so I went ahead. It says remaining time: 25:54 - will continue this post when the update is done.

Normally I wouldn't risk this, but I was having a problem with this device, so it can't get any worse.

While waiting for the firmware update to finish, I noticed a notification from HA:
[image]

BedTherm is the name of this TRV, and apparently it was on firmware version 85 and it's now updating to 87 

The udpate failed around 6% saying the device had not requested any blocks!? I was too slow to take a screenshot.

PetrMa, 2023-10-20:
Hi,

is there any update on this issue please?
Currently I have the same issue on all of my TuYa TS0601_thermostat and TS011F_plug_1 devices

thanks

mvzut, 2023-11-16:
I ran into this same issue and stumbled upon this page. I have six TS0601 TRVs and some updated without any problems, and some didn't. What I found out is that for a successful OTA firmware update, the devices must be close to the coordinator. It doesn't work when they are connected via an in-between router. Of course, in the case of TRVs or other "fixed" devices it's not very convenient to unscrew them to bring them closer to the coordinator, but this was the only solution for these three valves in my case.

xamadevp, 2023-12-05:
Same issue here. Worth reopening the issue.

@mvzut  Thanks for the report. Do you know what's in the firmware update? I'm afraid of bricking my TRVs and therefore weighing out the risk of updating.

mvzut, 2023-12-05:
> Same issue here. Worth reopening the issue.
> 
> 
> 
> @mvzut  Thanks for the report. Do you know what's in the firmware update? I'm afraid of bricking my TRVs and therefore weighing out the risk of updating.

To be honest I have no clue. I didn't notice any difference so far. I just upgraded because I always do, no other reason.

By the way, I'm actually quite happy with the valves. Of course the temperature stability is not very good due to the built-in sensor, which is very inaccurate during heating. But I solved that by creating generic thermostats for some rooms, based on a better sensor somewhere else in the room.

G3rry71, 2024-12-18:
Hi,

is there any update on this issue please?
Currently I have the same issue on all of my TuYa TS0601_thermostat
Thanks

RubenKelevra, 2024-12-18:
@mvzut wrote:
> To be honest I have no clue. I didn't notice any difference so far. I just upgraded because I always do, no other reason.
>
> By the way, I'm actually quite happy with the valves. Of course the [...]

Same here. With an external temperature probe it keeps the room nicely within 0.5 °C.

My old automation for this plus overswing protection can be found here:

https://github.com/RubenKelevra/Tuya-thermostatic-valve

Currently I use Better Thermostat, which does basically the same thing, with this bugfix:

https://github.com/KartoffelToby/better_thermostat/pull/1472

And a simple overswing protection, so I turn off better thermostat if the valve of the TRV is too wide open while approaching the setpoint, to quickly close it.

@G3rry71 unfortunately not. The version is still not shown for me.

@Koenkk can you please reopen this?

ferropolis, 2024-12-22:
I'am not sure, that's helpfull, but I have the same problems at my HA.

`2024-12-22 13:42:38z2m: Update of 'Heizung Kinderzimmer' failed (Timeout. Device did not start/finish firmware download after being notified. (Error: Timeout - 19239 - 1 - null - 25 - 3 after 150000ms))`

Pasha1402, 2024-12-28:
OTA for TS0601 might fail, as often happens, due to end device doing into sleep too fast. Try keeping it awake by poking sensing surface repeatedly. This would, likely, help to avoid timeout issue. I managed to update all my 5 thermostats after bunch of attempts.

On the other hand, initial issue still exists in some form: both Firmware Build Date and Firmware Version columns for TS0601-s are blank. It doesn't prevent Z2M to see there is an update, perform update and not asking for OTA again after successful update. That means internally everything works correct, but visuals quite broken

RubenKelevra, 2025-03-08:
Still the case for 2.1.3:

[image]

Ricc68, 2025-05-03:
Fixed by mentioned PR https://github.com/nurikk/zigbee2mqtt-frontend/pull/2470.