Koenkk/zigbee2mqtt #15841: Xiaomi Aqara H1 Switch is toggled by IKEA E1810 remote control

nulano, 2022-12-31:
### What happened?

I added both the [IKEA E1810 remote control](https://www.zigbee2mqtt.io/devices/E1524_E1810.html#ikea-e1524%252Fe1810) and the [Aqara Wall Switch H1 EU (with neutral, single rocker)](https://www.zigbee2mqtt.io/devices/WS-EUK03.html#xiaomi-ws-euk03) to the same network. After some time, the IKEA device decided to route itself through the Aqara wall switch. Now every time I press the toggle button on the IKEA remote control, it toggles the Aqara wall switch (in addition to my unrelated HA automation). I do not believe I changed any relevant settings from the defaults (I changed the led settings and coupled-uncoupled mode for the Aqara wall switch, but with no changes).

This issue seems to be similar to #13903, but that is for a different Aqara device, so I'm not sure if I should have opened a new issue or commented there.

Before adding the Aqara switch I had already updated the IKEA remote control to 2.3.080 from 2.3.014.
The Aqara switch is reporting firmware 11-10-2021 with no updates available.

I captured a snapshot of the ZigBee traffic when the IKEA remote control toggles the Aqara wall switch. 0xfdce is the IKEA remote control, 0x8432 is the Aqara wall switch, 0x0000 is the coordinator. Because I used the same dongle for sniffing I had to stop Zigbee2mqtt temporarily, so the wall switch is reporting a link failure. I can provide the pcapng file if that would be helpful.

```
No.	Time		Source	Destination	Dst Port Protocol	Length	Info
1	0.000000000	0xfdce	0x0000		17754	ZigBee HA	122	ZCL OnOff: Toggle, Seq: 57
2	0.139458300	0x8432	0xfdce		17754	ZigBee HA	124	ZCL: Default Response, Seq: 55
3	0.155410320	0x8432	0xfdce		17754	ZigBee		131	Network Status, 0x0000: Non-tree Link Failure
4	0.424617233	0x8432	Broadcast	17754	ZigBee		125	Route Request, Dst: 0x0000, Src: 0x8432
```

### What did you expect to happen?

The IKEA remote control does not randomly trigger a different device without being set up to do so.

### How to reproduce it (minimal and precise)

Pair IKEA remote control to Zigbee2mqtt via the Aqara Wall Switch and press the toggle button on the IKEA remote.

### Zigbee2MQTT version

1.28.4 commit: 52e545f

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus (ZBDongle-E)

cddu33, 2023-01-24:
Hi, 
same with Ikea toogle e1743 with last firmware

closed 2023-03-03 by github-actions[bot]

Censored3, 2023-03-25:
I'm in the same boat - did you manage to downgrade your firmware (or another solution)? I cannot seem to find it anywhere :-(
Fyi It happened to me also with the Sonoff SNZB-01 buttons.

cddu33, 2023-03-25:
sorry no... I included it asking to go through another device and it worked for a while. If you find a way to downgrade the firmware I'm interested

nulano, 2023-03-25:
I think the current firmware is the only one they have ever released for these.

I used a HomeAssistant automation as a workaround - if the light switch changes states within about 3 seconds of the IKEA button being pressed, I have it automatically turn off the light. But that is only usable in some rooms, not everywhere, so I'm not going to buy more of these like I planned until this is fixed.

Censored3, 2023-03-25:
Needless to say, I regret upgrading them. I'm surprised this firmware is still available @ z2m..

Meanwhile, I migrated them to a separate zigbee network until a fix is available. I'm lucky I had a Conbee still laying around...

baierjan, 2024-05-08:
Thank you for creating this issue. At least now I know I am not alone with this problem... I can only confirm the problem exists for both the WS-EUK03 (a.k.a. lumi.switch.n1aeu1) and WS-EUK04 (lumi.switch.n2aeu1) with combination with almost any IKEA remote (I tried E2001/E2002, E2123, E1744 and E2013). So far it seems that the only available workaround is indeed to isolate the Aqara stuff into separate zigbee network.

wagaman, 2024-06-30:
E2123 … the issue persists