Koenkk/zigbee2mqtt #15849: wrong TUYA device detected

BePalm2022, 2023-01-01:
### What happened?

Zigbee ID: TS011F | _TZ3000_mlswgkc3
is not detected as  a Tuya Universal Wall Socket 2 USB
shows the wrong model and function
instead is shows Model     TS011F_plug_1

correct wall socket looks like Zemismart ZIGBEE-B09-UK.

No issue to connect with the righrt model in the Tuya app, but not in Z2M

### What did you expect to happen?

How to change the type to the correct one , whichis already under the known devices?

my Tuya zigbee model is also called     TS011F
but only can be found under wall socket w 2USB 

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20220219

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Debug log

info  2023-01-01 14:03:26: Successfully interviewed '0xa4c138fc2f80c74d', device has successfully been paired
info  2023-01-01 14:03:26: Device '0xa4c138fc2f80c74d' is supported, identified as: TuYa Smart plug (with power monitoring) (TS011F_plug_1)
info  2023-01-01 14:03:26: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart plug (with power monitoring)","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":7,"description":"Recover state after power outage","name":"power_outage_memory","property":"power_outage_memory","type":"enum","values":["on","off","restore"]},{"access":7,"description":"LED indicator mode","name":"indicator_mode","property":"indicator_mode","type":"enum","values":["off","off/on","on/off","on"]},{"access":1,"description":"Instantaneous measured power","name":"power","property":"power","type":"numeric","unit":"W"},{"access":1,"description":"Instantaneous measured electrical current","name":"current","property":"current","type":"numeric","unit":"A"},{"access":1,"description":"Measured electrical potential value","name":"voltage","property":"voltage","type":"numeric","unit":"V"},{"access":1,"description":"Sum of consumed energy","name":"energy","property":"energy","type":"numeric","unit":"kWh"},{"features":[{"access":3,"description":"Enables/disables physical input on the device","name":"state","property":"child_lock","type":"binary","value_off":"UNLOCK","value_on":"LOCK"}],"type":"lock"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS011F_plug_1","options":[{"access":2,"description":"Calibrates the power value (percentual offset), takes into effect on next report of device.","name":"power_calibration","property":"power_calibration","type":"numeric"},{"access":2,"description":"Calibrates the current value (percentual offset), takes into effect on next report of device.","name":"current_calibration","property":"current_calibration","type":"numeric"},{"access":2,"description":"Calibrates the voltage value (percentual offset), takes into effect on next report of device.","name":"voltage_calibration","property":"voltage_calibration","type":"numeric"}],"supports_ota":true,"vendor":"TuYa"},"friendly_name":"0xa4c138fc2f80c74d","ieee_address":"0xa4c138fc2f80c74d","status":"successful","supported":true},"type":"device_interview"}'
debug 2023-01-01 14:03:30: Received Zigbee message from '0xa4c138fc2f80c74d', type 'raw', cluster '57344', data '{"data":[8,85,10,1,208,72,2,0,6],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2023-01-01 14:03:30: No converter available for 'TS011F_plug_1' with cluster '57344' and type 'raw' and data '{"data":[8,85,10,1,208,72,2,0,6],"type":"Buffer"}'
debug 2023-01-01 14:03:30: Received Zigbee message from '0xa4c138fc2f80c74d', type 'raw', cluster '57344', data '{"data":[8,85,10,1,208,72,2,0,6],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2023-01-01 14:03:30: No converter available for 'TS011F_plug_1' with cluster '57344' and type 'raw' and data '{"data":[8,85,10,1,208,72,2,0,6],"type":"Buffer"}'
debug 2023-01-01 14:03:30: Received Zigbee message from '0xa4c138fc2f80c74d', type 'raw', cluster '57344', data '{"data":[8,85,10,1,208,72,2,0,6],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2023-01-01 14:03:30: No converter available for 'TS011F_plug_1' with cluster '57344' and type 'raw' and data '{"data":[8,85,10,1,208,72,2,0,6],"type":"Buffer"}'
debug 2023-01-01 14:03:30: Received Zigbee message from '0xa4c138fc2f80c74d', type 'raw', cluster '57344', data '{"data":[8,85,10,1,208,72,2,0,6],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2023-01-01 14:03:30: No converter available for 'TS011F_plug_1' with cluster '57344' and type 'raw' and data '{"data":[8,85,10,1,208,72,2,0,6],"type":"Buffer"}'
debug 2023-01-01 14:03:30: Received Zigbee message from '0xa4c138fc2f80c74d', type 'raw', cluster '57344', data '{"data":[8,85,10,1,208,72,2,0,6],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2023-01-01 14:03:30: No converter available for 'TS011F_plug_1' with cluster '57344' and type 'raw' and data '{"data":[8,85,10,1,208,72,2,0,6],"type":"Buffer"}'
debug 2023-01-01 14:03:31: Received Zigbee message from '0xa4c138fc2f80c74d', type 'raw', cluster '57344', data '{"data":[8,86,10,2,208,72,2,0,10],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2023-01-01 14:03:31: No converter available for 'TS011F_plug_1' with cluster '57344' and type 'raw' and data '{"data":[8,86,10,2,208,72,2,0,10],"type":"Buffer"}'
debug 2023-01-01 14:03:31: Received Zigbee message from '0xa4c138fc2f80c74d', type 'attributeReport', cluster 'genOnOff', data '{"32768":0,"moesStartUpOnOff":2,"offWaitTime":0,"onOff":0,"onTime":0,"tuyaBacklightMode":1,"tuyaBacklightSwitch":1}' from endpoint 1 with groupID 0
info  2023-01-01 14:03:31: MQTT publish: topic 'zigbee2mqtt/0xa4c138fc2f80c74d', payload '{"child_lock":"UNLOCK","current":null,"device":{"applicationVersion":67,"dateCode":"","friendlyName":"0xa4c138fc2f80c74d","hardwareVersion":1,"ieeeAddr":"0xa4c138fc2f80c74d","manufacturerID":4417,"manufacturerName":"_TZ3000_mlswgkc3","model":"TS011F_plug_1","networkAddress":39637,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":null,"indicator_mode":"off/on","linkquality":120,"power":null,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":"unknown","state":null},"update_available":null,"voltage":null}'
debug 2023-01-01 14:03:31: Received Zigbee message from '0xa4c138fc2f80c74d', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"powerOnBehavior":2,"switchType":0}' from endpoint 1 with groupID 0
debug 2023-01-01 14:03:31: No converter available for 'TS011F_plug_1' with cluster 'manuSpecificTuya_3' and type 'attributeReport' and data '{"powerOnBehavior":2,"switchType":0}'
debug 2023-01-01 14:03:31: Received Zigbee message from '0xa4c138fc2f80c74d', type 'attributeReport', cluster 'genOnOff', data '{"offWaitTime":0,"onOff":0,"onTime":0}' from endpoint 2 with groupID 0
info  2023-01-01 14:03:31: MQTT publish: topic 'zigbee2mqtt/0xa4c138fc2f80c74d', payload '{"child_lock":"UNLOCK","current":null,"device":{"applicationVersion":67,"dateCode":"","friendlyName":"0xa4c138fc2f80c74d","hardwareVersion":1,"ieeeAddr":"0xa4c138fc2f80c74d","manufacturerID":4417,"manufacturerName":"_TZ3000_mlswgkc3","model":"TS011F_plug_1","networkAddress":39637,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":null,"indicator_mode":"off/on","linkquality":116,"power":null,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":"unknown","state":null},"update_available":null,"voltage":null}'
debug 2023-01-01 14:03:31: Received Zigbee message from '0xa4c138fc2f80c74d', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"powerOnBehavior":2}' from endpoint 2 with groupID 0
debug 2023-01-01 14:03:31: No converter available for 'TS011F_plug_1' with cluster 'manuSpecificTuya_3' and type 'attributeReport' and data '{"powerOnBehavior":2}'
debug 2023-01-01 14:03:32: Received Zigbee message from '0xa4c138fc2f80c74d', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2023-01-01 14:03:48: Saving state to file /app/data/state.json

Koenkk, 2023-01-01:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-01-01 by Koenkk

BePalm2022, 2023-01-04:
thanks
works as expected and the right unit is recognized and has the right function.

..as I bought it from China , I don't think the Manufacturer is Zemismart...but no issue on this