Koenkk/zigbee2mqtt #15853: Docker issue after 1.29.0

Tolriq, 2023-01-01:
### What happened?

After update image is broken

Logs shows:
```
Using '/app/data' as data directory

/usr/local/bin/docker-entrypoint.sh: exec: line 17: npm: not found
```

### Zigbee2MQTT version

1.29.0

### Adapter firmware version

N/A

### Adapter

Conbee2

### Debug log

Using '/app/data' as data directory

/usr/local/bin/docker-entrypoint.sh: exec: line 17: npm: not found

goessensj, 2023-01-01:
Same issue here

Tolriq, 2023-01-01:
From https://github.com/Koenkk/zigbee2mqtt/commit/8d8c46ecaef526a2077d740eb292f196f38a7962 seems like when updating docker via like portainer with a pull latest image, it does not run the necessary steps to install the missing pieces.

It probably require some migration steps, but I don't really know how.

Backup the config and restart from 1.29 probably works, will wait for @Koenkk feedback before attempting this.

Koenkk, 2023-01-02:
It seems you are overriding the default command (probably via portainer). `npm` has been removed from the image to reduce the image size. So either don't override the default command or change it to `/sbin/tini -- node index.js`

Tolriq, 2023-01-02:
I have not changed anything.

[image]

Koenkk, 2023-01-03:
And here https://docs.portainer.io/user/docker/containers/advanced ?

Tolriq, 2023-01-03:
Nope no override, but I just started fresh as a good test for my backup strategy and it works.

So I probably missed something at some point in a previous release, this container was quite old.

Anyway thanks for the help.

closed 2023-01-03 by Tolriq