Koenkk/zigbee2mqtt #15854: Sorting devices on Availability doesnt sort Disabled items together

ahd71, 2023-01-01:
### What happened?

Minor issue but sorting of Availability doesn't seemt to respect the new Disabled state yet.

### What did you expect to happen?

Group them together (online, offline and disabled devices in three different consecutive blocks)

### How to reproduce it (minimal and precise)

Disable at least a few devices and sort all

### Zigbee2MQTT version

1.29

### Adapter firmware version

NA

### Adapter

NA (but I have zzh)

ahd71, 2023-01-01:
[image]

ahd71, 2023-01-01:
[image]

Koenkk, 2023-01-02:
Please create issues related to the frontend here: https://github.com/nurikk/zigbee2mqtt-frontend

closed 2023-01-02 by Koenkk