Koenkk/zigbee2mqtt #15856: Update 1.29.0.1 - big problem: Front end not loading, devices not working > wife mad

kiekerjan18, 2023-01-01:
### What happened?

This morning I've updated to 1.29.0.1. 

Since then it is not possible for me anymore to load the frond end and see what is going on. Also my devices like switches, lights are not responding anymore. I see that the sensors like temperature, motion, door sensors are still working.

On hindsight I had similar issues in the dev versions of last days (I was trying to fix OTA update of some devices, like Philips Hue, which I read that was fixed). But I had identical issues as current stable version. But when I reverted back to stable 1.24.0 I had no issues at all and everything worked accordingly (beside OTA updates). 

But now my stable 1.29.0.1 has same issues I can't roll-back. Any option to rollback? Please help!

The logs don't produce that much information:

`error 2023-01-01 12:39:41: Failed to read state of 'Light Media' after reconnect (Read 0x001788010b82070f/11 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
`

### Zigbee2MQTT version

1.29.0.1 stable

### Adapter firmware version

20221102

### Adapter

Sonoff Zigbee Dongle Plus P

kiekerjan18, 2023-01-01:
I used the option in Home Assistant to restore a back-up before the update back to 1.28.4. Gladly everything is again back to normal. 

I've tried the latest dev / edge version and has the same isssue, so I can use that one to simulate and help / validate for potential future fixes.

dereksanders7, 2023-01-01:
For those that don't have the ability to roll back, you can use `git checkout tags/1.28.4` in place of the `git pull` command you would normally use in your update process. All other steps remain the same.

Chacsam, 2023-01-01:
Same issue here. 
No front-end, all devices unavailable since update to 1.29.0-1 :(
How could I downgrade on HA on a VM?

kiekerjan18, 2023-01-01:
I did in HA:

Settings > System > Backups 

In then restored the latest backup (automatically made by HA)
- addon_45df7312_zigbee2mqtt_1.28.4-1

Hope this helps

lux73, 2023-01-01:
i would never understood why you Guys don't use Snapshot's in your VM Envirnment - with that you could easily rollback the whole VM

Chacsam, 2023-01-01:
> I did in HA:
> 
> Settings > System > Backups
> 
> In then restored the latest backup (automatically made by HA)
> 
> * addon_45df7312_zigbee2mqtt_1.28.4-1
> 
> Hope this helps

Indeed, it fixed it

kiekerjan18, 2023-01-01:
> > I did in HA:
> > Settings > System > Backups
> > In then restored the latest backup (automatically made by HA)
> > 
> > * addon_45df7312_zigbee2mqtt_1.28.4-1
> > 
> > Hope this helps
> 
> Indeed, it fixed it

Don't forget to uncheck the auto-update function in the Addons, otherwise you need to redo this after an hour (like I had to do)

kiekerjan18, 2023-01-01:
> i would never understood why you Guys don't use Snapshot's in your VM Envirnment - with that you could easily rollback the whole VM

That's just briljant.. never came to my mind. Will experiment with it, thanks for the tip!

Xitro01, 2023-01-01:
Same here, all my lights work properly, but my curtains are now showing as “opening” or “closing” in the frontend and nothing else. While the current state shows correctly in Z2M, just not in the frontend.

BjornHgh, 2023-01-02:
Rolled back a snapshot due to all devices offline after updating to 1.29.0.1
Z2M back online, but all devices stay offline, apparently have to reset all of them.
Our Financial Manager is not amused either. :) Shortcut anyone?

Koenkk, 2023-01-02:
@kiekerjan18 can you provide the herdsman debug log from starting z2m until this error?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

gras55, 2023-01-02:
Have noticed that starting from 1.29.0.1 my dimmers (TS110E) were not reporting their state to HA.  Upon further analysis I discovered this 
"state": "OFF",
"state_1": "ON"
it was toggling state_1 rather than state upon reverting to 1.28 all problems vanished

BjornHgh, 2023-01-02:
Finally I went for 1.29.0.1 and re-pairing all my zigbee hardware.
New VM snapshots were immediately taken.

kiekerjan18, 2023-01-02:
> @kiekerjan18 can you provide the herdsman debug log from starting z2m until this error?
> 
> See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

I am quite new to this and I got stuck in the guide in fetching the log finding the container:

[image]

I am using Home Assistant with the following repository integrated:
https://github.com/zigbee2mqtt/hassio-zigbee2mqtt

Home Assistant 2022.12.8
Supervisor 2022.12.1
Operating System 9.4
Frontend 20221213.1 - latest

jjansen85, 2023-01-02:
Same issue here (docker install on Synology NAS). _Wife not mad yet ;-)_
Frontend not always loading or very slow. Devices did not respond or delayed.

Restored to 1.28.4 which solved the issue.

maxbec, 2023-01-02:
Same for me (docker on Synology NAS).

szymucha94, 2023-01-02:
> Same here, all my lights work properly, but my curtains are now showing as “opening” or “closing” in the frontend and nothing else. While the current state shows correctly in Z2M, just not in the frontend.

#15895 
Hopefully it gets fixed :(

jjansen85, 2023-01-02:
> Same issue here (docker install on Synology NAS). _Wife not mad yet ;-)_ Frontend not always loading or very slow. Devices did not respond or delayed.
> 
> Restored to 1.28.4 which solved the issue.

Seems to be resolved in the dev version
_1.29.0-dev commit: [826f032](https://github.com/Koenkk/zigbee2mqtt/commit/826f032)_

kiekerjan18, 2023-01-02:
> Seems to be resolved in the dev version _1.29.0-dev commit: [826f032](https://github.com/Koenkk/zigbee2mqtt/commit/826f032)_

Unfortunately not for me.

jjansen85, 2023-01-02:
> Unfortunately not for me.

Apparently you are right. After about an hour, it started malfunctioning again :-(

kiekerjan18, 2023-01-02:
Logging:

`[21:08:13] INFO: Preparing to start...
[21:08:14] INFO: Socat not enabled
[21:08:16] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:error 2023-01-02 21:10:03: Failed to configure 'Light Voordeur', attempt 1 (Error: Bind 0xa4c13804100bdc0b/1 genOnOff from '0x00124b0024c04c3e/1' failed (SRSP - ZDO - bindReq after 6000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:559:17)
    at processTimers (node:internal/timers:502:7))
Zigbee2MQTT:error 2023-01-02 21:13:37: Not connected to MQTT server!
Zigbee2MQTT:error 2023-01-02 21:14:39: Not connected to MQTT server!
Zigbee2MQTT:error 2023-01-02 21:14:39: Cannot send message: topic: 'zigbee2mqtt/Thermometer WC', payload: '{"battery":70,"humidity":60.3,"linkquality":142,"power_outage_count":20335,"pressure":1023.77,"temperature":18.41,"voltage":2955}
Zigbee2MQTT:error 2023-01-02 21:14:39: Not connected to MQTT server!
Zigbee2MQTT:error 2023-01-02 21:14:39: Cannot send message: topic: 'zigbee2mqtt/Light Schuur', payload: '{"child_lock":"UNLOCK","current":0.06,"energy":17.08,"indicator_mode":"off/on","linkquality":43,"power":8,"power_outage_memory":"off","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":false,"voltage":246}
Zigbee2MQTT:error 2023-01-02 21:14:39: Not connected to MQTT server!
Zigbee2MQTT:error 2023-01-02 21:14:39: Cannot send message: topic: 'zigbee2mqtt/Light Schuur', payload: '{"child_lock":"UNLOCK","current":0.06,"energy":17.08,"indicator_mode":"off/on","linkquality":43,"power":8,"power_outage_memory":"off","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":false,"voltage":234}
Zigbee2MQTT:error 2023-01-02 21:14:39: Not connected to MQTT server!
Zigbee2MQTT:error 2023-01-02 21:14:39: Cannot send message: topic: 'zigbee2mqtt/Blinds Achter Smal', payload: '{"battery":76,"charging_status":false,"device_temperature":20,"linkquality":21,"motor_state":"pause","position":0,"power_outage_count":0,"running":false,"state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":false}
Zigbee2MQTT:error 2023-01-02 21:14:39: Not connected to MQTT server!
Zigbee2MQTT:error 2023-01-02 21:14:39: Cannot send message: topic: 'zigbee2mqtt/Light Zwembad', payload: '{"child_lock":"UNLOCK","current":0,"energy":2.49,"indicator_mode":"off/on","linkquality":47,"power":0,"power_outage_memory":"off","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":false,"voltage":234}
Zigbee2MQTT:error 2023-01-02 21:14:39: Not connected to MQTT server!
Zigbee2MQTT:error 2023-01-02 21:14:39: Cannot send message: topic: 'zigbee2mqtt/Light Tuinlampen Achter', payload: '{"child_lock":"UNLOCK","current":0.06,"energy":32.6,"indicator_mode":"off/on","linkquality":72,"power":11,"power_outage_memory":"off","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":false,"voltage":234}
Zigbee2MQTT:error 2023-01-02 21:14:39: Not connected to MQTT server!
Zigbee2MQTT:error 2023-01-02 21:14:39: Cannot send message: topic: 'zigbee2mqtt/Zolder werkkamer', payload: '{"battery":100,"humidity":67.39,"linkquality":153,"temperature":19.38,"voltage":3000}
Zigbee2MQTT:error 2023-01-02 21:14:39: Not connected to MQTT server!
Zigbee2MQTT:error 2023-01-02 21:15:37: Not connected to MQTT server!
Zigbee2MQTT:error 2023-01-02 21:15:37: Cannot send message: topic: 'zigbee2mqtt/Switch Media', payload: '{"child_lock":"UNLOCK","current":0.67,"energy":836.56,"indicator_mode":"off/on","linkquality":153,"power":114,"power_outage_memory":"off","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":false,"voltage":232}
Zigbee2MQTT:error 2023-01-02 21:15:37: Not connected to MQTT server!
Zigbee2MQTT:error 2023-01-02 21:15:37: Cannot send message: topic: 'zigbee2mqtt/Thermometer Keuken', payload: '{"battery":63,"humidity":55.59,"linkquality":134,"power_outage_count":11881,"pressure":1034.78,"temperature":21.18,"voltage":2945}
Zigbee2MQTT:error 2023-01-02 21:15:37: Not connected to MQTT server!
Zigbee2MQTT:error 2023-01-02 21:15:37: Cannot send message: topic: 'zigbee2mqtt/Switch Media', payload: '{"child_lock":"UNLOCK","current":0.67,"energy":836.57,"indicator_mode":"off/on","linkquality":153,"power":114,"power_outage_memory":"off","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":false,"voltage":232}
Zigbee2MQTT:error 2023-01-02 21:15:37: Not connected to MQTT server!
Zigbee2MQTT:error 2023-01-02 21:15:37: Cannot send message: topic: 'zigbee2mqtt/Switch Zwembad Heating', payload: '{"child_lock":null,"current":0,"energy":0.02,"indicator_mode":null,"linkquality":61,"power":0,"power_outage_memory":null,"state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":false,"voltage":233}
Zigbee2MQTT:error 2023-01-02 21:15:38: Not connected to MQTT server!
Zigbee2MQTT:error 2023-01-02 21:15:38: Not connected to MQTT server!
Zigbee2MQTT:error 2023-01-02 21:15:38: Cannot send message: topic: 'zigbee2mqtt/Motion Sensor Woonkamer', payload: '{"battery":100,"device_temperature":23,"illuminance":34,"illuminance_lux":34,"linkquality":94,"occupancy":false,"power_outage_count":16,"voltage":3005}`

Koenkk, 2023-01-03:
@kiekerjan18 it seems your adapter is disconnecting. Since you are using HA, can you make sure ZHA is disabled? This could interfere with z2m.

kiekerjan18, 2023-01-03:
> @kiekerjan18 it seems your adapter is disconnecting. Since you are using HA, can you make sure ZHA is disabled? This could interfere with z2m.

ZHA is not installed. When I switch to 1.24.0 it is all working according. For the record, these logs are of latest dev / edge release since symptoms are the same as one stable release.

jjansen85, 2023-01-03:
Below my herdsman logs
[_zigbee2mqtt_herdsman_logs.txt](https://github.com/Koenkk/zigbee2mqtt/files/10336550/_zigbee2mqtt_herdsman_logs.txt)

thefeli73, 2023-01-03:
Same issue for me, but only when trying to access it externally (https://ha.mydomain.com i.e. through cloudflare -> reverse proxy -> frontend). 
When accessing locally (http://homeassistant:8123/[z2m]) it works fine. Also no problem with devices
From what i can tell from browser console the issue seems to be `Uncaught SyntaxError: Invalid or unexpected token (at index.3d898417.js:1:5577)`

kiekerjan18, 2023-01-03:
Ok, it took me a while to figure out how to collect the herdsman log. But hereby the logging. Hope it can be fixed.

[herdsman_log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10340320/herdsman_log.txt)

Koenkk, 2023-01-04:
@kiekerjan18 I see you are running a dev version of the firmware (`20221102`). This one is known to be unstable. Can you try with the following fw: https://github.com/Koenkk/Z-Stack-firmware/tree/6.10.01.01/coordinator/Z-Stack_3.x.0/bin

jjansen85, 2023-01-04:
@Koenkk I also use the `20221102` firmware. 
It ran for weeks without any problem on the previous releases. Why do you suspect the firmware to create issues with the current release?

Koenkk, 2023-01-04:
@jjansen85 I'm not sure what in 1.29.0 could cause this, but I just want to rule out an unstable firmware causing this

kiekerjan18, 2023-01-04:
> @kiekerjan18 I see you are running a dev version of the firmware (`20221102`). This one is known to be unstable. Can you try with the following fw: https://github.com/Koenkk/Z-Stack-firmware/tree/6.10.01.01/coordinator/Z-Stack_3.x.0/bin

sadly this didn't do the trick.

New log file:
[herdsman_log_20230104.txt](https://github.com/Koenkk/zigbee2mqtt/files/10346701/herdsman_log_20230104.txt)

jjansen85, 2023-01-04:
Updated to the suggested firmware; no success. Had the idea the problem got worse. Strange thing is, sometimes it operates OK for a while after restart (like half an hour / hour). Sometimes it 'fails' after a few minutes.

Koenkk, 2023-01-05:
@kiekerjan18 in https://github.com/Koenkk/zigbee2mqtt/issues/15856#issuecomment-1370319158 you mention that everything is working fine on 1.24.0 and provided a log file. However this log is of 1.28.4 (line 8: `Zigbee2MQTT:info  2023-01-04 00:16:41: Starting Zigbee2MQTT version 1.28.4 (commit #unknown)`). Can you confirm that by just downgrading to 1.28.4 the problems are fixed? (and no other changes like HA-OS downgrades)

kiekerjan18, 2023-01-05:
Hi @Koenkk, I have Z2M & Z2M Edge both installed. What I did is start the Z2M Edge (i update it everytime to latest 1.29.0) and if that one still has issues like described I go back to Z2M which is still on 1.28.4.

Earlier I had installed the 1.29.0 stable version and then I did a partial restore backup only on Z2M addon and went back to 1.28.4 and everything worked fine.

Btw, yesterday I upgraded firmware, today I also have some issues in 1.28.4 of losing devices and their connectivity. I will dive into that this evening when I am back home.

thefeli73, 2023-01-05:
> Same issue for me, but only when trying to access it externally (https://ha.mydomain.com i.e. through cloudflare -> reverse proxy -> frontend). When accessing locally ([http://homeassistant:8123/[z2m]](http://homeassistant:8123/%5Bz2m%5D)) it works fine. Also no problem with devices From what i can tell from browser console the issue seems to be `Uncaught SyntaxError: Invalid or unexpected token (at index.3d898417.js:1:5577)`
> 
> Edit: Running in Home Assistant. Doesn't seem to matter whether it is accessed through the HA sidepanel (http://[host]/zigbee2mqtt/dashboard) or "directly" (http://[host]/api/hassio_ingress/[numbers]/#/dashboard)

Updated to Home Assistant 2023.1.0 and did a [Empty Cache and Hard Reload] and now it works again, no clue what was wrong but probably not the same issue as OP anyway

kiekerjan18, 2023-01-05:
I've tried that as well, but both internal as external same issues. Also on HA 2023.1.0. I think different issues, but good to hear yours are solved!

jjansen85, 2023-01-05:
> Btw, yesterday I upgraded firmware, today I also have some issues in 1.28.4 of losing devices and their connectivity. I will dive into that this evening when I am back home.

Experienced the same issue. After FW upgrade to `20221226` I started to have a lot of connectivity issues (and timeouts and routing errors in Z2M logs) within 1.28.4. 
I downgraded FW to the last stable `20220219`, but issues where not solved.
I did some power cycles and reconfigurations for the impacted devices, which seems to solve the issues.
Now back at `20221226`, which seems to be stable :-)
Will try updating to 1.29 later.

kiekerjan18, 2023-01-05:
> Experienced the same issue. After FW upgrade to `20221226` I started to have a lot of connectivity issues (and timeouts and routing errors in Z2M logs) within 1.28.4. I downgraded FW to the last [...]

Yes, I had similar experience. Still issues with 1.29, but firmware seems to be ok now.

jjansen85, 2023-01-06:
> Yes, I had similar experience. Still issues with 1.29, but firmware seems to be ok now.

1.29 (latest dev) + 20221226 is also very unstable for me.

Koenkk, 2023-01-06:
@jjansen85 I think it is indeed related the the serialport update (from version 9 to 10). However I don't understand why only you are affected by this. Are you also getting the `failed (SRSP - AF - dataRequest after 6000ms))` errors? If yes, can you provide the log when running with `DEBUG=*` instead of `DEBUG=zigbee-herdsman*` as explained on https://www.zigbee2mqtt.io/guide/usage/debug.html#enabling-logging

jjansen85, 2023-01-06:
@Koenkk Yes, I get this error. Enabled `DEBUG=*` on latest dev.

[_zigbee2mqtt_logs (1).txt](https://github.com/Koenkk/zigbee2mqtt/files/10359227/_zigbee2mqtt_logs.1.txt)

Gives also a lot of these messages:
```
2023-01-06T09:59:02.539Z serialport/stream _read reading { start: 23334, toRead: 42202 }
2023-01-06T09:59:02.539Z serialport/bindings-cpp read
2023-01-06T09:59:02.539Z serialport/bindings-cpp/unixRead Starting read
2023-01-06T09:59:02.540Z serialport/bindings-cpp/unixRead read error [Error: EAGAIN: resource temporarily unavailable, read] {
  errno: -11,
  code: 'EAGAIN',
  syscall: 'read'
}
2023-01-06T09:59:02.540Z serialport/bindings-cpp/unixRead waiting for readable because of code: EAGAIN
```

Koenkk, 2023-01-06:
@jjansen85 interesting.

Let's first confirm it is indeed caused by serialport 10. I've downgraded to serialport 9 in the dev/edge version (changes will be available in 1 hour from now). Can you check that?

cloudbr34k84, 2023-01-06:
Just changing the serial port cause all this instability and slow responsiveness?? Sorry I have no idea about any of this.

kiekerjan18, 2023-01-06:
Sadly, this didn't do the trick. Although there seems some delayed response now. I still can't access the Z2M front end.

[log_herdsman_20230106.txt](https://github.com/Koenkk/zigbee2mqtt/files/10359569/log_herdsman_20230106.txt)

I made a small video
https://user-images.githubusercontent.com/63964079/210999842-2f89f026-505d-4472-b9b0-c40e90c7fbbd.mp4

cloudbr34k84, 2023-01-06:
> Sadly, this didn't do the trick. Although there seems some delayed response now. I still can't access the Z2M front end. [...]

I'm experiencing this as well

jjansen85, 2023-01-06:
> @jjansen85 interesting.
> 
> Let's first confirm it is indeed caused by serialport 10. I've downgraded to serialport 9 in the dev/edge version (changes will be available in 1 hour from now). Can you check that?

Unfortunately, no positive effect.

Koenkk, 2023-01-07:
@kiekerjan18 I reverted more changes (**only for HA addon**), can you check again with the latest edge version in 30 mins from now?

@jjansen85 the `EAGAIN: resource temporarily unavailable, read` are normal indeed.

kiekerjan18, 2023-01-07:
> @kiekerjan18 I reverted more changes (**only for HA addon**), can you check again with the latest edge version in 30 mins from now?
> 
> @jjansen85 the `EAGAIN: resource temporarily unavailable, read` are normal indeed.

Sorry, no success... are we running out of options?

Hereby the log. 
[log_herdsman_20230107.txt](https://github.com/Koenkk/zigbee2mqtt/files/10366111/log_herdsman_20230107.txt)

Koenkk, 2023-01-08:
@kiekerjan18 can you contact me on telegram or discord? (@koenkk)

kiekerjan18, 2023-01-08:
On discord. Sent you a direct message

cloudbr34k84, 2023-01-08:
Came home to even more devices offline. Lights on in the house but not showing Z2M. Nothing working.

Sundown89, 2023-01-09:
Downgrade to 1.28.4 did not fix the issue for me. Restored the addon through HA as mentioned here before. No access to frontend and same issue with delayed or no responses. Do I also have to rollback to a certain version of HA? I run complete HA OS in a VirtualBox VM.

vulpes2, 2023-01-09:
I can't be of much help in debugging but I thought it might be useful to share my experiences here. 1.29 broke all of my devices but not the web interface. Battery powered ones were all offline, mains powered devices (Tradfri bulbs) were online but would not respond to any commands. I'm running z2m in a docker container so rolling back to 1.28.4 was fairly easy, and that fixed all but two devices. 

There's an E1812 (Tradfri shortcut button) and an E1810 (Tradfri remote) that were still completely unresponsive, no signs of life even when the pairing button was pressed or held down. Batteries in both are tested to be good, and I've tried new batteries as well. The E1812 was able to reconnect after a power-cycle, but the E1810 seems to be completely dead even after leaving the battery out for a few minutes. 

I don't have much time to debug this any further right now, but I'll try leaving the batteries out of the E1810 for a day and see if it will work again. If that doesn't work, I would probably need to try powering off all zigbee routers and my coordinator before powering on the remote, and maybe buy a new remote if that still doesn't work.

cloudbr34k84, 2023-01-09:
Yeah I'm cutting my loses and getting a new coordinator.

kiekerjan18, 2023-01-09:
Working with @Koenkk on debugging of this issue. We've found which change caused the problems. Hopefully we can also find a solution, work in progress.

jjansen85, 2023-01-09:
@kiekerjan18 which change caused the issue?

kiekerjan18, 2023-01-09:
> @kiekerjan18 which change caused the issue?

Update to: zigbee-herdsman 0.14.81 it seems.

Bastiencc, 2023-01-09:
Thank you very much guys your work !

kiekerjan18, 2023-01-09:
Fix should be available in dev / edge. It worked here.

cloudbr34k84, 2023-01-09:
> Fix should be available in dev / edge. It worked here.

I didnt realise im now getting the same error as you
```
bug 10-01-2023 09:29:40: Loaded state from file /config/zigbee2mqtt/state.json
info  10-01-2023 09:29:40: Logging to console and directory: '/config/zigbee2mqtt/log/2023-01-10.09-29-40' filename: log.txt
debug 10-01-2023 09:29:40: Removing old log directory '/config/zigbee2mqtt/log/2023-01-10.07-26-20'
info  10-01-2023 09:29:40: Starting Zigbee2MQTT version 1.28.4 (commit #unknown)
info  10-01-2023 09:29:40: Starting zigbee-herdsman (0.14.76)
debug 10-01-2023 09:29:40: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[25],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyUSB0"}}'
error 10-01-2023 09:30:01: Error while starting zigbee-herdsman
error 10-01-2023 09:30:01: Failed to start zigbee
error 10-01-2023 09:30:01: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 10-01-2023 09:30:01: Exiting...
**error 10-01-2023 09:30:01: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)**
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)
```

kiekerjan18, 2023-01-09:
> I didnt realise im now getting the same error as you
>
> ```
> bug 10-01-2023 09:29:40: Loaded state from file /config/zigbee2mqtt/state.json
> info  10-01-2023 09:29:40: Logging to console and directory: [...]

tried restarting again? I sometimes see this, but when restarting it started.

[38 more replies]

closed 2023-01-11 by Koenkk