Koenkk/zigbee2mqtt #15857: Update of 'TVrum fönsterlampa Vänster' failed (Image padding contains invalid bytes)

NBonline, 2023-01-01:
### What happened?

In OTA Philips devices is marked for update. When i try to do that the error message i presented.

### What did you expect to happen?

To update firmware of my Philips devices.

### How to reproduce it (minimal and precise)

Every time I try to update Philips devices.

### Zigbee2MQTT version

1.26.0

### Adapter firmware version

20220507

### Adapter

Sonoff ZIGBEE 3.0 USB DONGLE PLUS

lukyjay, 2023-01-01:
Supposed to be fixed in the latest version.

Please search existing issues and update to the latest version of Zigbee2MQTT before opening an issue :)

NBonline, 2023-01-01:
Thank you for quick answer, I will look better next time.

NBonline, 2023-01-01:
Strange, I followed this to upgrade today, (https://www.zigbee2mqtt.io/guide/installation/01_linux.html#for-later-update-zigbee2mqtt-to-the-latest-version). But still on 1.26.0, have I done something wrong?

lukyjay, 2023-01-01:
Please go to the folder where Zigbee2MQTT is installed and run the update.sh script:
./update.sh

It should do everything for you. Not sure why it isn't listed on that page.

In case you don't have that script, here are the contents:
```
!/bin/bash -e

if [ -d data-backup ]; then
   echo "ERROR: Backup directory exists. May be previous restoring was failed?"
   echo "1. Save 'data-backup' and 'data' dirs to safe location to make possibility to restore config later."
   echo "2. Manually delete 'data-backup' dir and try again."
   exit 1
fi

if which systemctl 2> /dev/null > /dev/null; then
       echo "Stopping Zigbee2MQTT..."
       sudo systemctl stop zigbee2mqtt
else
       echo "Skipped stopping Zigbee2MQTT, no systemctl found"
fi

echo "Creating backup of configuration..."
cp -R data data-backup

echo "Updating..."
git pull

echo "Installing dependencies..."
npm ci

echo "Restore configuration..."
cp -R data-backup/* data
rm -rf data-backup

if which systemctl 2> /dev/null > /dev/null; then
       echo "Starting Zigbee2MQTT..."
       sudo systemctl start zigbee2mqtt
else
       echo "Skipped starting Zigbee2MQTT, no systemctl found"
fi

echo "Done!"

```

Just save that as update.sh, then run chmod +x update.sh, then ./update.sh

NBonline, 2023-01-01:
get the following:
ppdaterar bc4ffc00..dddce571
error: Dina lokala ändringar av följande filer skulle skrivas över av sammanslagning:
	package-lock.json
Checka in dina ändringar eller använd "stash" innan du byter gren.

Howto proceed?

lukyjay, 2023-01-01:
Did you edit package-lock.json? If not, something might have.. Save a copy by changing the filename and try again:

mv package-lock.json package-lock.json.bak

Then try again?

NBonline, 2023-01-01:
Now it seems to work.
I will check i update Philips work later.

Thanks for super support!

NBonline, 2023-01-01:
Everything works as expected.

Once again thanks for the swift support.

Regards

closed 2023-01-01 by NBonline