Koenkk/zigbee2mqtt #15858: Updating to 1.29.0 fails to start Z2M

cadavre, 2023-01-01:
### What happened?

I've updated 1.28.4 to 1.29.0.
Updated docker container via docker-compose (as I usually do).

While restarting container the Z2M won't launch due to error.

I'm using a single custom converter for Hue lightbulb.

### What did you expect to happen?

Z2M shall start.

### How to reproduce it (minimal and precise)

I'm using a single custom converter for some broken Hue lightbulb:

```js
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const ota = require('zigbee-herdsman-converters/lib/ota');
const extend = require('zigbee-herdsman-converters/lib/extend');

const hueExtend = {
    light_onoff_brightness: (options={}) => ({
        ...extend.light_onoff_brightness(options),
        toZigbee: extend.light_onoff_brightness(options).toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
    light_onoff_brightness_colortemp: (options={}) => ({
        ...extend.light_onoff_brightness_colortemp(options),
        toZigbee: extend.light_onoff_brightness_colortemp(options).toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
    light_onoff_brightness_color: (options={}) => ({
        ...extend.light_onoff_brightness_color({supportsHS: true, ...options}),
        toZigbee: extend.light_onoff_brightness_color({supportsHS: true, ...options}).toZigbee
            .concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
    light_onoff_brightness_colortemp_color: (options={}) => ({
        ...extend.light_onoff_brightness_colortemp_color({supportsHS: true, ...options}),
        toZigbee: extend.light_onoff_brightness_colortemp_color({supportsHS: true, ...options})
            .toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
};

module.exports = [
    {
        zigbeeModel: ['LTA001', '4080130P6'],
        model: '9290022169',
        vendor: 'Philips',
        description: 'Hue white ambiance E27 with Bluetooth',
        meta: {turnsOffAtBrightness1: true},
        extend: hueExtend.light_onoff_brightness_colortemp(),
        ota: ota.zigbeeOTA,
    },
];
```

### Zigbee2MQTT version

1.29.0

### Adapter firmware version

20220507

### Adapter

LAUNCHXL-CC1352P-2

### Debug log

```
zigbee2mqtt        | Using '/app/data' as data directory
zigbee2mqtt        | 
zigbee2mqtt        | /app/node_modules/zigbee-herdsman-converters/index.js:91
zigbee2mqtt        |         if (converter.options) {
zigbee2mqtt        |                       ^
zigbee2mqtt        | TypeError: Cannot read properties of undefined (reading 'options')
zigbee2mqtt        |     at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:91:23)
zigbee2mqtt        |     at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
zigbee2mqtt        |     at new Controller (/app/lib/controller.ts:84:58)
zigbee2mqtt        |     at start (/app/index.js:106:18)
```

cadavre, 2023-01-01:
OK, I removed custom converter and Z2M started.

Though, I can't remember the reason why I had to use the converter. I'll keep it for some time without it and see.

clintkev251, 2023-01-01:
I'm seeing the same thing. A custom converter I was using to add support for a hue bulb which isn't yet supported (and apologies for not yet having written a pull request for that) and this causes a failure to start. Removed it and Zigbee2MQTT starts successfully, though the devices no longer work due to the missing converter

Koenkk, 2023-01-02:
@cadavre `9290022169` is supported out-of-the-box so no need for an external converter. 

@clintkev251 please fill in https://github.com/Koenkk/zigbee2mqtt/issues/new?assignees=&labels=new+device+support&template=new_device_support.yaml&title=%5BNew+device+support%5D%3A+. then I can add it.

closed 2023-02-09 by github-actions[bot]